espressif/esp-zigbee-sdk #281: HA_ON_OFF_SWITCH cannot run:Network(0x9761) closed, devices joining not allowed.

zhang-zi-ang, 2024-03-12:
### Question

I am using the ESP32C6 development board with the official HA_ON_OFF_SWITCH example. After successful compilation, I encounter an error at runtime: 'Network(0x9761) closed, devices joining not allowed.' What could be the possible reasons for this error?

zhang-zi-ang, 2024-03-13:
I (23) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (24) boot: compile time Mar 13 2024 08:02:00
I (24) boot: chip revision: v0.0
I (26) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32c6: SPI Speed      : 80MHz
I (36) boot.esp32c6: SPI Mode       : QIO
I (41) boot.esp32c6: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 000e1000
I (85) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (92) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=42088020 size=10708h ( 67336) map
I (126) esp_image: segment 1: paddr=00020730 vaddr=40800000 size=078e8h ( 30952) load
I (134) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=86e10h (552464) map
I (245) esp_image: segment 3: paddr=000aee38 vaddr=408078e8 size=07e18h ( 32280) load
I (256) boot: Loaded app from partition at offset 0x10000
I (257) boot: Disabling RNG early entropy source...
I (268) cpu_start: Unicore app
I (269) cpu_start: Pro cpu up.
W (277) clk: esp_perip_clk_init() has not been implemented yet
I (284) cpu_start: Pro cpu start user code
I (284) cpu_start: cpu freq: 160000000 Hz
I (284) cpu_start: Application information:
I (287) cpu_start: Project name:     light_switch
I (292) cpu_start: App version:      1
I (297) cpu_start: Compile time:     Mar 13 2024 08:01:22
I (303) cpu_start: ELF file SHA256:  4c2dd97a1a2dc350...
I (309) cpu_start: ESP-IDF:          v5.1.3
I (314) cpu_start: Min chip rev:     v0.0
I (318) cpu_start: Max chip rev:     v0.99
I (323) cpu_start: Chip rev:         v0.0
I (328) heap_init: Initializing. RAM available for dynamic allocation:
I (335) heap_init: At 408192E0 len 00063330 (396 KiB): D/IRAM
I (341) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (348) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (355) spi_flash: detected chip: generic
I (359) spi_flash: flash io: qio
W (363) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (376) sleep: Configure to isolate all GPIO pins in sleep state
I (383) sleep: Enable automatic switching of GPIO sleep configuration
I (390) coexist: coex firmware version: 77cd7f8
I (396) coexist: coexist rom version 5b8dcfa
I (401) app_start: Starting scheduler on CPU0
I (405) main_task: Started on CPU0
I (405) main_task: Calling app_main()
I (415) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (415) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
I (465) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
I (475) main_task: Returned from app_main()
I (485) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (485) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
W (485) ESP_ZB_ON_OFF_SWITCH: Network(0x9761) closed, devices joining not allowed.
I (495) ESP_ZB_ON_OFF_SWITCH: Device started up in non factory-reset mode
I (495) ESP_ZB_ON_OFF_SWITCH: Device rebooted

xieqinan, 2024-03-14:
@zhang-zi-ang ，

It is a reasonable behavior for a Zigbee device, as having the network always open can be unsafe. The Zigbee network will automatically close after 108 seconds or after a reboot, which is beneficial for network safety. You can use `esp_zb_bdb_open_network()` and `esp_zb_bdb_close_network()` functions to manually open or close the network, and also enable or disable other devices from joining the network.

zhang-zi-ang, 2024-03-15:
> @zhang-zi-ang ，
>
> It is a reasonable behavior for a Zigbee device, as having the network always open can be unsafe. The Zigbee network will automatically close after 108 seconds or after a reboot, [...]

Is the network default closed in the official example HA_ON_OFF_SWITCH?According to the README.md file of the official example, I feel that it shouldn't be necessary for me to manually configure the network to use.

xieqinan, 2024-03-15:
When the HA_ON_OFF_SWICH device is in the factory-reset state and forms a network for the first time, it will keep the network open for 180 seconds. However, if it is rebooted, the network will be closed. In such a case, you can call `esp_zb_bdb_open_network()` in the `ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT` signal according to the requirements of your product.

zhang-zi-ang, 2024-03-19:
> When the HA_ON_OFF_SWICH device is in the factory-reset state and forms a network for the first time, it will keep the network open for 180 seconds. However, if it is rebooted, the network will be [...]

谢谢，问题已经解决

xieqinan, 2024-03-19:
@zhang-zi-ang ，

好的， 如果问题解决了，可以考虑关闭当前的issue。其他的关于esp-zigbee-sdk 的问题，可以另外开issue来讨论。 谢谢。

closed 2024-04-01 by zhang-zi-ang