espressif/esp-zigbee-sdk #282: Reportable attributes of ESP32-h2 zigbee thermostat cluster

ak201290, 2024-03-13:
Hi
I am trying to build ESP32-h2 based zigbee thermostat and connect it with zigbee2mqtt coordinator. The question is: how to make the thermostat attributes to be reportable, for example, SYSTEM MODE? Could you please give some code example how to do it? Thanks

xieqinan, 2024-03-14:
@ak201290 ，

The `thermostat` and `temperature sensor` device examples have been tested internally and will be made available within 1 week.

ak201290, 2024-03-15:
Thank you very much

chshu, 2024-03-15:
@ak201290 The thermostat example is available here: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_thermostat

ak201290, 2024-03-15:
Thank you very much

closed 2024-04-01 by espressif-bot

Hubro, 2025-05-22:
> [@ak201290](https://github.com/ak201290) The thermostat example is available here: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_thermostat

This example doesn't show how to make attributes of the Thermostat cluster reportable. I'm trying to make SYSTEM_MODE reportable, but if I just add `ESP_ZB_ZCL_ATTR_ACCESS_REPORTING` to the `access` field, the Zigbee stack crashes catastrophically:

```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

--- Stack dump detected
Core  0 register dump:
MEPC    : 0x42072b08  RA      : 0x42072b08  SP      : 0x40825500  GP      : 0x40810c24  
--- 0x42072b08: zb_zcl_get_next_reporting_info at ??:?
0x42072b08: zb_zcl_get_next_reporting_info at ??:?

TP      : 0x40825660  T0      : 0x40030dca  T1      : 0x0000000f  T2      : 0x000000ff  
--- 0x40030dca: memset in ROM

S0/FP   : 0x40824424  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00000100  
A2      : 0x420f45b8  A3      : 0x420f3d48  A4      : 0x00000267  A5      : 0x00000000  
A6      : 0x408254cc  A7      : 0x00000000  S2      : 0x00000001  S3      : 0x408243ec  
S4      : 0x00000000  S5      : 0x00000017  S6      : 0x00000000  S7      : 0x420f4000  
S8      : 0x00000002  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000003  
--- 0x40800001: _vector_table at /nix/store/d4n5hxa6jz5wbrfpiap341hgq73y61x2-esp-idf-v5.4.1/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000  

--- Backtrace:

0x42072b08 in zb_zcl_get_next_reporting_info ()
#0  0x42072b08 in zb_zcl_get_next_reporting_info ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

xieqinan, 2025-05-22:
@Hubro Please share your changes.

Hubro, 2025-05-23:
@xieqinan It seems clearing the flash on my device after modifying the "access" field solves the issue. I have repeated this several times.

Adding "REPORTABLE" to an access field after initial startup and joining a network will cause the device to crash with the above error any time reporting is attempted, until the device flash is cleared (leaving/rejoining network was not enough).