espressif/esp-zigbee-sdk #283: Сorrect procedure for removing a zigbee device from the gateway

iamalek, 2024-03-13:
### Question

I'm making my Zigbee gateway based on ESP32-C6. I figured out how to bind the zigbee device, but the functional sequence for removing the device from the gateway is completely unclear. Please tell me the correct procedure.
You need to understand how to connect all these functions **esp_zb_zdo_device_leave_req()**, **esp_zb_zdo_device_unbind_req()** and other necessary ones.

xieqinan, 2024-03-14:
Hello @iamalek,

If the `esp_zb_zdo_device_leave_req()` is executed successfully, it will clear all information for the network, including non-volatile data. However, the endpoint information will still exist. You can use `esp_zb_zcl_reset_all_endpoints_to_factory_default()` to clear it.

iamalek, 2024-03-14:
> Hello @iamalek,
> 
> If the `esp_zb_zdo_device_leave_req()` is executed successfully, it will clear all information for the network, including non-volatile data. However, the endpoint information will still exist. You can use `esp_zb_zcl_reset_all_endpoints_to_factory_default()` to clear it.

It’s still unclear the order of the procedure, what order of calling functions, what signals to expect, how to connect them all with the main process **esp_zb_app_signal_handler()**
The execution of the esp_zb_zdo_device_leave_req() function tries to remove the device, but since it is sleeping and does not respond to network requests, the removal does not occur.

`  
esp_zb_zdo_mgmt_leave_req_param_t leave_req;
esp_zb_ieee_addr_t ieee_addr;
uint16_t short_addr = 0xfdc1;
esp_zb_ieee_address_by_short(short_addr, ieee_addr);
memcpy(leave_req.device_address, zdev_leave->ieee_addr, sizeof(esp_zb_ieee_addr_t));
leave_req.dst_nwk_addr = short_addr;
esp_zb_zdo_device_leave_req(&leave_req, zdevice_leave_cb, NULL);
`
I (115653) ESP_ZIGBEE_API_ZDO: ZDO device leave request timeout!
I (118263) ESP_ZB_GATEWAY: ZDO signal: NLME Status Indication (0x32), status: ESP_OK, (short: 0xeb06), capability: 0x0

xieqinan, 2024-03-19:
Hi @iamalek ,

If the device expected to receive the leave request is a sleep device, its wakeup time should be less than the timeout (5 seconds) of the leave request. Otherwise, the leave request may not receive a response, potentially triggering the log message "ZDO device leave request timeout!". The wakeup time of a sleep device can be set in `ED_KEEP_ALIVE`.

RahulRGaikwad, 2024-06-18:
Hi @xieqinan , 

I have a similar use case for gateway, which needs to remove an end device from the network.

So I'm trying to implement a reset flow.

The reset flow should - 
1. Reset the application data(non-zigbee)
2. Zigbee application layer/cluster data
3. Zigbee stack layer(network details etc.)

What would be the Zigbee way of doing it, or is their esp-zigbee-sdk way of doing it?

xieqinan, 2024-07-26:
@iamalek ,

> It’s still unclear the order of the procedure, what order of calling functions, what signals to expect, how to connect them all with the main process esp_zb_app_signal_handler().

There is no complex process to leave a device from the network. You only need to call `esp_zb_zdo_device_leave_req()`.
If the `esp_zb_zdo_device_leave_req()` is called to leave an end device, the `ESP_ZB_ZDO_SIGNAL_LEAVE_INDICATION` signal will be triggered in the `esp_zb_app_signal_handler()`, indicating that the leave operation has been completed with status on the sender. The related data for the left device will be cleared by the stack. The receiver will receive the `ESP_ZB_ZDO_SIGNAL_LEAVE` signal, indicating it is required to leave the network. The network information will be cleared by the stack based on the parameters of the leave command.

Please note that the timeout for `esp_zb_zdo_device_leave_req()` is triggered after 5 seconds and it does not support modification in the current version.

chshu, 2024-09-29:
@iamalek @RahulRGaikwad Feel free to reopen if you have any further questions.

closed 2024-09-29 by chshu