espressif/esp-zigbee-sdk #284: Stack size to small in example switch_driver

janjaeger, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-2527-ga7d1da94b9

### esp-zigbee-lib version.

1.1.2 @ 32fe28631a1a47d3cd8c7685a12cb765ce9581e5

### esp-zboss-lib version.

1.1.2

### Espressif SoC revision.

ESP32H2

### What is the expected behavior?

abort in printf

### What is the actual behavior?

message being displayed

### Steps to reproduce.

Run a the HA light + switch example, with a debug and trace messages enabled

### More Information.

The below fix solves the problem, however, one could also make a change where one passes the stack size for the button press task on the switch_driver_init() routine.

```
diff --git a/examples/common/switch_driver/src/switch_driver.c b/examples/common/switch_driver/src/switch_driver.c
index 5cb6dcd..d641487 100644
--- a/examples/common/switch_driver/src/switch_driver.c
+++ b/examples/common/switch_driver/src/switch_driver.c
@@ -136,7 +136,7 @@ static bool switch_driver_gpio_init(switch_func_pair_t *button_func_pair, uint8_
         return false;
     }
     /* start gpio task */
-    xTaskCreate(switch_driver_button_detected, "button_detected", 2048, NULL, 10, NULL);
+    xTaskCreate(switch_driver_button_detected, "button_detected", 4096, NULL, 10, NULL);
     /* install gpio isr service */
     gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
     for (int i = 0; i < button_num; ++i) {

```

xieqinan, 2024-03-18:
@janjaeger ，

You are correct. We will increase the default size. However, this is an application logic, and you can also modify it as needed.

janjaeger, 2024-03-18:
Thank you @xieqinan !

closed 2024-04-01 by espressif-bot

xieqinan, 2024-04-01:
Hi,

The stack size of switch driver task has been enlarged to 4096, meeting the requirement. Therefore, the issue will be closed.