espressif/esp-zigbee-sdk #285: Custom cluster Error after last update  V1.2.1

diazmanuel, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.3

### esp-zigbee-lib version.

1.2.1

### esp-zboss-lib version.

1.2.1

### Espressif SoC revision.

ESP32-H2 & ESP32-C6

### What is the expected behavior?

keep it working just like it was doing

### What is the actual behavior?

It stopped working after the last update

### Steps to reproduce.

creates a custom cluster with an ID in the range of 0xFD00 to 0xFFFF

### More Information.

After updating from version 1.2.0 to 1.2.1, everything corresponding to the custom clusters stopped working, the logs indicate that it is a problem with the id out of range but until version 1.2.0 there were no problems, and Reviewing the release notes of the final version (1.2.1) does not seem to indicate any changes in this regard.
I leave below the logs that it shows me and the custom cluster ID used

```
#define CUSTOM_ON_OFF_CLUSTER_ID 0xFDEE
#define CUSTOM_OT_CLUSTER_ID 0xFFFF
#define CUSTOM_BASIC_CLUSTER_ID 0xFDE8
```

```
E (566) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
E (568) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
E (575) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
E (583) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
E (591) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
E (598) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
E (606) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
```
I ask you if you can solve this problem as soon as possible since everything stopped working and I cannot include the new changes in this version

xieqinan, 2024-03-18:
@diazmanuel ，

Could you demonstrate the process of creating a custom cluster? An update for esp-zigbee-sdk v1.2.1 is related to the issue where the attr_list is freed after calling esp_zb_cluster_list_add_xxx_cluster().

diazmanuel, 2024-03-18:
> @diazmanuel ，
> 
> Could you demonstrate the process of creating a custom cluster? An update for esp-zigbee-sdk v1.2.1 is related to the issue where the attr_list is freed after calling esp_zb_cluster_list_add_xxx_cluster().

Hello, yes of course, here is the code 
```
void MyZigbeeHandlertTask(void * parameter){

    esp_zb_ep_list_t *esp_zb_ep_list;
    esp_zb_cluster_list_t *esp_zb_cluster_list;
    esp_zb_attribute_list_t *esp_zb_attribute_list;
    esp_zb_endpoint_config_t esp_zb_endpoint_config;
    esp_zb_endpoint_config.app_profile_id=ESP_ZB_AF_HA_PROFILE_ID;
    esp_zb_endpoint_config.app_device_id=ESP_ZB_HA_CUSTOM_ATTR_DEVICE_ID;
    /* initialize Zigbee stack with Zigbee end-device config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZR_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_ep_list = esp_zb_ep_list_create();

    esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_BASIC_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_COMPANY_NAME, ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,datos_task_zigbee_handler.cluster_data.basic.company_name);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_DEVICE_NAME, ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,datos_task_zigbee_handler.cluster_data.basic.device_name);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_FIRMWARE_VERSION, ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,datos_task_zigbee_handler.cluster_data.basic.firmware_version);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_HARDWARE_VERSION, ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,datos_task_zigbee_handler.cluster_data.basic.hardware_version);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_SERIAL_NUMBER, ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,datos_task_zigbee_handler.cluster_data.basic.serial_number);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_RESET_DEFAULT, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_WRITE_ONLY,&datos_task_zigbee_handler.cluster_data.basic.reset_default);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_RESET, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_WRITE_ONLY,&datos_task_zigbee_handler.cluster_data.basic.reset);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_DEVICE_MAC, ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,datos_task_zigbee_handler.cluster_data.basic.device_mac);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_DEVICE_CLEAR, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_WRITE_ONLY,&datos_task_zigbee_handler.cluster_data.basic.device_clear);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_BASIC_CLUSTER_ATTR_DEVICE_TYPE, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,&datos_task_zigbee_handler.cluster_data.basic.device_type);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);  

    // cluster de ot en ep 1 para eerores
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_OT_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&datos_task_zigbee_handler.cluster_data.boiler.ot_error);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE );
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    // clsuter de on off ep 1 para reportes al boiler
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_ON_OFF_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_ON_OFF_CLUSTER_ATTR_STATE_ID, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&datos_task_zigbee_handler.cluster_data.boiler.state);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_ON_OFF_CLUSTER_ATTR_OFF_DELAY, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,&datos_task_zigbee_handler.cluster_data.boiler.delay_off);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_ON_OFF_CLUSTER_ATTR_ON_DELAY, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,&datos_task_zigbee_handler.cluster_data.boiler.delay_on);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_ON_OFF_CLUSTER_ATTR_INV, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,&datos_task_zigbee_handler.cluster_data.boiler.inv);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_endpoint_config.endpoint=ZIGBEE_BOILER_ENDPOINT;
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, esp_zb_endpoint_config);

    // cluster on off de bomba  ep 2
    esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
[... cut ...]
            report->u.send_info.delta.u32=REPORTABLE_DELTA_ON_OFF;
            if(esp_zb_zcl_update_reporting_info(report)!=ESP_OK) ESP_LOGE(TAG, "REPORTE NO CONFIGURADO");
        }else ESP_LOGE(TAG, "CONFIGURACION DE REPORTE NO ENCONTRADA");
    }

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    esp_zb_set_tx_power(20);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
    
}
```

xieqinan, 2024-03-18:
@diazmanuel ,

Suggestion:
```
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_OT_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&datos_task_zigbee_handler.cluster_data.boiler.ot_error);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE );
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_OT_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&datos_task_zigbee_handler.cluster_data.boiler.ot_error);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    
```

diazmanuel, 2024-03-18:
okay so the problem is when creating both server and client endpoint. will this be fixed or i have to create it as you mention ?
Thanks for the quick response

HOSTEER, 2024-03-20:
> okay so the problem is when creating both server and client endpoint. will this be fixed or i have to create it as you mention ? Thanks for the quick response

@diazmanuel did you mean creating on the same endpoint a cluster's client and a cluster's server ?

diazmanuel, 2024-03-20:
> > okay so the problem is when creating both server and client endpoint. will this be fixed or i have to create it as you mention ? Thanks for the quick response
> 
> @diazmanuel did you mean creating on the same endpoint a cluster's client and a cluster's server ?

The error occurs when trying to add a cluster as a server and client continuously, this could previously be done without problems.Apparently when you use the esp_zb_cluster_list_add_custom_cluster function the cluster list is released so you have to create it again

I leave the code of how it was done before and how it should be done now

Before:
```
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_OT_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&datos_task_zigbee_handler.cluster_data.boiler.ot_error);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE );
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

```
After:
```
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_OT_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&datos_task_zigbee_handler.cluster_data.boiler.ot_error);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE );
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_OT_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID, ESP_ZB_ZCL_ATTR_TYPE_S16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&datos_task_zigbee_handler.cluster_data.boiler.ot_error);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

```

xieqinan, 2024-03-21:
@diazmanuel @HOSTEER 

> Apparently when you use the esp_zb_cluster_list_add_custom_cluster function the cluster list is released so you have to create it again

The current method for creating the cluster list may not be optimal, and it will be improved in the next version.

closed 2024-04-01 by espressif-bot

xieqinan, 2024-04-01:
@diazmanuel @HOSTEER ,

The esp-zigbee-sdk v1.2.2 has added support for reusing attribute, cluster, and endpoint lists. As a result, the issue will be closed. Please don't hesitate to reach out with any further issues or questions regarding the esp-zigbee-sdk.