espressif/esp-zigbee-sdk #286: "attribute not reportable" Error Despite Successful Reporting

acha666, 2024-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### esp-zigbee-lib version.

1.1.2

### esp-zboss-lib version.

1.1.2

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

Attributes, including `scaled_value` and `scale` from the pressure measurement cluster, should be reportable without errors. When reporting these attributes, no errors should occur, and the attributes should be successfully reported to the coordinator (using the ZHA plugin in Home Assistant).

### What is the actual behavior?

Upon attempting to report attributes `scaled_value` and `scale`, the log generates an error stating: "ESP_ZIGBEE_COMMAND: esp_zb_zcl_report_attr_cmd_req(281): This attribute: 0x10 (or 0x14) is not reportable!" Additionally, the `esp_zb_zcl_report_attr_cmd_req` function returns an error code of 0x106. Despite these errors, when attempting to read these attributes from the coordinator (ZHA plugin in Home Assistant), the attributes appear to have been normally reported.

### Steps to reproduce.

1. Initialize the pressure measurement cluster with the provided code snippet.
2. Attempt to report the scaled_value and scale attributes using the specified reporting method.
3. Observe the error in the logs and the error code 0x106 from the esp_zb_zcl_report_attr_cmd_req function.
4. Verify the reporting status by reading the attributes from the coordinator.

### More Information.

Code for initializing the cluster:
``` cpp
    esp_zb_pressure_meas_cluster_cfg_t pressure_meas_cfg = {
        .measured_value = 0xFFFF,
        .min_value = 300,
        .max_value = 1100,
    };
    uint16_t pressure_scaled_value = 0xFFFF;
    uint16_t pressure_min_scaled_value = 300;
    uint16_t pressure_max_scaled_value = 1100;
    uint8_t pressure_scale = 0;
    esp_zb_attribute_list_t *esp_zb_pressure_meas_cluster = esp_zb_pressure_meas_cluster_create(&pressure_meas_cfg);
    ESP_ERROR_CHECK(esp_zb_pressure_meas_cluster_add_attr(esp_zb_pressure_meas_cluster, ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_SCALED_VALUE_ID, &pressure_scaled_value));
    ESP_ERROR_CHECK(esp_zb_pressure_meas_cluster_add_attr(esp_zb_pressure_meas_cluster, ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_MIN_SCALED_VALUE_ID, &pressure_min_scaled_value));
    ESP_ERROR_CHECK(esp_zb_pressure_meas_cluster_add_attr(esp_zb_pressure_meas_cluster, ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_MAX_SACLED_VALUE_ID, &pressure_max_scaled_value));
    ESP_ERROR_CHECK(esp_zb_pressure_meas_cluster_add_attr(esp_zb_pressure_meas_cluster, ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_SCALE_ID, &pressure_scale));
```

Code for reporting attributes:
``` cpp
esp_err_t reportAttribute(uint8_t endpoint, uint16_t clusterID, uint16_t attributeID, void *value, uint8_t value_length)
{
    esp_zb_zcl_status_t status;
    status = esp_zb_zcl_set_attribute_val(endpoint, clusterID, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, attributeID, value, false);

    if (status != ESP_ZB_ZCL_STATUS_SUCCESS)
    {
        ESP_LOGE(TAG, "Setting attribute %04x:%04x failed(0x%02x)!", clusterID, attributeID, status);
        return ESP_FAIL;
    }

    esp_zb_zcl_report_attr_cmd_t cmd = {
        .zcl_basic_cmd = {
            .dst_addr_u.addr_short = 0x0000,
            .dst_endpoint = endpoint,
            .src_endpoint = endpoint,
        },
        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .clusterID = clusterID,
        .attributeID = attributeID,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
    };

    status = esp_zb_zcl_report_attr_cmd_req(&cmd);

    if (status != ESP_ZB_ZCL_STATUS_SUCCESS)
    {
        ESP_LOGE(TAG, "Updating attribute %04x:%04x failed(0x%02x)!", clusterID, attributeID, status);
        return ESP_FAIL;
    }

    return ESP_OK;
}
```

``` cpp
            int16_t pressure_hpa_int = data.pressure; // data.pressure is float in hPa
            int16_t pressure_0_1_hpa_int = data.pressure * 10;
            uint8_t pressure_scale = 1;

            reportAttribute(HA_ESP_LIGHT_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT, ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_VALUE_ID, &pressure_hpa_int, 2); // basic

            reportAttribute(HA_ESP_LIGHT_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT, ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_SCALED_VALUE_ID, &pressure_0_1_hpa_int, 2);
            reportAttribute(HA_ESP_LIGHT_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT, ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_SCALE_ID, &pressure_scale, 1);
```

[link](https://github.com/acha666/esp32-h2-air-sensor/tree/dfb8aff08447ee6d41bc2c86f0fd0752616a2b0a) to the complete project (may contain unrelated content and is currently in an incomplete state, hence the code might be a bit messy)

acha666, 2024-03-16:
After running `idf.py erase-flash`, removing the device from the coordinator, and reconnecting, everything is now working properly, and the error has disappeared. I'm not sure which operation fixed the issue.

xieqinan, 2024-03-18:
@acha666 ，

We have provided a related example for this issue that you can refer to: [temperature_sensor and thermostat](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor).

xieqinan, 2024-03-18:
According to the displayed log, it appears that the attributes expected by reporting do not have the `ESP_ZB_ZCL_ATTR_ACCESS_REPORTING` access.

acha666, 2024-03-20:
@xieqinan ,

You are right.
[image]
Accroading to ZCL doc, these attr only have read access.

Thank you

acha666, 2024-03-20:
btw, may I ask if there is a low-power application example that I can refer to?

closed 2024-03-20 by acha666

chshu, 2024-04-03:
> btw, may I ask if there is a low-power application example that I can refer to?

@acha666 Please refer to the sleep examples: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep

acha666, 2024-04-03:
its usefulthanks