espressif/esp-zigbee-sdk #287: Example esp_zigbee_cli crash at the start

mancausoft, 2024-03-17:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2

### esp-zigbee-lib version.

master

### esp-zboss-lib version.

master

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

work

### What is the actual behavior?

Crash at start for no memory

I (454) main_task: Started on CPU0
I (454) main_task: Calling app_main()
I (464) Zigbee CLI: Command history disabled
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4201831a
file: "/home/xieqinan/ESP/esp-zboss/components/zboss_cli_command/zb_esp_cli_console_task.c" line 102
func: initialize_console
expression: esp_console_init(&console_config)

### Steps to reproduce.

1. build example esp_zigbee_cli with IDF_TARGET=esp32c6
2. flash
3. crash

mancausoft, 2024-03-17:
I also tried with idf 5.2.1 same result

xieqinan, 2024-03-18:
@mancausoft ，

Would you like to use the esp-idf v5.1.3? We have fixed the esp-idf version to v5.1.3 for esp-zigbee-sdk. The esp-zigbee-lib and esp-zboss-lib are also compiled based on v5.1.3.

mancausoft, 2024-03-22:
@xieqinan thanks, works fine with v5.1.3, i wanted to use 5.2.0-5.2.1 because is what i'm using on my project, but for now is good like that :)

xieqinan, 2024-04-11:
@mancausoft ,

If the issue has been resolved, please close it. Please feel free to open other issues for esp-zigbee-sdk.

closed 2024-04-15 by mancausoft