espressif/esp-zigbee-sdk #289: Replacement of Gateway (coordinator) without re-pairing all devices

pieterjanbuntinx, 2024-03-18:
### Question

Hi,

I am currently in the process of developing a Zigbee gateway using an ESP32-C6 and ESP32-S3 with RCP. We would like that in the unlikely event that a gateway would break, we can just replace the gateway and load a config with the correct network details, secur_key, bindings... without having to pair all of the devices again. This would avoid us having to send an installer to the customer to redo all of the pairing.

At this moment I can backup the secur_key, network address and the short address of each individual device. To make this work, I think I would also need to backup what is stored in NVS. Is this something that is possible at this moment? How can I best handle this?

Best Regards,
Pieter-Jan

xieqinan, 2024-04-19:
@pieterjanbuntinx 

> At this moment I can backup the secur_key, network address and the short address of each individual device. To make this work, I think I would also need to backup what is stored in NVS. Is this something that is possible at this moment? How can I best handle this?

I believe the above requirement is not very secure, so the feature has not been implemented. Do you have another way to repair all devices, we can discuss it here?

pieterjanbuntinx, 2024-04-19:
I don't see why this would be a problem security-wise. If for example, there is a vulnerability in my MQTT server, these keys and the binding table could be read out. But more importantly, also the sensor data. This sensor data is much more important than these keys with all of GDPR rules here in Europe. 

With Zigbee2MQTT and TI coordinators (to the same type of chip) it is possible to migrate without repairing [https://www.zigbee2mqtt.io/guide/faq/#how-do-i-migrate-from-one-adapter-to-another](url).
The same also seems possible with Philips Hue bridges [https://readwrite.com/tutorial-upgrade-new-philips-hue-bridge/](url)

xieqinan, 2024-04-26:
@pieterjanbuntinx 

It is possible to use esptool to implement the NVRAM migration feature.

1. Read `zb_storage` from the existing gateway partition:
   ```
   esptool.py -p [PORT] read_flash 0xf1000 16384 zb_storage
   ```

2. Write `zb_storage` to the new gateway partition:
   ```
   esptool.py -p [PORT] write 0xf1000 zb_storage
   ```

3. Flash and restart the new gateway.

xieqinan, 2024-05-30:
@pieterjanbuntinx ，

Do you have some updates for this issue? Whether the above solution  takes effect for your project?
By the way, if the issue has been resolved, please cosider closing it.

pieterjanbuntinx, 2024-05-30:
Hi,
Sorry for the late response. Doing it using esptool is not what I was looking for but I guess I could also do the same using the SPI_flash API. I will test this soon and report back here.

pieterjanbuntinx, 2024-06-06:
It works fine for me with the flash tool and the esp_partition API. Thanks for the help!

Bosemani, 2024-06-13:
> It works fine for me with the flash tool and the esp_partition API. Thanks for the help!

Hai @pieterjanbuntinx 
I'm also trying the same method.
   Can you give a code snippet of how to read and write? It will be useful for everyone.

pieterjanbuntinx, 2024-06-13:
Hi @Bosemani,

Here are some code snippets for reading and writing the zb_storage partition. 

Reading:
``` c
#include <esp_partition.h>

const esp_partition_t* zb_storage_partition         = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_FAT, "zb_storage");

uint8_t data[1024] = {0};
int block = 0;
esp_partition_read(zb_storage_partition, block, data, sizeof(data)
```

Writing:
``` c
#include <esp_partition.h>

const esp_partition_t* zb_storage_partition         = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_FAT, "zb_storage");
uint8_t data[1024] = {0};
int block = 0;

esp_partition_write(zb_storage_partition, block , data, sizeof(data));
```

In my code, I split up the flash partition in blocks of 1024 bytes and transfer them base64 encoded over mqtt. I added a special boot mode wherein all the Zigbee initialization is not done so nothing would be read or written to zb_storage from code while doing the transfer.

Bosemani, 2024-06-13:
Hai @pieterjanbuntinx 

Thank you for quick response!