espressif/esp-zigbee-sdk #290: How to connect to multiple devices with multiple endpoints

pepijndevos, 2024-03-20:
### Question

I had some fun with the on-off switch example, which worked with my smart socket unchanged!
Except it switches all 5 outputs at the same time, and I can't quite figure out how to switch them individually.
Maybe I'm being dense, I kinda caught a virus, so apologies for asking dumb questions, maybe the answers will reveal themselves with a fresh mind.

I see `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE` which I guess is when a device announces its presence and then I see `esp_zb_zdo_find_on_off_light` which calls the callback that gets endpoint 1 only which for some reason switches all the outputs.

What does that code need to do to detect all the endpoints on the device?
This appear to be all the endpoints it has: https://github.com/zigpy/zha-device-handlers/blob/2a41331d7aec3b5c309de0da4dd928863e418290/zhaquirks/tuya/ts011f_plug.py#L416

The thermostat example also seems interesting, which seems to have a whole lot of stuff going on creating clusters and what not. I feel like maybe I'm just missing some context on Zigbee concepts and the PDK architecture, any recommended reading?

Both examples seem very hardcoded on one specific type of device with specific endpoints and clusters. Eventually I want to connect both the socket and a temperature+humidity sensor. Preferably without too much hardcoding of the specific device, so that it could also work with devices from a different manufacturer. Is that possible?

djorr5, 2024-03-20:
See if the example code in here helps.

https://github.com/espressif/esp-zigbee-sdk/issues/246#issuecomment-1996839407

pepijndevos, 2024-04-05:
I'm looking more into the thermostat example, trying to understand how to add multiple devices.

My impression is I need another function like `find_temperature_sensor` that would match each of the devices I want to find and pair? That would then hopefully run the callback for the matching device.

https://github.com/espressif/esp-zigbee-sdk/blob/c86f305efc693cef958705ad55b2092bcdde16d5/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c#L266

But I'm a bit puzzled about the details of binding that go on in `user_find_cb` and `bind_cb`.  I guess I need to make copies of those that bind different kind of devices?

I'm also a bit puzzled about the endpoints on the coordinator. Do I need to add another device that has endpoints corresponding to those for each device I want to bind?

https://github.com/espressif/esp-zigbee-sdk/blob/c86f305efc693cef958705ad55b2092bcdde16d5/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c#L448

I guess maybe what I'm missing is a good overview of the Zigbee protocol and how the SDK API relates to it. If I look for Zigbee explanations I get really high level enthusiast articles, and if I go down to the specification it goes into the wire protocol that is not relevant to me either. What I need is kind of a description of the roles and concepts and an overview of the transactions that happen. Kind of like all those Syn-ack diagrams of TCP.

For example one big question I have is how to persist the device pairings. I suspect with a reasonable understanding of the protocol that would be obvious, like, just store the MAC address of the sensor or whatever the Zigbee equivalent is.

xieqinan, 2024-04-15:
@pepijndevos,

> My impression is I need another function like `find_temperature_sensor` that would match each of the devices I want to find and pair? That would then hopefully run the callback for the matching device.

I think the `esp_zb_zdo_active_ep_req()` API is helpful for you to get the all endpoints in a device.

> I guess maybe what I'm missing is a good overview of the Zigbee protocol and how the SDK API relates to it. If I look for Zigbee explanations I get really high level enthusiast articles, and if I go [...]

We are enhancing our documents. However, we can discuss the specific requirements you need.

pepijndevos, 2024-04-21:
I don't fully understand the meaning of this code. Why does it create both client and server clusters?
Is the client endpoint sort of a local representation of the remote endpoint we want to bind?

If that is so, why is there a humidity measurement cluster, but no hygrometer cluster?
It's unclear to me what I need to do to bind the humidity cluster that my temperature sensor has.

```
    ESP_ERROR_CHECK(esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, MANUFACTURER_NAME));
    ESP_ERROR_CHECK(esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, MODEL_IDENTIFIER));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(cluster_list, basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create(&(thermostat->identify_cfg)), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY), ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_thermostat_cluster(cluster_list, esp_zb_thermostat_cluster_create(&(thermostat->thermostat_cfg)), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    /* Add temperature measurement cluster for attribute reporting */
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_temperature_meas_cluster(cluster_list, esp_zb_temperature_meas_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE));
```

chshu, 2024-07-26:
@pepijndevos please reopen if any follow up questions.

closed 2024-07-26 by chshu

pepijndevos, 2024-07-28:
Thank you. For now we paused this effort. I do have a dream of using this PDK to implement Zigbee support in ESPHome. So you could automatically expose sensors to Zigbee much like the existing mqtt support. So that would require a data-driven approach to setting up endpoints and such based on a yaml configuration. This is not something I'm actively pursuing, but that I believe would be of great value to the ecosystem.