espressif/esp-zigbee-sdk #291: unable to communicate with coordinator after

diazmanuel, 2024-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.3

### esp-zigbee-lib version.

1.2.1

### esp-zboss-lib version.

1.2.1

### Espressif SoC revision.

ESP32-H2 & ESP32-C6

### What is the expected behavior?

The expected behavior is to let the device in as usual

### What is the actual behavior?

I'm having trouble contacting the network coordinator.
I am currently testing a system which has 1 coordinator with the ESP32-C6, 2 repeaters with ESP32-C6 and 6 end-devices with the ESP32-H2.
In addition, the end-devices operate in low consumption with light sleep.
At some point during the testing (which we left for about 2 weeks) one of the devices lost the ability to connect to the coordinator, although it was still on the network because it was connecting to the network from a repeater. To try to locate the problem, I tried to restart the device and it connected to the zigbee network (since it was connected through the repeaters) but when performing the ieee_req and trying to reach the coordinator it fails repeatedly. In the second instance we disconnect all the repeaters and restart the end-device, in this case it directly failed when trying to enter the network, that is, for some reason the coordinator does not allow it to communicate with it, in the third instance we restart the coordinator and the end-device , and finally it started working correctly

### Steps to reproduce.

try to replicate the ecosystem and leave it for a long time

### More Information.

What should I do when this error occurs, because it is very difficult to identify the error and replicate it, because the ieee_request fails, if upon restarting the coordinator it manages to get in again, that is, it saved the keys and everything necessary to connect, clearly the The problem is in the coordinator. Did you find any similar problem?
Is there a way to reset just the zigbee part without restarting the entire chip? Any suggestions on how to solve this problem? I think it's something important to solve and I don't know if I'm doing something wrong.

xieqinan, 2024-04-01:
@diazmanuel ,

We haven't encountered similar problems before, but this issue is significant.

Could you please provide details on how you made the end device connect to the network from a repeater? Additionally, could you describe the network topology?

>  I tried to restart the device and it connected to the zigbee network (since it was connected through the repeaters) but when performing the ieee_req and trying to reach the coordinator it fails repeatedly.

If you perform the `ieee_req()` to obtain the repeater which serves as the parent of the endpoint, does it yield any results?

> In the second instance we disconnect all the repeaters and restart the end-device, in this case it directly failed when trying to enter the network, that is, for some reason the coordinator does not allow it to communicate with it

If the end device had already joined this network, it should attach to other repeaters within this network. Therefore, the behavior described above is not normal. I suspect that the end device may have lost the network information or has not updated the new link key from the coordinator. I will attempt to reproduce this issue; however, it would be helpful to have the Wireshark packet of this process.

diazmanuel, 2024-04-04:
hello @xieqinan, 

> Could you please provide details on how you made the end device connect to the network from a repeater? Additionally, could you describe the network topology?

I estimate that the end device connects to the repeater through proximity since it was located next to the repeater device. In any case, if the gateway is disconnected, the device does not lose the connection as explained above, this is because it was connected to the repeater.

The topology is as I mentioned before. I am currently testing a system which has 1 coordinator with the ESP32-C6, 2 repeaters with ESP32-C6 and 6 end-devices with the ESP32-H2. I don't know if you need any more information.
 
System works this way

1.  The gateway interprets commands sent to it by mqtt and modifies a parameter of one of the end devices via zigbee write command
2. The end device reports a change to the repeater . the change can be a write command that comes from the gateway or a change that is made by pressing a physical button on that end device)
3. The repeater takes the report of that change and activates a relay

So what failed was that the changes that were sent from the gateway never reached the end device, that is, the writing failed. But on the contrary, the changes that were made with the physical button of the end device did generate a report that successfully reached the repeater and activated its relay. On the other hand, this failure only occurred in 1 of the 6 end devices, the other 5 worked correctly, and the error occurred after many hours of testing.In short, the network itself continued to function. The problem is that the gateway denies communication.

> If you perform the ieee_req() to obtain the repeater which serves as the parent of the endpoint, does it yield any results?

I really don't know because the ieee_req() is only sent when connecting to the zigbee network, I give an excerpt of the code where the ieee_req() function is called
```
static void ieee_cb(esp_zb_zdp_status_t zdo_status, esp_zb_ieee_addr_t ieee_addr, void *user_ctx)
{
    
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {

        ESP_LOGI(TAG, "ieee_cb address is: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x",
                 ieee_addr[7], ieee_addr[6], ieee_addr[5], ieee_addr[4],
                 ieee_addr[3], ieee_addr[2], ieee_addr[1], ieee_addr[0]);

        esp_zb_zdo_bind_req_param_t bind_req;
        esp_zb_get_long_address(bind_req.src_address);

        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        memcpy(bind_req.dst_address_u.addr_long, ieee_addr, sizeof(esp_zb_ieee_addr_t));
        
        ESP_LOGI(TAG, "local address is: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x",bind_req.src_address[7], bind_req.src_address[6], bind_req.src_address[5],bind_req.src_address[4], bind_req.src_address[3], bind_req.src_address[2],bind_req.src_address[1], bind_req.src_address[0]);
        ESP_LOGI(TAG,"local short address is: 0x%04x",esp_zb_get_short_address());

        //binds al hub
        bind_req.dst_endp = ZIGBEE_HUB_ENDPOINT;
        bind_req.req_dst_addr = esp_zb_get_short_address();

        bind_req.src_endp = ZIGBEE_BOILER_ENDPOINT;
        bind_req.cluster_id = CUSTOM_ON_OFF_CLUSTER_ID;
        esp_zb_zdo_device_bind_req(&bind_req, bind_cb,0);

    }else{
        ESP_LOGW(TAG,"unable to communicate with coordinator");
        set_ieee_addr_req_alarm(5000);     
    }
}
static void bdb_start_ieee_addr_req_cb(uint8_t param)
{
    esp_zb_zdo_ieee_addr_req_param_t ieee_req;
    ieee_req.addr_of_interest = 0;
    ieee_req.dst_nwk_addr = 0;
    ieee_req.request_type = 0;
    ieee_req.start_index = 0;
    esp_zb_zdo_ieee_addr_req(&ieee_req, ieee_cb, NULL);
}

static void set_ieee_addr_req_alarm(uint32_t time_ms){
    esp_zb_scheduler_alarm_cancel((esp_zb_callback_t)bdb_start_ieee_addr_req_cb, 0);
    esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_ieee_addr_req_cb, 0, time_ms);
}
static void set_steering_alarm(uint32_t time_ms){
    esp_zb_scheduler_alarm_cancel((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING);
    esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, time_ms);
}
static void set_zigbee_init_alarm(uint32_t time_ms){
[... 64 lines cut ...]
        break;
    }
}
```
Maybe there is something wrong but what I end up getting is the "unable to communicate with coordinator"  message continuously because it is a kind of recursive function. Anyway, the problem is that the hub does not respond to the ieee_req()

I understand that this always makes the request to the gateway/coordinator
```
static void bdb_start_ieee_addr_req_cb(uint8_t param)
{
    esp_zb_zdo_ieee_addr_req_param_t ieee_req;
    ieee_req.addr_of_interest = ZIGBEE_HUB_SHORT_ADDRESS;
    ieee_req.dst_nwk_addr = ZIGBEE_HUB_SHORT_ADDRESS;
    ieee_req.request_type = 0;
    ieee_req.start_index = 0;
    esp_zb_zdo_ieee_addr_req(&ieee_req, ieee_cb, NULL);
}
```
so I don't know what the router would respond when surveyed.

> If the end device had already joined this network, it should attach to other repeaters within this network. Therefore, the behavior described above is not normal. I suspect that the end device may [...]

I really don't understand well how Zigbee access keys work, what I can tell you is that when restarting the coordinator the device was able to re-enter the network and communicate with the coordinator without problems, t [... cut ...] of the network kept it stored, in the case of the key I no longer have knowledge to ensure whether or not it is the cause. If I understand correctly, it is a key that would have an expiration date and that from time to time has to be updated and that for some reason this end device could not update its key, but I don't know.

Regarding the wireshark package, unfortunately I do not have the possibility of capturing it at this time, especially because of how difficult it is to replicate this case.

Suxsem, 2024-04-06:
I'm experiencing the same behavior, after some weeks the device loses the connectivity to the coordinator and is not able to reconnect to it without restarting the coordinator

xieqinan, 2024-04-19:
@diazmanuel ,

We attempted to reproduce the phenomenon based on your description, but the test has run normally for about 1 week. Is it common for the issue to occur frequently? We may need more detailed information to reproduce it accurately.

In my environment, the end devices report messages to the coordinator every 30 seconds, and then the coordinator reads the attribute from the end device that reported the message. Additionally, we press a button to trigger an `ieee_req()` from the end device to the coordinator every 4 hours. What might I be missing or overlooking?

diazmanuel, 2024-04-23:
hello @xieqinan, 
it is not common for it to happen, and I think the biggest difference I find with your testing is that you are not using devices as a repeater, in fact I think the problem appears when repeaters are used. Try inserting the repeaters and making the end device connect to the network through the repeater. In this condition is when the commands from the coordinator stop reaching the end device and when the ieee req from the end device to the coordinator fails

xieqinan, 2024-04-25:
@diazmanuel ，

The network topology for the test is as follows, and it should be the same as yours.
```
Coordinator(C6)|----Router1(C6)---- |end_device 1(H2)
               |                    |end_device 2(H2)
               |                    |end_device 3(H2)
               |     
               |----Router2(C6)--- |end_device 4(H2)
               |                   |end_device 5(H2)
               |                   |end_device 6(H2)
```
Would you mind sharing a simplified reproducing code? It would be helpful for us to resolve the issue.

xieqinan, 2024-05-30:
@diazmanuel ，

> On the other hand, I would like to take this opportunity to ask you if the following chain of actions is correct or would it be saturating the coordinator too much. This chain repeats every time the device restarts (due to lack of power) or loss of network.

I have reviewed the code you provided. I believe this is normal device behavior and should not cause payload pressure on the coordinator.
> I wonder if it is necessary to do the iee_req() to the coordinator (since in the examples they do it that way) 

The `ieee_req()` is not necessary; it is only used to obtain the long address for the binding operation.

> if it is a problem to make the bindings to the coordinator every time the network connection is reestablished

It is not a problem; the device will only store a single binding relationship for the same device.

closed 2024-06-05 by diazmanuel