espressif/esp-zigbee-sdk #292: Unable to pair device with homeassistant ZHA for deep sleep sketch

tilman, 2024-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-2032-g4d90eedb6e

### esp-zigbee-lib version.

1.2.1

### esp-zboss-lib version.

1.2.1

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

The device should pair with my Homeassistant ZHA via a Conbee Stick 2

### What is the actual behavior?

The device is not showing up while I search for new devices.

### Steps to reproduce.

1. flash light sleep sketch
2. search for devices
3. `Network steering was not successful (status: ESP_FAIL)` keeps being shown

To verify that hardware is working i also flashed `HA_color_dimmable_switch` example
1. flash file
2. search for devices
3. device is found and shown in Homeassistant with light property. 
4. turning light on and off via Homeassistant works

Hedda, 2024-03-28:
@tilman slightly off-topic but if using Home Assistant then you might be interested in this ESPHome project that is using Espressif's esp-zigbee-sdk to  -> https://github.com/luar123/esphome_zb_sensor

smg-tilman, 2024-03-28:
Thanks a lot! Thats a good tip of a project. I'm already a huge esphome fan and the combinations of these two project is great!

Hedda, 2024-03-28:
> Thats a good tip of a project. I'm already a huge esphome fan and the combinations of these two project is great!

Then also check out the very related feature request for native Zigbee (and Thread) support in ESPHome firmware here:

* https://github.com/esphome/feature-requests/issues/1397

lhespress, 2024-05-13:
@tilman Could you check it again? i checked it with Conbee II as Coordinator:

### IDF version
tag v5.2.1
### esp-zigbee-lib version.
1.3.1
### esp-zboss-lib version.
1.3.1
### Espressif SoC revision.
ESP32-H2

it works fine, the LOG as follows: 

```
I (23) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (24) boot: compile time May 13 2024 14:40:42
I (25) boot: chip revision: v0.1
I (27) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (36) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 000e1000
I (79) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=42060020 size=12008h ( 73736) map
I (130) esp_image: segment 1: paddr=00022030 vaddr=40800000 size=05fe8h ( 24552) load
I (139) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=5f9b8h (391608) map
I (259) esp_image: segment 3: paddr=000879e0 vaddr=40805fe8 size=09170h ( 37232) load
I (273) esp_image: segment 4: paddr=00090b58 vaddr=4080f160 size=018a4h (  6308) load
I (276) esp_image: segment 5: paddr=00092404 vaddr=50000000 size=00058h (    88) load
I (283) boot: Loaded app from partition at offset 0x10000
I (285) boot: Disabling RNG early entropy source...
I (302) cpu_start: Unicore app
W (311) clk: esp_perip_clk_init() has not been implemented yet
I (317) cpu_start: Pro cpu start user code
I (318) cpu_start: cpu freq: 96000000 Hz
I (318) cpu_start: Application information:
I (320) cpu_start: Project name:     sleepy_end_device
I (326) cpu_start: App version:      658a05c-dirty
I (332) cpu_start: Compile time:     May 13 2024 14:41:05
I (338) cpu_start: ELF file SHA256:  990338b2c...
I (343) cpu_start: ESP-IDF:          v5.2.1
I (348) cpu_start: Min chip rev:     v0.0
I (353) cpu_start: Max chip rev:     v0.99 
I (358) cpu_start: Chip rev:         v0.1
I (362) heap_init: Initializing. RAM available for dynamic allocation:
I (369) heap_init: At 40815250 len 00038130 (224 KiB): RAM
I (376) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (383) spi_flash: detected chip: generic
I (386) spi_flash: flash io: dio
W (390) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (404) sleep: Enable automatic switching of GPIO sleep configuration
I (411) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (419) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (427) sleep_sys_periph: Interrupt Matrix sleep retention initialization
I (435) sleep_sys_periph: HP System sleep retention initialization
I (442) sleep_sys_periph: TEE/APM sleep retention initialization
I (448) sleep_sys_periph: UART sleep retention initialization
I (455) sleep_sys_periph: Timer Group sleep retention initialization
I (462) sleep_sys_periph: IO Matrix sleep retention initialization
I (469) sleep_sys_periph: SPI Mem sleep retention initialization
I (475) sleep_sys_periph: SysTimer sleep retention initialization
I (493) main_task: Started on CPU0
I (493) main_task: Calling app_main()
I (503) pm: Frequency switching config: CPU_MAX: 96, APB_MAX: 96, APB_MIN: 96, Light sleep: ENABLED
[... 58 lines cut ...]
I (43768) ESP_ZB_SLEEP: Zigbee can sleep
I (43983) ESP_ZB_SLEEP: Zigbee can sleep
I (47809) ESP_ZB_SLEEP: Zigbee can sleep
I (47953) ESP_ZB_SLEEP: Zigbee can sleep
I (48069) ESP_ZB_SLEEP: Zigbee can sleep
I (48265) ESP_ZB_SLEEP: Zigbee can sleep
I (48408) ESP_ZB_SLEEP: Zigbee can sleep
I (52307) ESP_ZB_SLEEP: Zigbee can sleep
I (52426) ESP_ZB_SLEEP: Zigbee can sleep
I (52533) ESP_ZB_SLEEP: Zigbee can sleep
I (52766) ESP_ZB_SLEEP: Zigbee can sleep
I (56576) ESP_ZB_SLEEP: Zigbee can sleep
I (56717) ESP_ZB_SLEEP: Zigbee can sleep
I (56843) ESP_ZB_SLEEP: Zigbee can sleep
I (57000) ESP_ZB_SLEEP: Zigbee can sleep
I (57184) ESP_ZB_SLEEP: Zigbee can sleep
I (61047) ESP_ZB_SLEEP: Received message: endpoint(10), cluster(0x6), attribute(0x0), data size(1)
I (61098) ESP_ZB_SLEEP: Light sets to On
I (61115) ESP_ZB_SLEEP: Zigbee can sleep
I (61250) ESP_ZB_SLEEP: Zigbee can sleep
I (61365) ESP_ZB_SLEEP: Received message: endpoint(10), cluster(0x6), attribute(0x0), data size(1)
I (61416) ESP_ZB_SLEEP: Light sets to Off
I (61426) ESP_ZB_SLEEP: Zigbee can sleep

```

chshu, 2024-07-26:
@tilman please check https://github.com/espressif/esp-zigbee-sdk/issues/292#issuecomment-2106813913, and reopen if any follow up questions.

closed 2024-07-26 by chshu