espressif/esp-zigbee-sdk #293: Assert in zb_bufpool_mult.c:1022

nomis, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-2320-ge4f167df25

### esp-zigbee-lib version.

1.2.0

### esp-zboss-lib version.

1.2.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Does not crash

### What is the actual behavior?

Assert in zb_bufpool_mult.c:1022

### Steps to reproduce.

Unknown

### More Information.

Source: https://github.com/nomis/candle-dribbler/tree/0.7.1
Binary: [candle-dribbler.elf.gz](https://github.com/espressif/esp-zigbee-sdk/files/14711720/candle-dribbler.elf.gz)
Core dump: 
[core-dump-2024-03-20.txt](https://github.com/espressif/esp-zigbee-sdk/files/14711744/core-dump-2024-03-20.txt)

It looks like this is in `zb_buf_initial_alloc_func()`?

nomis, 2024-03-23:
It looks like it's trying to allocate more than `ZB_IO_BUF_SIZE` (152), but there's no indication of which function called this one in the core dump.
```
4202cb46:       0aa97163                bgeu    s2,a0,4202cbe8 <zb_buf_initial_alloc_func+0xd2>
```
```
s2             0xb3	179
```

nomis, 2024-03-27:
This has happened again in the same place, with version 1.2.1

Source: https://github.com/nomis/candle-dribbler/tree/0.7.2
Binary: [candle-dribbler.elf.gz](https://github.com/espressif/esp-zigbee-sdk/files/14770010/candle-dribbler.elf.gz)
Core dump: [core-dump-2024-03-27.txt](https://github.com/espressif/esp-zigbee-sdk/files/14770015/core-dump-2024-03-27.txt)

```
4202dde6: zb_init_common.c.obj:?
42031eaa: zb_scheduler.c.obj:?
42032b16: zb_scheduler.c.obj:?
```

Again, it looks like it's trying to allocate too much:
```
#3  0x4202dde6 in ?? ()
(gdb) info registers 
s2             0xbe	190
```

xieqinan, 2024-04-02:
@nomis ,

The `zb_buf_initial_alloc_func()` function is used to allocate memory space from a buffer that has been previously allocated from the buffer pool. This indicates that the size of the memory buffer is fixed, either 152 bytes or a multiple of 152 bytes. Therefore, both sending a large packet from the application and receiving a large packet from a remote device can potentially trigger the crashing issue.

After reviewing the source code of `zb_init_common.c`, it seems that the function `zb_buf_initial_alloc_func()` is not called.

Additionally, could you please provide a new `core dump` log? The previous log appears to contain random code.

nomis, 2024-04-02:
> sending a large packet from the application

I'm not sending anything specifically, this is just running `esp_zb_main_loop_iteration()`.

>  receiving a large packet from a remote device can potentially trigger the crashing issue.

This must never happen and indicates a bug in the zboss library.

> After reviewing the source code of `zb_init_common.c`, it seems that the function `zb_buf_initial_alloc_func()` is not called.

I've already reported that the core dumps don't contain any useful information (#157) so you'll need to fix that problem first? None of the other tasks have problems showing a full stack trace.
 
> Additionally, could you please provide a new `core dump` log? The previous log appears to contain random code.

Please explain what "random code" you think is in my core dump.

nomis, 2024-04-02:
It has happened again with a different line number: #304

lpy4105, 2024-04-22:
This issue related to https://github.com/espressif/esp-zigbee-sdk/issues/304, I think we could close this one.

lpy4105, 2024-04-22:
> That issue is with the `zb_get_buf_tail_ptr` function and this issue is with the `zb_buf_initial_alloc_func` function.

Surely. And both asserts happened because we received a frame with unexpected length.

chshu, 2024-04-28:
@nomis This issue shares the same root cause with https://github.com/espressif/esp-zigbee-sdk/issues/304, that Mac layer received an 802.15.4 frame which is longer than 127 bytes (https://github.com/espressif/esp-zigbee-sdk/issues/304#issuecomment-2067705304), but it's unexpect to happen in our hardware. We assume the issue is casued by an unexpected overwritten, but not sure whether it's caused by the applicaiton code or the SDK itself.

Please help to capture the log following this comment: https://github.com/espressif/esp-zigbee-sdk/issues/304#issuecomment-2071877536.

I'm closing the issue, and let's follow up in https://github.com/espressif/esp-zigbee-sdk/issues/304.

closed 2024-04-28 by chshu