espressif/esp-zigbee-sdk #294: Add manufacturer specific code in custom cluster/attribute

grebenyuk51, 2024-03-24:
### Is your feature request related to a problem?

ZCL specification requires manufacturer specific code to be transmitted with manufacturer specific extensions (clusters/attributes/commands)

[image]

SDK provides possibility to add custom cluster and custom attribute. But there is no possibility to add manufacturer specific code

### Describe the solution you'd like.

add additional parameter to `esp_zb_cfg_s `or `esp_zb_endpoint_config_t`

### Describe alternatives you've considered.

add additional parameter uint16_t to function `esp_zb_cluster_list_add_custom_cluster`

### Additional context.

also if device receives command without manuf spec code ignore it

xieqinan, 2024-03-25:
Hi @grebenyuk51 ,

We will take your suggestion into consideration. Before that, there is a workaround for your issue. You can call `esp_zb_zcl_get_attribute()` to get the pointer of `esp_zb_zcl_attr_t` after `esp_zb_device_register()`, then modify the `manuf_code` directly to achieve the goal.

grebenyuk51, 2024-03-25:
oh, that is a good point! it`s better than iterate all attributes. thanks

xieqinan, 2024-04-15:
@grebenyuk51 ，

We do not plan to make a breaking change for `esp_zb_custom_cluster_add_custom_attr()` in the near future. If the workaround is helpful for you, please consider closing the issue. We welcome discussions on any issues related to the esp-zigbee-sdk.

grebenyuk51, 2024-04-16:
ok, looking forward for changes but solution in this [massage](https://github.com/espressif/esp-zigbee-sdk/issues/294#issuecomment-2017122492) do a job

closed 2024-04-16 by grebenyuk51