espressif/esp-zigbee-sdk #295: Odd behavior with multiple on/off endpoints

john159753, 2024-03-25:
### Question

I am new to this SDK, though I have worked with a XBee before. I'm sure this is a problem my code, but its such a simple change off the main example i don't know how i could have introduced this behavior. I just added two endpoints off the "HA_On_Off_Light", and in the attribute handler/callback I added a second conditional, to handle the second endpoint.

The specific issue has to to with two on/off endpoints. I can turn on and then off the 1st endpoint, and then turn on and off the second endpoint. But if i try to turn on endpoint 1 THEN endpoint 2, endpoint two never turns on.  Eventually clicking off then on (on the misbehaving endpoint) gets both LEDs on. The same behavior happens in reverse, i cant turns off in sequence either. The coordinator doesn't report any timeout or errors (which is what would happen if the ESP wasnt online), so it seems like the command is received by the ESP, but i have added println's at the very start of "esp_zb_core_action_handler_register" and it doesn't seem like the callback gets fired off at all.

I'm at a loss here, any help would be appreciated

### Additional context.

My coordinator is zigbee2mqtt, I haven't ruled out the issue isn't on that side yet either..

I am using the Arduino IDE with the ESP32C6 board selected

john159753, 2024-03-27:
I'm continuing to throw the kitchen sink at this, trying different examples from the web on how it should be setup. I've tried to clean up my code to help others see what I'm doing. My latest discovery that prompted me to update this, is that if i use the "local control/setting a pin to ground, it also causes the issue.
#### Working scenario - all commands from zigbee2mqtt
1. Turn Endpoint 1 **on** (works)
2. Turn Endpoint 1 **off** (works)
1. Turn Endpoint 2 **on** (works)
2. Turn Endpoint 1 **off** (works)
#### Broken scenario - all commands from zigbee2mqtt
1. Turn Endpoint 1 **on** (works)
2. Turn Endpoint 2 **on** (doesnt work)
3. Turn Endpoint 2 **off** (no change to report)
2. Turn Endpoint 2 **on** (works)
#### New broken scenario, with initial ON being triggered locally 
1. Turn Endpoint 1 **on** with LOCAL control via GPIO to ground (works)
2. Turn Endpoint 2 **on** (doesnt work)
3. Turn Endpoint 2 **off** (no change to report)
2. Turn Endpoint 2 **on** (works)

I'm starting to think this is a bug somewhere. Like i said I'm using the Arduino IDE with the 3.0 Alpha3 version of the boards (which is  using ESP-IDF v5.1)

```c
static void esp_zb_task(void *pvParameters)
{  /* initialize Zigbee stack with Zigbee end-device config */
  esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
  esp_zb_init(&zb_nwk_cfg);
  esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

  /* basic cluster attribute list */

  esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_basic_cluster_create(NULL);
  uint8_t powerSource = ZCL_BASIC_POWER_SOURCE;
  esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &powerSource);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
  /* identify cluster attribute list */
  esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_identify_cluster_create(NULL);
  /* group cluster attribute list */
  esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_groups_cluster_create(NULL);
  /* scenes cluster attribute list */
  esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(NULL);
  /* on-off cluster attribute list */
  esp_zb_on_off_cluster_cfg_t on_off_cfg;
  on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
  esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);

  //Add all the clusters to esp_zb_cluster_list
  esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
  esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
  esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
  esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
  esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
  esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

  esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
  esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINTRED, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
  esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINTGREEN, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);

  //add the endpoint list
  esp_zb_device_register(esp_zb_ep_list);
  
  esp_zb_core_action_handler_register(zb_action_handler);
  ESP_ERROR_CHECK(esp_zb_start(false));

  esp_zb_main_loop_iteration();
}
```
And my task i run for detections on two inputs (local control) - when i bring these pins to ground the leds get toggled and that state gets set.
```c

static void test_task(void *pvParameters) {

for (;;) {
redLastButtonState    = redCurrentButtonState;      // save the last state
  redCurrentButtonState = digitalRead(REDSW); // read new state
 if(redLastButtonState == HIGH && redCurrentButtonState == LOW) {
    
 //   toggle state of LED
   redLedState = !redLedState;
 //   control LED arccoding to the toggled state
   digitalWrite(REDLED, redLedState); 
   esp_zb_zcl_set_attribute_val(HA_ESP_LIGHT_ENDPOINTRED, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID, &redLedState,false);
 }
  greenLastButtonState    = greenCurrentButtonState;      // save the last state
  greenCurrentButtonState = digitalRead(GREENSW); // read new state
  if(greenLastButtonState == HIGH && greenCurrentButtonState == LOW) {
    
    // toggle state of LED
    greenLedState = !greenLedState;
    // control LED arccoding to the toggled state
    digitalWrite(GREENLED, greenLedState); 
    esp_zb_zcl_set_attribute_val(HA_ESP_LIGHT_ENDPOINTGREEN, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID, &greenLedState,false);
  }
  vTaskDelay(100 / portTICK_PERIOD_MS);
}
}
I don't know if these are going to embed correctly, but one example of me triggering all from zigbee2mqtt, and another example of me turning on endpoint 1 locally

https://github.com/espressif/esp-zigbee-sdk/assets/12056448/88d20bb6-ba31-46b1-8819-f269572b1723

https://github.com/espressif/esp-zigbee-sdk/assets/12056448/a2ddfce1-abeb-4323-a613-cab9745cfe04

xieqinan, 2024-03-27:
Hello @john159753,

I attempted to reproduce the issue based on your description using a local button to simulate the process. I utilized the esp-zigbee-sdk v1.2.0 and esp-idf v5.1.3. I think that it produces the same result as `zigbee2mqtt` if the command is the same. However, everything seems to be working fine. I've attached the code here for your reference.

Upon reviewing your code, I believe that the endpoint creation is correct. However, it seems that the version of the library and ESP-IDF is outdated.
 
By the way, the esp-zigbee-lib and esp-zboss-lib are compiled on the esp-idf v5.1.3. To avoid any unexpected behavior, please ensure that you are using esp-idf v5.1.3 as your version.

[HA_on_off_example.zip](https://github.com/espressif/esp-zigbee-sdk/files/14767602/HA_on_off_example.zip)

xieqinan, 2024-03-27:
Hello @john159753 
> ```c
>  esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINTRED, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
>   esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINTGREEN, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
> ```

It should not reuse the `esp_zb_cluster_list`, which is the root cause of this issue.

john159753, 2024-03-27:
You are totally right -- I think it might be smart for me to drop the Arduino IDE and switch to use esp-idf with VScode or something, so i can run the latest versions (it looks like your example was using a struct esp_zb_endpoint_config_s that didn't exist in the dev esp32 pack from December in Arduino).

By totally splitting out the two endpoints to two totally different attribute lists and cluster lists, they were no longer stepping on each other's toes in the background.

Thanks so much for your help on this!

closed 2024-03-27 by john159753