espressif/esp-zigbee-sdk #296: Transmit JSON data

superbagus1712, 2024-03-26:
### Question

Can i transmit json data via zigbee within 100miliseconds delay ? i have tried but failed esp32c6 keep restarting

xieqinan, 2024-03-26:
@superbagus1712,

The esp-zigbee-sdk supports transmitting APS data (https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/aps/esp_zigbee_aps.h#L118), which can help with your issue. However, you need to adjust the transmission time according to the data size. If the data size is less than 1024, a 500 ms delay is acceptable.

superbagus1712, 2024-03-27:
@xieqinan do you have an example to use APS data ? sorry im actually little bit confused with zigbee api

xieqinan, 2024-03-28:
@superbagus1712,

The ESP-Zigbee-SDK does not provide an example for APS, but I can provide guidance on its usage.

The sender:
```
static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        esp_zb_apsde_data_req_t req;
        // Data
        for (int i = 0; i < sizeof(s_value); i++) {
            s_value[i] = i;
        }
        req.dst_addr_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        // The basic information of destination
        req.dst_short_addr = 0xd47d;       
        req.dst_endpoint = 10;
        req.profile_id = 0x104;
        req.cluster_id = 0xFFF1;
        req.src_endpoint = 1;
        // Payload
        req.asdu_length = sizeof(s_value);
        req.asdu = s_value;
        // routing hops
        req.radius = 2;
        req.tx_options = (ESP_ZB_APSDE_TX_OPT_ACK_TX | ESP_ZB_APSDE_TX_OPT_FRAG_PERMITTED);
        req.use_alias = false;
        ESP_EARLY_LOGI(TAG, "Send aps data request");
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_aps_data_request(&req);
        esp_zb_lock_release();
    }
}
``` 

The receiver:
```
static bool zb_apsde_data_indication_handler(esp_zb_apsde_data_ind_t ind)
{
    if (!ind.status && ind.cluster_id == 0xFFF1 && ind.profile_id == 0x0104 && ind.dst_endpoint == 10) {
        ESP_LOGI(TAG, "APS data size: %d", ind.asdu_length);
        return true;
    }
    return false;
}
esp_zb_aps_data_indication_handler_register(zb_apsde_data_indication_handler);
```

superbagus1712, 2024-04-01:
@xieqinan thanks for your support but i still have problem on esp_zigbee_aps.h  No such file or directory
    2 | #include "esp_zigbee_aps.h" .How do i fix this ? i keep trying to reconfigure the project also write the library on header file it doesnt work

superbagus1712, 2024-04-01:
[image]

superbagus1712, 2024-04-01:
@xieqinan thanks for your support . but it have problem when i build the project
[image]

superbagus1712, 2024-04-01:
how i declare s_value @xieqinan

superbagus1712, 2024-04-01:
thanks for your support @xieqinan . How about the zigbee task ? i still dont understand to add the zigbee APS task

superbagus1712, 2024-04-01:
@xieqinan i also have this problem on receiver side 
[image]

xieqinan, 2024-04-01:
@superbagus1712 ,

It's just a data formatting issue. You can refer to either `ESP_LOGI(TAG, "APS data size: %lld", ind.asdu_length);` or `ESP_LOGI(TAG, "APS data size: %d", (uint16_t)ind.asdu_length);`.

superbagus1712, 2024-04-01:
@xieqinan there s nothing happen on receiver after pressing button
[image]

superbagus1712, 2024-04-01:
here my senderfile @xieqinan  
`typedef struct light_bulb_device_params_s {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t  endpoint;
    uint16_t short_addr;
} light_bulb_device_params_t;

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

static const char *TAG = "APSsend";

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, , TAG, "Failed to start Zigbee bdb commissioning");
}

static uint8_t s_value[10];

static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        esp_zb_apsde_data_req_t req;
        // Data
        for (int i = 0; i < sizeof(s_value); i++) {
            s_value[i] = i;
        }
        req.dst_addr_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        // The basic information 
        req.dst_short_addr = 0xd47d;       
        req.dst_endpoint = 10;
        req.profile_id = 0x104;
        req.cluster_id = 0xFFF1;
        req.src_endpoint = 10;
        // Payload
        req.asdu_length = sizeof(s_value);
        req.asdu = s_value;
        // routing 
        req.radius = 2;
        req.tx_options = (ESP_ZB_APSDE_TX_OPT_ACK_TX | ESP_ZB_APSDE_TX_OPT_FRAG_PERMITTED);
        req.use_alias = false;
        ESP_EARLY_LOGI(TAG, "Send aps data request");
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_aps_data_request(&req);
        esp_zb_lock_release();
    }
}

static esp_err_t deferred_driver_init(void)
{
    ESP_RETURN_ON_FALSE(switch_driver_init(button_func_pair, PAIR_SIZE(button_func_pair), esp_zb_buttons_handler), ESP_FAIL, TAG,
                        "Failed to initialize switch driver");
    return ESP_OK;
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    esp_zb_zdo_signal_device_annce_params_t *dev_annce_params = NULL;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Initialize Zigbee stack");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
[... 54 lines cut ...]

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZC_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_switch_cfg_t switch_cfg = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_switch_ep = esp_zb_on_off_switch_ep_create(HA_ONOFF_SWITCH_ENDPOINT, &switch_cfg);
    esp_zb_device_register(esp_zb_on_off_switch_ep);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));

    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}`

superbagus1712, 2024-04-01:
here my receiver file 
`static const char *TAG = "APSReceive";
/********************* Define functions **************************/
static esp_err_t deferred_driver_init(void)
{
    light_driver_init(LIGHT_DEFAULT_OFF);
    return ESP_OK;
}

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, , TAG, "Failed to start Zigbee commissioning");
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Initialize Zigbee stack");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Deferred driver initialization %s", deferred_driver_init() ? "failed" : "successful");
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
            /* commissioning failed */
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());
        } else {
            ESP_LOGI(TAG, "Network steering was not successful (status: %s)", esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
    default:
        ESP_LOGI(TAG, "ZDO signal: %s (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type,
                 esp_err_to_name(err_status));
        break;
    }
}
static bool zb_apsde_data_indication_handler(esp_zb_apsde_data_ind_t ind)
{
    if (!ind.status && ind.cluster_id == 0xFFF1 && ind.profile_id == 0x0104 && ind.dst_endpoint == 10) {
        ESP_LOGI(TAG, "APS data size: %d", (uint16_t)ind.asdu_length);
        return true;
    }
    return false;
}

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    esp_zb_device_register(esp_zb_on_off_light_ep);
    esp_zb_aps_data_indication_handler_register(zb_apsde_data_indication_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}`

superbagus1712, 2024-04-01:
@xieqinan it seems i didnt see any problem

xieqinan, 2024-04-01:
Hi,
If you use the code provided by me directly, I think it has some problems.

If you use the code provided by me directly, there might be some issues:

- The `req.dst_short_addr` should be set to the remote device address (0xaed4).
- It's recommended to create an endpoint identifier  1 with a custom cluster (0xFFF1).
-

superbagus1712, 2024-04-01:
@xieqinan still same problem i have set the `req.dst_short_addr` to  (0xaed4)

xieqinan, 2024-04-01:
> @xieqinan there s nothing happen on receiver after pressing button ![WhatsApp Image 2024-04-01 at 3 21 23 PM](https://private-user-images.githubusercontent.com/141626820/318381642-d9c02c70-7325-4a0b-95aa-fb9ea3a7fb5b.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5NjAyMzMsIm5iZiI6MTcxMTk1OTkzMywicGF0aCI6Ii8xNDE2MjY4MjAvMzE4MzgxNjQyLWQ5YzAyYzcwLTczMjUtNGEwYi05NWFhLWZiOWVhM2E3ZmI1Yi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDFUMDgyNTMzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjdjYzhiNzNjNmZkMTg0YjQzZTU2ZmE0ODBhM2I1NmRkNDE3ZWU5MWY2Njg5NjMzODliOGE0NTBiMjM0YzAwOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.kUgvUk3zeR2HDKASyy-eX4Dyz9JZydJIgRR5p-0Qfv0)

The short address value 0xaed4 is taken from the picture, but it will change if the device is factory reset. Have you updated it accordingly?

superbagus1712, 2024-04-01:
@xieqinan yes already

xieqinan, 2024-04-01:
@superbagus1712,

Could you utilize the sniffer to check the packet whether has been sent by the switch device?

superbagus1712, 2024-04-02:
@xieqinan  what kind sniffer that i can use ?

xieqinan, 2024-04-02:
@superbagus1712 

Refer to https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark.

chshu, 2024-04-28:
@superbagus1712 The APS usage is commented here: https://github.com/espressif/esp-zigbee-sdk/issues/296#issuecomment-2024779903. It doesn't care what's the data format (no matter JSON or any others). Feel free to reopen if you have any further questions.

closed 2024-04-28 by chshu