espressif/esp-zigbee-sdk #297: Reduce the lib size

xmow49, 2024-03-27:
### Question

Hello,
Im using the Zigbee lib and all work great.
In my project, the partition size of my main image is about 1.9MB.
Currently, I have only 3% space left.

I tested to run ` idf.py size-components` and the biggest components is the zboss stack:
```
Per-archive contributions to ELF file:
            Archive File DRAM .data & 0.bss .rtc.text .eh_frame IRAM0 .text ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
     libzboss_stack.ed.a        176   28638         0         0           0        28814      254356      2205                0          0      256737
            libble_app.a        146     529         0         0       15786        16461      170474      7602                0          0      194008
            libpthread.a          0      12         0         0           0           12        1144    177599                0          0      178743
           libnet80211.a       1527    8700         0         0        5332        15559      140026     16110            10805          0      162995
         libmbedcrypto.a        184     288         0         0          84          556       82022     35150                0          0      117440
[....]
```

I can see that the `.text` is about 248KB.
Is it possible to compile the lib without all of this text or add an option in sdkconfig ?
All ZigBee assert contain strings of the path from the computer that compiled the lib.

Currently, Zigbee trace and Zigbee sniffer are disabled.
I have only enable Zigbee and set to End Device.

Many Thanks

chshu, 2024-03-28:
@xmow49 Thanks for the suggestion. We will support two zigbee lib versions in the coming release:
- release version for production, it keeps the error log only
- debug version for development, it enabled all debug logs

We will also investigate and optimize the zigbee lib code size with more options.

xieqinan, 2024-04-07:
@xmow49 @Hedda ,

> @xmow49 Thanks for the suggestion. We will support two zigbee lib versions in the coming release:
> 
> * release version for production, it keeps the error log only
> * debug version for development, it enabled all debug logs

The feature has been supported in esp-zigbee-sdk v1.2.2. In the release version, the code size has been reduced by about 100KB compared to esp-zigbee-sdk v1.2.1 in the [switch example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch). If you follow the [esp-idf compiler optimization](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/size.html#reducing-overall-size) method for your project, the code size can be further decreased.

xmow49, 2024-04-07:
Hi @xieqinan 
I updated the lib to 1.2.2 and unchecked debug mode.
Now I have this:
``` 
Per-archive contributions to ELF file:
            Archive File DRAM .data & 0.bss .rtc.text .eh_frame IRAM0 .text ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
     libzboss_stack.ed.a        213    9693         0         0           0         9906      203730      2205                0          0      206148
            libble_app.a        146     529         0         0       15786        16461      170474      7602                0          0      194008
[...]
```

So yes, there is a decrease of 50 KB but not 100KB as you mentioned.
My compiler is already in size optimisation mode.
Is there another thing to do? 
[image]

xieqinan, 2024-04-09:
@xmow49 
> Is there another thing to do?

The size of the "libzboss_stack.ed.a" has been reduced by 50K. By following the "Reducing Overall Size" guidance, it's possible to further decrease the total size of the firmware.
We are currently focused on optimizing the code size further, but for now, that's all we have accomplished in 1.2.2 version.

chshu, 2024-04-14:
@xmow49 In the recent releases, we have introduced the following changes to reduce the Zigbee lib size:
- enable the release build options for the release version lib, it keeps the error log only
- register the callbacks only for the enabled clusters

Please follow the [minimizing-binary-size](https://docs.espressif.com/projects/esp-idf/en/v5.2.1/esp32h2/api-guides/performance/size.html#minimizing-binary-size) instruction to further reduce the overall application binary.

xmow49, 2024-04-14:
Hello,
Thanks for the feedback.
I tried the 1.2.3 and I gained 2 KB.
I will test the minimizing-binary-size instruction

Thanks

chshu, 2024-04-16:
Closing the issue since the planed optimizations all have been done.

closed 2024-04-16 by chshu