espressif/esp-zigbee-sdk #300: How does the write command work?

HOSTEER, 2024-03-28:
### Question

Hello, I'm trying to write multiple attributes simultaneously, I'm using a custom cluster. When I use the write command the ESP32-C6 reset itself. Can someone help me to know how to properly make the cmd_req to write ?

- Here is the custom cluster init, it is the same on the Coordinator and End Device
```c
    int16_t value = 0;
    uint64_t value2 = 0;
    uint32_t array_alue[3];
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_OT_CLUSTER_ID);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE );
    esp_zb_attribute_list = esp_zb_zcl_attr_list_create(CUSTOM_OT_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID, ESP_ZB_ZCL_ATTR_TYPE_16BIT, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&value);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID+1, ESP_ZB_ZCL_ATTR_TYPE_U64, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,&value2);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID+2, ESP_ZB_ZCL_ATTR_TYPE_U32, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,&array_alue[0]);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID+3, ESP_ZB_ZCL_ATTR_TYPE_U32, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,&array_alue[1]);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list, CUSTOM_OT_CLUSTER_ATTR_ERROR_ID+4, ESP_ZB_ZCL_ATTR_TYPE_U32, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,&array_alue[2]);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_attribute_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
``` 

- Then there is the code calling the write comand
```c
                    uint32_t reported_data[3];
                    reported_data[0] = ((uint32_t) * data);
                    reported_data[1] = ((uint32_t) * data) + 1;
                    reported_data[2] = ((uint32_t) * data) + 2;
                    gpio_set_level(GPIO_OUTPUT_IO_0, 1);
                    esp_zb_lock_acquire(portMAX_DELAY);
                    esp_zb_zcl_write_attr_cmd_t cmd_req = {
                        .zcl_basic_cmd = {
                            .dst_addr_u.addr_short = 0x000,
                            .dst_endpoint = HA_ESP_LIGHT_ENDPOINT,
                            .src_endpoint = HA_ONOFF_SWITCH_ENDPOINT,
                        },
                        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
                        .clusterID = CUSTOM_OT_CLUSTER_ID,
                        .attr_number = 3,
                    };
                    cmd_req.attr_field->id = CUSTOM_OT_CLUSTER_ATTR_ERROR_ID+2;
                    cmd_req.attr_field->data.type = ESP_ZB_ZCL_ATTR_TYPE_U32;
                    cmd_req.attr_field->data.size = 4;
                    cmd_req.attr_field->data.value= &reported_data[0];
                    esp_zb_zcl_write_attr_cmd_req(&cmd_req);
                    esp_zb_lock_release();
                    vTaskDelay(10 / portTICK_PERIOD_MS);
                    gpio_set_level(GPIO_OUTPUT_IO_0, 0);
```

xieqinan, 2024-03-29:
@HOSTEER 

suggestion；
```
esp_zb_zcl_attribute_t attr_field[] = 
{
   {CUSTOM_OT_CLUSTER_ATTR_ERROR_ID + 0, {ESP_ZB_ZCL_ATTR_TYPE_U32, sizeof(uint32_t), reported_data[0]}},
   {CUSTOM_OT_CLUSTER_ATTR_ERROR_ID + 1, {ESP_ZB_ZCL_ATTR_TYPE_U32, sizeof(uint32_t), reported_data[1]}},
   {CUSTOM_OT_CLUSTER_ATTR_ERROR_ID + 2, {ESP_ZB_ZCL_ATTR_TYPE_U32, sizeof(uint32_t), reported_data[2]}},
}
esp_zb_zcl_write_attr_cmd_t cmd_req = {
                        .zcl_basic_cmd = {
                            .dst_addr_u.addr_short = 0x000,
                            .dst_endpoint = HA_ESP_LIGHT_ENDPOINT,
                            .src_endpoint = HA_ONOFF_SWITCH_ENDPOINT,
                        },
                        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
                        .clusterID = CUSTOM_OT_CLUSTER_ID,
                        .attr_number = 3,
                        . attr_field = attr_field,
                    };
esp_zb_lock_acquire(portMAX_DELAY);
esp_zb_zcl_write_attr_cmd_req(&cmd_req);
esp_zb_lock_release();
```

HOSTEER, 2024-03-29:
Thank you a lot ! It works now

closed 2024-03-29 by HOSTEER