espressif/esp-zigbee-sdk #301: HA_thermostat: undefined reference to `esp_zb_zcl_cluster_list_create'

pepijndevos, 2024-03-29:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### esp-zigbee-lib version.

1.2.1

### esp-zboss-lib version.

1.2.1

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Build and run the HA_thermostat example

### What is the actual behavior?

I get a linking error

### Steps to reproduce.

1. `cd examples/esp_zigbee_HA_sample/HA_thermostat`
2.  `idf.py set-target esp32c6`
3. `idf.py build`

Note that this worked perfectly for the HA_on_off_switch example which seems almost identical in terms of configuration files. So I'm having a hard time figuring out why the one works and the other doesn't.

### More Information.

```
$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/pepijn/code/floracle/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_thermostat/build
Executing "ninja all"...
[1/1] cd /home/pepijn/code/floracle/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_thermostat/build/bootloader/esp-idf/esptool_py && /h...000 bootloader 0x0 /home/pepijn/code/floracle/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_thermostat/build/bootloader/bootloader.bin
Bootloader binary size 0x54e0 bytes. 0x2b20 bytes (34%) free.
[4/6] Linking CXX executable thermostat.elf
FAILED: thermostat.elf 
: && /home/pepijn/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=/home/pepijn/code/floracle/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_thermostat/build/thermostat.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/thermostat.elf.dir/project_elf_src_esp32c6.c.obj -o thermostat.elf -L/home/pepijn/code/floracle/esp-idf/components/soc/esp32c6/ld   -L/home/pepijn/code/floracle/esp-idf/components/esp_rom/esp32c6/ld   -L/home/pepijn/code/floracle/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_thermostat/build/esp-idf/esp_system/ld   -L/home/pepijn/code/floracle/esp-idf/components/esp_phy/lib/esp32c6   -L/home/pepijn/code/floracle/esp-idf/components/esp_coex/lib/esp32c6   -L/home/pepijn/code/floracle/esp-idf/components/esp_wifi/lib/esp32c6 esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/ieee802154/libieee802154.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/switch_driver/libswitch_driver.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/ieee802154/libieee802154.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/console/libconsole.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  esp-idf/mbedtls/mbedtls/3rdparty/everest/libeverest.a  esp-idf/mbedtls/mbedtls/3rdparty/p256-m/libp256m.a  /home/pepijn/code/floracle/esp-idf/components/esp_coex/lib/esp32c6/libcoexist.a  /home/pepijn/code/floracle/esp-idf/components/esp_wifi/lib/esp32c6/libcore.a  /home/pepijn/code/floracle/esp-idf/components/esp_wifi/lib/esp32c6/libespnow.a  /home/pepijn/code/floracle/esp-idf/components/esp_wifi/lib/esp32c6/libmesh.a  /home/pepijn/code/floracle/esp-idf/components/esp_wifi/lib/esp32c6/libnet80211.a  /home/pepijn/code/floracle/esp-idf/components/esp_wifi/lib/esp32c6/libpp.a  /home/pepijn/code/floracle/esp-idf/components/esp_wifi/lib/esp32c6/libsmartconfig.a  /home/pepijn/code/floracle/esp-idf/componen [... cut ...]
/home/pepijn/code/floracle/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c:462:(.text.app_main+0x1c): undefined reference to `esp_zb_platform_config'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/pepijn/code/floracle/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_thermostat/build/log/idf_py_stderr_output_41782 and /home/pepijn/code/floracle/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_thermostat/build/log/idf_py_stdout_output_41782
```

pepijndevos, 2024-04-01:
A funny thing is that if I copy the thermostat C and H file into the light switch directory it compiles fine, so there seems to just be some subtle configuration error with the thermostat example. I will try your suggestions and report back.

I was seeing a similar error when trying to compile the example using Platform.io, so I do want to get to the bottom of what causes these linking errors.

pepijndevos, 2024-04-06:
Neither fullclean nor sdkconfig helped, but `git clean -fdX` solved the problem. Still not totally sure why.

closed 2024-04-06 by pepijndevos

Li-Gru, 2024-05-09:
> is the `Zigbee enable` option in menuconfig setted?

Thank you