espressif/esp-zigbee-sdk #302: Where do i define ZB_ED_ROLE?

Runepx4, 2024-03-29:
When I try to compile the Temperature Sensor example I get an error stating that I need to define **ZV_ED_ROLE** in the idf.py menuconfig. This error makes sense based on the code, but where do I do that (and what would be the standard value for the temperature sensor example)?

```
#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile sensor (End Device) source code.
#endif
```

0rion12, 2024-03-30:
I got exactly the same problem with the light example. 
As mentionned in the error, I checked the menuconfig, but I did not see any parameter related to ZB_ED_ROLE or zigbee.

amidarius, 2024-03-31:
Same here and also during compilation of HA_color_dimmable_light.

`#error Define ZB_ZCZR in idf.py menuconfig to compile light (Router) source code.`

and HA_on_off_light

` #error Define ZB_ED_ROLE in idf.py menuconfig to compile light (End Device) source code.`

For each example i did this :

`idf.py set-target esp32h2`

followed with this

`idf.py build`

ESP-IDF v5.1.2

xieqinan, 2024-04-01:
@Runepx4 , @0rion12 , @amidarius ,

Regarding the issue, I'm sorry, but we have encountered a CI issue that needs to be fixed. As a result, there is a  error in the release version that matches esp-zigbee-lib and esp-zboss-lib. We will fix it as soon as possible.

amidarius, 2024-04-01:
Changed and confirming that **HA_temperature_sensor** example now compiles and seen by Z2M after flashing the ESP32H2 device.

[image]
[image]

xieqinan, 2024-04-01:
Hello,

This issue has been resolved. Could you please update the esp-zigbee-sdk to version 1.2.2 and test it again? If the problem is resolved, please close this issue. Thank you.

0rion12, 2024-04-01:
For me version 1.2.2 is working.

0rion12, 2024-04-01:
> Changed and confirming that **HA_temperature_sensor** example now compiles and seen by Z2M after flashing the ESP32H2 device.
> 
> ![HA_temperature_sensor 2024-04-01 085045](https://private-user-images.githubusercontent.com/29826378/318360401-1370e4f0-5f3e-47b0-8547-0fbe437ec99d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5NzQ0NDQsIm5iZiI6MTcxMTk3NDE0NCwicGF0aCI6Ii8yOTgyNjM3OC8zMTgzNjA0MDEtMTM3MGU0ZjAtNWYzZS00N2IwLTg1NDctMGZiZTQzN2VjOTlkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAxVDEyMjIyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdmNzBmMDZiZWZjNmYyOWUwYmZiMmRiYjdiODMxZDZiOGY5ZjVjZTM5ZjhhYmJmYTM0OTk1ZGZhNDY1NjU0NTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sqFIqSObNFfU5ScIj9gLfbvKEkdv3bLtWRXPjp7evBI) ![HA_temperature_sensor 2024-04-01 085151](https://private-user-images.githubusercontent.com/29826378/318360409-42544c51-f2a0-43eb-aed3-253a1d4c8e29.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5NzQ0NDQsIm5iZiI6MTcxMTk3NDE0NCwicGF0aCI6Ii8yOTgyNjM3OC8zMTgzNjA0MDktNDI1NDRjNTEtZjJhMC00M2ViLWFlZDMtMjUzYTFkNGM4ZTI5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAxVDEyMjIyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiN2NmMzEzZTA2ZGFkMzYzZTk1YWRhZjc1ZWE2OTBlMWQxMzE3OTU2NDE4MmJjNDZiODQ1M2RiM2NjMzAwYzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.oouzXWb5FDEAWasbeb-RlaqHOuaQ19HAKPa8Sso-yEc)

Hello, I try to use the light example. I use Z2M with HA. I am able to make it join the network. 
But then since it is not a supported device, I got nothing under the expose tab.  
[image]
[image]

How did you manage to get data from your device ? Did you create a custom config somewhere ?

lhespress, 2024-04-02:
@0rion12 `Z2M`, what do you mean [zigbee2mqtt](https://github.com/zigbee2mqtt) ? BTW, Could you create a new issue because of this issue has been resolved?

closed 2024-04-03 by chshu