espressif/esp-zigbee-sdk #303: Gateway Button Handler

Nicqhhq, 2024-04-01:
### Question

Hi! 

I would like to know if there is any code example that makes the esp32c6 a zigbee gateway (Router) and that can receive a warning when a button that connected to the network is pressed, for example a sonoff SNZB-01 button

xieqinan, 2024-04-02:
@Nicqhhq ,

Could you provide more details about your question? Are you asking whether you expect the router to successfully connect to the network using a button press, and then the stack triggers a log?

Nicqhhq, 2024-04-02:
I have a sonoff SNZB-01 button and an esp32C6 I want to build a zigbee hub with the esp32c6, the button would connect to it and when I pressed the button I wanted the esp32c6 to make an http request for example

xieqinan, 2024-04-03:
@Nicqhhq ,

Do you mean that you expect the ESP32-C6 to act as a bridge between Wi-Fi and Zigbee? If so, what commands will be sent by the SNZB-01?

Nicqhhq, 2024-04-03:
From what I researched, it works with the ESP_ZB_ZCL_CLUSTER_ID_ON_OFF cluster, how could I capture it in the code? Another question can I ask to capture any event on the network so I can use it as if it were a "sniffer"

Nicqhhq, 2024-04-11:
I got a cc2531 antenna to sniff the network, is there any code that I can make the button connect to the esp32c6 so I can see what action it sends when clicked?

xieqinan, 2024-04-11:
@Nicqhhq 

> how could I capture it in the code? Another question can I ask to capture any event on the network so I can use it as if it were a "sniffer"

The ESP provides a sniffer method; you can refer to the [sniffer documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) for more details.

> I got a cc2531 antenna to sniff the network, is there any code that I can make the button connect to the esp32c6 so I can see what action it sends when clicked?

What role does Zigbee play in your button? The ESP-Zigbee SDK provides some [examples](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample) for users as a reference.

chshu, 2024-04-28:
@Nicqhhq Hope your question is fully resolved by https://github.com/espressif/esp-zigbee-sdk/issues/303#issuecomment-2049001948, please reopen if you have any further questions.

closed 2024-04-28 by chshu