espressif/esp-zigbee-sdk #306: Possible to get date & time from router/hub?

darianbjohnson, 2024-04-04:
### Question

Is is possible for an end device to obtain/request the current data and time from the zigbee router?

xieqinan, 2024-04-07:
@darianbjohnson ,

The Zigbee cluster provides the timer cluster for managing time, and Zigbee devices attach a timestamp when sending frames. Which way better suits your requirements?

darianbjohnson, 2024-04-07:
I just reviewed the time cluster API Documentation…. This is what I need. Is there an example that shows how to use the time cluster API?

xieqinan, 2024-04-07:
@darianbjohnson ,

We have not provided an example for the timer cluster temporarily. However, you can implement it based on the ZCL specification and refer to the examples of other clusters, such as the [temperature cluster](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor).

chshu, 2024-04-28:
@darianbjohnson please reopen if you have any further questions on this topic.

closed 2024-04-28 by chshu