espressif/esp-zigbee-sdk #307: Crash when receiving empty "binding table response"

simonborje, 2024-04-06:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.3

### esp-zigbee-lib version.

1.2.2

### esp-zboss-lib version.

1.2.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Be able to receive a binding table with zero entries without crashing.

### What is the actual behavior?

When a device responds to a `esp_zb_zdo_binding_table_req` with an empty table, the software crashes. When a non-empty binding table is returned the `esp_zb_zdo_binding_table_callback_t` callback is called and I can print the entries successfully. 

### Steps to reproduce.

1. Request the binding table from a device without any bindings using `esp_zb_zdo_binding_table_req`.
2. When the empty binding table is received, the application crashes every time.

### More Information.

The received empty table:
[image]

The crash logs can be seen below:
```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42015ffa  RA      : 0x42015ff6  SP      : 0x4087e280  GP      : 0x4080dfb0
0x42015ffa: zdo_binding_table_resp at esp_zigbee_zdo_command.c.obj:?
0x42015ff6: zdo_binding_table_resp at esp_zigbee_zdo_command.c.obj:?

TP      : 0x40867014  T0      : 0x40022494  T1      : 0xffffffe0  T2      : 0x73206263
0x40022494: multi_heap_internal_unlock in ROM

S0/FP   : 0x40819f1f  S1      : 0x40819f24  A0      : 0x000000ff  A1      : 0x40818260
A2      : 0x00000001  A3      : 0x00000004  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x00000004  A7      : 0x30203a6c  S2      : 0x00000006  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x420061d4  S6      : 0x00000000  S7      : 0x00000000
0x420061d4: bind_table_req_cb(esp_zb_zdo_binding_table_info_s const*, void*) at C:/temp/esp32-c6-c++/main/esp_zigbee_gateway.cpp:136

S8      : 0x00000001  S9      : 0x00000000  S10     : 0x00000006  S11     : 0x00000007
T3      : 0x202c3020  T4      : 0x3a746e75  T5      : 0x6f63202c  T6      : 0x30203a73
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000117
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4087e280: 0x00000000 0x00000002 0x00000000 0x000000ff 0x00000000 0x42015f46 0x7fffffff 0x000000ff
0x42015f46: zdo_binding_table_resp at esp_zigbee_zdo_command.c.obj:?

4087e2a0: 0x0000000f 0x00000000 0x40810270 0x420247ae 0x00000000 0x00000000 0x00000000 0x0000008f
0x420247ae: zb_sched_loop_iteration at ??:?

4087e2c0: 0x00000682 0x001f0040 0x4081e984 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e2e0: 0x00000000 0x4081e918 0x4081e984 0x4081eacc 0x4081efb4 0x4081effc 0x4081f274 0x42014e62
0x42014e62: esp_zb_main_loop_iteration at ??:?

4087e300: 0x4081efb4 0x4081effc 0x4081f274 0x420064e8 0x00000000 0x03000000 0x00000000 0x00000000
0x420064e8: esp_zb_task(void*) at C:/temp/esp32-c6-c++/main/esp_zigbee_gateway.cpp:971 (discriminator 2)

4087e320: 0x00000000 0x00000000 0x00000014 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e340: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e360: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
4087e380: 0x4087df40 0x00000123 0x4080f624 0x40821c58 0x4087e380 0x4080f61c 0x0000000f 0x4081ea38
4087e3a0: 0x4081ea38 0x4087e380 0x00000000 0x0000000a 0x4087d37c 0x6267697a 0x6d5f6565 0x006e6961
4087e3c0: 0x00000000 0x00000000 0x4087e370 0x00000005 0x00000001 0x00000000 0x00000000 0x00000009
4087e3e0: 0x408167b4 0x4081681c 0x40816884 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
4087e400: 0x00000000 0x4209f2e2 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x4209f2e2: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4087e420: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e440: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e460: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e480: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e4a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e4c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x40000000 0x00000014 0x4267695a 0x6c416565
0x40000000: _start in ROM

4087e4e0: 0x6e61696c 0x39306563 0x00000000 0x0000000c 0x3fffffff 0x00000000 0x40870000 0x00000280
4087e500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e520: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e540: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e5a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e5c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e5e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e600: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e620: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e640: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e660: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: c5cc0e767bd3b664

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1c (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd10
load:0x4086e610,len:0x2d0c
load:0x40875728,len:0x17d0
entry 0x4086c410
I (22) boot: ESP-IDF v5.1.3 2nd stage bootloader
```

xieqinan, 2024-04-07:
@simonborje ，

Does the crashing also occur in other versions of the esp-zigbee-sdk?

simonborje, 2024-04-07:
@xieqinan 
I do get the crash also on version 1.0.7 with a slightly different log. In that case the log from within the callback function has time to be printed but the program crashes right after.

The callback function looks like this:
```c++
static void bind_table_req_cb(const esp_zb_zdo_binding_table_info_t *table_info, void *user_ctx) {
  if (table_info != nullptr) {
    ESP_LOGI(TAG, "bind_table_req_cb status: %d, count: %d, total: %d",
             table_info->status, table_info->count, table_info->total);
  }
}

```

Logs after crash:

```
I (25001) ESP_ZB_GATEWAY: bind_table_req_cb status: 0, count: 0, total: 0

assert failed: heap_caps_free heap_caps.c:387 (heap != NULL && "free() target pointer is outside heap areas")
Core  0 register dump:
MEPC    : 0x4080062a  RA      : 0x40807602  SP      : 0x4087e130  GP      : 0x4080df90
0x4080062a: panic_abort at C:/Users/simon/esp/v5.1.3/esp-idf/components/esp_system/panic.c:472
0x40807602: __ubsan_include at C:/Users/simon/esp/v5.1.3/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x40866134  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000001  S1      : 0x0000005a  A0      : 0x4087e178  A1      : 0x4080e5b5
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4081a000
A6      : 0x00000008  A7      : 0x76757473  S2      : 0x4087e178  S3      : 0x4087e2ad
S4      : 0x4080e5b4  S5      : 0x4087e178  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4087e130: 0x00000026 0x40810e31 0x420c8e74 0x4080c814 0x00373833 0x00000000 0x00000000 0x00000000
0x4080c814: esp_ptr_in_drom at C:/Users/simon/esp/v5.1.3/esp-idf/components/esp_hw_support/include/esp_memory_utils.h:301
 (inlined by) __assert_func at C:/Users/simon/esp/v5.1.3/esp-idf/components/newlib/assert.c:63

4087e150: 0x00000000 0x4080eab8 0x420c8e74 0x4080ee08 0x420c8df2 0x4080eac8 0x4087e140 0x4080eacc
4087e170: 0x420c8d9c 0x4080e5b4 0x65737361 0x66207472 0x656c6961 0x68203a64 0x5f706165 0x73706163
4087e190: 0x6572665f 0x65682065 0x635f7061 0x2e737061 0x38333a63 0x68282037 0x20706165 0x4e203d21
4087e1b0: 0x204c4c55 0x22202626 0x65657266 0x74202928 0x65677261 0x6f702074 0x65746e69 0x73692072
4087e1d0: 0x74756f20 0x65646973 0x61656820 0x72612070 0x22736165 0x00000029 0x00000001 0x00000000
4087e1f0: 0x4081a000 0x203a6c61 0x420cb000 0x0000000b 0x00000000 0x42006208 0x00000000 0x00000000
0x42006208: bind_table_req_cb(esp_zb_zdo_binding_table_info_s const*, void*) at C:/Users/simon/Documents/ESP-IDF/esp32-c6-c++/main/esp_zigbee_gateway.cpp:212

4087e210: 0x00000000 0x00000000 0x00000000 0x00000000 0x2c30203a 0x746e756f 0x63202c30 0x203a7375
4087e230: 0x00000000 0x00000016 0x00000003 0x4207c462 0x0000001c 0x42006208 0x00000000 0x0000000b
0x4207c462: zb_osif_scheduler_event at ??:?
0x42006208: bind_table_req_cb(esp_zb_zdo_binding_table_info_s const*, void*) at C:/Users/simon/Documents/ESP-IDF/esp32-c6-c++/main/esp_zigbee_gateway.cpp:212

4087e250: 0x00000000 0x00000000 0x07934bd8 0x40800a9c 0x00000000 0x00000000 0x40810e64 0x42017472
0x40800a9c: heap_caps_free at C:/Users/simon/esp/v5.1.3/esp-idf/components/heap/heap_caps.c:388
0x42017472: zdo_binding_table_resp at esp_zigbee_zdo_command.c.obj:?

4087e270: 0x00000000 0x00000000 0x00000000 0x4202969e 0x00000000 0x4087e8b0 0x00000043 0x0000065c
0x4202969e: zb_mac_logic_iteration at ??:?

4087e290: 0x00000000 0x00000000 0x00000001 0x42027e0e 0x00000000 0x420172fa 0x000b0003 0x424f4300
0x42027e0e: zb_sched_loop_iteration at ??:?
0x420172fa: zdo_binding_table_resp at esp_zigbee_zdo_command.c.obj:?

4087e2b0: 0x00000078 0x00000008 0x4087e8fc 0x4087ea2c 0x4087eddc 0x4087ee04 0x4087efac 0x42056080
0x42056080: zboss_main_loop_iteration at ??:?

4087e2d0: 0x00000000 0x00000000 0x4087f104 0x420162b4 0x4087eddc 0x4087ee04 0x4087efac 0x42006524
[... 1 lines cut ...]
0x42006524: esp_zb_task(void*) at C:/Users/simon/Documents/ESP-IDF/esp32-c6-c++/main/esp_zigbee_gateway.cpp:1033 (discriminator 2)

4087e2f0: 0x00000000 0x00000000 0x00000000 0x03000000 0x00000000 0x00000000 0x00000014 0x00000000
4087e310: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e330: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4087e350: 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4087e1b0 0x000009ab 0x4080f560 0x4080f560 0x4087e35c
4087e370: 0x4080f558 0x00000014 0x4087e9f4 0x4087e9f4 0x4087e35c 0x00000000 0x00000005 0x4087d358
4087e390: 0x6267695a 0x6d5f6565 0x006e6961 0x00ffffff 0x00000000 0x4087e350 0x00000005 0x00000000
4087e3b0: 0x00000000 0x00000000 0x00000009 0x4081afb4 0x4081b01c 0x4081b084 0x00000000 0x00000000
4087e3d0: 0x00000001 0x00000000 0x00000000 0x00000000 0x420b359a 0x00000000 0x00000000 0x00000000
0x420b359a: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4087e3f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e410: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[... cut ...]
4087e490: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40000000
0x40000000: _start in ROM

4087e4b0: 0x00000060 0x00000000 0x00000101 0x00000036 0x00000000 0x00000000 0x4087e518 0xb33fffff
4087e4d0: 0x00000000 0x00000001 0x00000001 0x00000000 0x00000000 0x00000000 0x4087e570 0xb33fffff
4087e4f0: 0x00000000 0xffffffff 0x00000000 0x00000000 0x00000000 0x00000000 0x4087e5c8 0x00000000
4087e510: 0x00000000 0x00000054 0x00000000 0x4087e518 0x00000000 0x00000000 0x00000000 0x4087e530
```

simonborje, 2024-04-09:
@xieqinan 
I get the crash with v1.2.0 as well:

```
assert failed: heap_caps_free heap_caps.c:387 (heap != NULL && "free() target pointer is outside heap areas")
Core  0 register dump:
MEPC    : 0x4080062a  RA      : 0x4080760e  SP      : 0x4087e090  GP      : 0x4080dfa0
0x4080062a: panic_abort at C:/Users/simon/esp/v5.1.3/esp-idf/components/esp_system/panic.c:472
0x4080760e: __ubsan_include at C:/Users/simon/esp/v5.1.3/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x40865ac4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000001  S1      : 0x0000005a  A0      : 0x4087e0d8  A1      : 0x4080e5cd
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4081a000
A6      : 0x00000008  A7      : 0x76757473  S2      : 0x4087e0d8  S3      : 0x4087e20d
S4      : 0x4080e5cc  S5      : 0x4087e0d8  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4087e090: 0x00000026 0x408110b9 0x420c8e74 0x4080c820 0x00373833 0x00000000 0x00000000 0x00000000
0x4080c820: esp_ptr_in_drom at C:/Users/simon/esp/v5.1.3/esp-idf/components/esp_hw_support/include/esp_memory_utils.h:301
 (inlined by) __assert_func at C:/Users/simon/esp/v5.1.3/esp-idf/components/newlib/assert.c:63

4087e0b0: 0x00000000 0x4080ead0 0x420c8e74 0x4080ee20 0x420c8df2 0x4080eae0 0x4087e0a0 0x4080eae4
4087e0d0: 0x420c8d9c 0x4080e5cc 0x65737361 0x66207472 0x656c6961 0x68203a64 0x5f706165 0x73706163
4087e0f0: 0x6572665f 0x65682065 0x635f7061 0x2e737061 0x38333a63 0x68282037 0x20706165 0x4e203d21
4087e110: 0x204c4c55 0x22202626 0x65657266 0x74202928 0x65677261 0x6f702074 0x65746e69 0x73692072
4087e130: 0x74756f20 0x65646973 0x61656820 0x72612070 0x22736165 0x00000029 0x00000003 0x420b90ce
0x420b90ce: write at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/syscalls/syswrite.c:11

4087e150: 0x4087e1d9 0x0000001c 0x00000003 0x420b90ce 0x4087e1df 0x4087e19c 0x00000003 0x00000000
0x420b90ce: write at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/syscalls/syswrite.c:11

4087e170: 0x00000000 0x000000ff 0x420174d6 0x4207dfce 0x0000001c 0x00000042 0x00000003 0x00000000
0x420174d6: device_binding_table_req_timeout at esp_zigbee_zdo_command.c.obj:?
0x4207dfce: zb_esp_set_event at ??:?

4087e190: 0x00000000 0x00000016 0x00000003 0x4207e3c2 0x0000001c 0x4200650a 0x00000000 0x0000000b
0x4207e3c2: zb_osif_scheduler_event at ??:?
0x4200650a: bind_table_req_cb(esp_zb_zdo_binding_table_info_s const*, void*) at C:/Users/simon/Documents/ESP-IDF/esp32-c6-c++/main/esp_zigbee_gateway.cpp:137

4087e1b0: 0x00000000 0x00000000 0x07934bd8 0x40800a9c 0x00000000 0x408110e7 0x4081cd9c 0x420177da
0x40800a9c: heap_caps_free at C:/Users/simon/esp/v5.1.3/esp-idf/components/heap/heap_caps.c:388
0x420177da: zdo_binding_table_resp at esp_zigbee_zdo_command.c.obj:?

4087e1d0: 0x00000000 0x00000000 0x00000000 0x4202b29c 0x00000000 0x00000049 0x00000012 0x00000000
0x4202b29c: zb_mac_logic_iteration at ??:?

4087e1f0: 0x00000000 0x00000000 0x00000001 0x42029946 0x42017662 0xff0b0000 0x4087e42c 0x40808360
0x42029946: zb_sched_loop_iteration at ??:?
0x42017662: zdo_binding_table_resp at esp_zigbee_zdo_command.c.obj:?
0x40808360: xQueueTakeMutexRecursive at C:/Users/simon/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:788

4087e210: 0x4087edbc 0x4087e890 0x4087e8dc 0x4087ea0c 0x4087edbc 0x4087ede4 0x4087ef8c 0x42058a7e
0x42058a7e: zboss_main_loop_iteration at ??:?

4087e230: 0x00000000 0x00000000 0x4087f0ec 0x4201651c 0x4087edbc 0x4087ede4 0x4087ef8c 0x420063ce
0x4201651c: esp_zb_main_loop_iteration at ??:?
0x420063ce: esp_zb_task(void*) at C:/Users/simon/Documents/ESP-IDF/esp32-c6-c++/main/esp_zigbee_gateway.cpp:972 (discriminator 2)

4087e250: 0x00000000 0x03000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000014 0x00000000
4087e270: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e290: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4087e2b0: 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4087e070 0x00004e47 0x4080f580 0x4080f580 0x4087e2bc
4087e2d0: 0x4080f578 0x00000014 0x4087e9c4 0x4087e9c4 0x4087e2bc 0x00000000 0x00000005 0x4087d2b8
4087e2f0: 0x6267697a 0x6d5f6565 0x006e6961 0x00000000 0x00000000 0x4087e2b0 0x00000005 0x00000001
4087e310: 0x00000000 0x00000000 0x00000009 0x4081b444 0x4081b4ac 0x4081b514 0x00000000 0x00000000
4087e330: 0x00000001 0x00000000 0x00000000 0x00000000 0x420b55b0 0x00000000 0x00000000 0x00000000
0x420b55b0: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4087e350: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e370: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e390: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[... cut ...]
4087e3f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40000000
0x40000000: _start in ROM

4087e410: 0x00000014 0x4267695a 0x6c416565 0x6e61696c 0x39306563 0x00000000 0x00000054 0x00000000
4087e430: 0x4087e42c 0x4087e2bc 0x00000001 0x00000000 0x4087e444 0xffffffff 0x4087e444 0x4087e444
4087e450: 0x00000000 0x4087e458 0xffffffff 0x4087e458 0x4087e458 0x00000000 0x00000001 0x00000000
4087e470: 0x4000ffff 0x00000000 0xb33fffff 0x00000000 0x00000060 0x00000000 0x00000101 0x00000036
0x4000ffff: ppRxFragmentProc in ROM
```

xieqinan, 2024-04-09:
@simonborje ,

Thank you for your test, I will fixed it in next version.

closed 2024-04-12 by espressif-bot

xieqinan, 2024-04-12:
@simonborje ,

The issue has been fixed in esp-zigbee-sdk v1.2.2, please update version and test again.

simonborje, 2024-04-12:
@xieqinan
It seems to work fine with v1.2.3, thank you!