espressif/esp-zigbee-sdk #308: Why need to config the reporting info?

acha666, 2024-04-07:
### Question

I'm checking the HA_temperature_sensor example, on 

https://github.com/espressif/esp-zigbee-sdk/blob/c86f305efc693cef958705ad55b2092bcdde16d5/examples/esp_zigbee_HA_sample/HA_temperature_sensor/main/esp_zb_temperature_sensor.c#L174-L190

In this code snippet, esp_zb_zcl_update_reporting_info() is utilized, with the comment indicating it's for "Config the reporting info."

Why is this configuration necessary? What impact does it have on the functionality?

Thank you.

xieqinan, 2024-04-09:
@acha666 ，

The example aims to inform users that configuring the reporting information locally is an option. Whether this is necessary for you depends on your project's requirements.

acha666, 2024-04-09:
@xieqinan 

Does the config make any changes from default settings? What is the default setting?

xieqinan, 2024-04-09:
@acha666 ，

The defult setting can refer to the belowed code.
```
 .u.send_info.min_interval = 0, 
 .u.send_info.max_interval = 0, 
 .u.send_info.def_min_interval = 5, 
 .u.send_info.def_max_interval = 0, 
 .u.send_info.delta = 0, 
     
```
The esp-zigbee-sdk provides `esp_zb_zcl_find_reporting_info()` API for user to find the report information, which is helpful for you.

closed 2024-04-09 by acha666