espressif/esp-zigbee-sdk #310: EndDevice cannot sleep before start commissioning

grebenyuk51, 2024-04-07:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### esp-zigbee-lib version.

1.2.2

### esp-zboss-lib version.

1.2.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Devices started in factory reset mode with autostart=false, without starting commissioning and start sleeping

### What is the actual behavior?

Devices started in factory reset mode with autostart=false, without starting commissioning, start sleeping and immediately waking up

### Steps to reproduce.

I created esp_zb_sleepy_end_device example with one difference: I don't want to start searching network without button click

```
void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;

    ESP_LOGI(TAG, "ZDO signal: %s (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type,
                 esp_err_to_name(err_status));

    switch (sig_type)
    {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
    //Notifies the application that ZBOSS framework (scheduler, buffer pool, etc.) has started, but no join/rejoin/formation/BDB initialization has been done yet.
        ESP_LOGI(TAG, "SKIP_STARTUP. Device started up in %sfactory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non ");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        ESP_LOGI(TAG, "Zigbee stack initialized");
        break;
    case ESP_ZB_ZDO_SIGNAL_DEFAULT_START:
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    //Device started for the first time after the NVRAM erase
        if(err_status == ESP_OK)
        {
            ESP_LOGI(TAG, "FIRST_START. Device started up in %sfactory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non ");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            }
        }
        else
        {
            ESP_LOGE(TAG, "FIRST_START.  Device started up in %sfactory-reset mode with an error %d (%s)", esp_zb_bdb_is_factory_new() ? "" : "non ", err_status, esp_err_to_name(err_status));
        }
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
[... 113 lines cut ...]
E (9394) OutDoorSensor: Network steering was not successful (status: ESP_FAIL)
I (9398) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (9405) OutDoorSensor: Zigbee can sleep
Returned from light sleep, reason: timer (4), t=9555 ms, slept for 168 ms
I (9640) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (9640) OutDoorSensor: Zigbee can sleep
Returned from light sleep, reason: timer (4), t=10390 ms, slept for 783 ms
I (12686) OutDoorSensor: ZDO signal: BDB Steering (0xa), status: ESP_FAIL
E (12687) OutDoorSensor: Network steering was not successful (status: ESP_FAIL)
I (12691) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (12698) OutDoorSensor: Zigbee can sleep
Returned from light sleep, reason: timer (4), t=13657 ms, slept for 998 ms
I (15969) OutDoorSensor: ZDO signal: BDB Steering (0xa), status: ESP_FAIL
E (15970) OutDoorSensor: Network steering was not successful (status: ESP_FAIL)
I (15974) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (15981) OutDoorSensor: Zigbee can sleep
Returned from light sleep, reason: timer (4), t=16928 ms, slept for 998 ms
I (19253) OutDoorSensor: ZDO signal: BDB Steering (0xa), status: ESP_FAIL
E (19254) OutDoorSensor: Network steering was not successful (status: ESP_FAIL)
I (19258) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (19265) OutDoorSensor: Zigbee can sleep
[and stops]
```

I think allowing sleep before searching network for battery powered devices saving a little bit of battery and useful. What do you think? or it`s bad behavior?

xieqinan, 2024-04-09:
@grebenyuk51 ，

If I have not miss you mean, I believe that the issue is not a bug. Could you attempt to follow the below code for a test?

```
static void zb_buttons_handler(switch_func_pair_t* button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        /* Get the switch ieee address */
        ESP_LOGI(TAG, "Start network steering");
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
        esp_zb_lock_release();
    }
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Initialize Zigbee stack");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Deferred driver initialization %s", deferred_driver_init() ? "failed" : "successful");
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Factory new");
            } else {
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
            /* commissioning failed */
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %d)", err_status);
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());
        } else {
            ESP_LOGI(TAG, "Network steering was not successful (status: %d)", err_status);
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
      case ESP_ZB_COMMON_SIGNAL_CAN_SLEEP:
          ESP_LOGI(TAG, "Zigbee can sleep");
          esp_zb_sleep_now();
          break;
    default:
        ESP_LOGI(TAG, "ZDO signal: %s (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type, esp_err_to_name(err_status));
        break;
    }
}

```

grebenyuk51, 2024-04-09:
and this code leads to not sleep

> I (626) OutDoorSensor: SKIP_STARTUP. Device started up in factory-reset mode
> I (643) OutDoorSensor: Zigbee stack initialized
> I (651) OutDoorSensor: ZDO signal: BDB Device Start (0x5), status: ESP_OK
> I (656) OutDoorSensor: FIRST_START. Device started up in factory-reset mode
> I (664) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
> I (671) OutDoorSensor: Zigbee can sleep
> Returned from light sleep, reason: other (0), t=676 ms, slept for 0 ms
> I (682) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
> I (688) OutDoorSensor: Zigbee can sleep
> Returned from light sleep, reason: other (0), t=693 ms, slept for 0 ms
> I (681) main_task: Returned from app_main()
> I (700) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
> I (711) OutDoorSensor: Zigbee can sleep
> Returned from light sleep, reason: other (0), t=716 ms, slept for 0 ms
> I (723) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
> I (729) OutDoorSensor: Zigbee can sleep
> Returned from light sleep, reason: other (0), t=734 ms, slept for 0 ms
> I (739) HTU21D: Temperature: 21.76
> I (740) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
> I (750) OutDoorSensor: Zigbee can sleep
> Returned from light sleep, reason: other (0), t=755 ms, slept for 0 ms
> I (762) OutDoorSensor: ZDO signal: Can Sleep (0x16), status: ESP_OK
> I (768) OutDoorSensor: Zigbee can sleep
> Returned from light sleep, reason: other (0), t=773 ms, slept for 0 ms
>[...infinite...]

if add `esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);` to `case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:` device will sleep

kelin6, 2024-04-24:
@grebenyuk51 You are correct. Once the Zigbee main loop is up and running without Zigbee commissioning, the end-device should also be able to enter sleep mode. Thank you very much for reporting the issue. 

We have identified the root cause of the problem and have already fixed it in an internal MR. The next version of the esp-zigbee-sdk will include this fix.

grebenyuk51, 2024-04-24:
thank you. I will close the issue after my check

closed 2024-04-28 by espressif-bot

grebenyuk51, 2024-05-03:
with library version 1.3.1 device sleeps without Zigbee commissioning
thanks