espressif/esp-zigbee-sdk #312: Require devices information connected to network coordinator

nalves23, 2024-04-09:
### Question

I have been exploring the zigbee stack, its behavior in the relationship between coordinator and devices connected to the zigbee network such as routers and end devices. However, for my network coordinator application, I need to obtain information about the devices that are authenticated in the ZigBee network and I cannot find any function in the API that does this.

Can you tell me how I can do this? The objective will be to obtain a list of all network addresses, MAC addresses, capabilities and other information from devices connected to the network.

Best regards
nalves23

xieqinan, 2024-04-09:
@nalves23 ,

The esp-zigbee-sdk currently does not offer the aforementioned support; however, we will consider implementing this feature in the future.

nalves23, 2024-04-09:
Thankyou @xieqinan.

I will implement in FAT or SPIFFS memory the association devices table and add information of device when it join the network and remove it when they are removed.

Each storage type that you suggest to store this information?

Kind regards
nalves23

closed 2024-05-24 by espressif-bot

xieqinan, 2024-05-24:
Hello @nalves23 

> Can you tell me how I can do this? The objective will be to obtain a list of all network addresses, MAC addresses, capabilities and other information from devices connected to the network.

The esp-zigbee-sdk v1.3.2 provides APIs to retrieve information about network connections. Please refer to [this link](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/nwk/esp_zigbee_nwk.h#L263). Could you please update the libraries and test it? The issue can be reopened if you encounter any problems with these APIs.