espressif/esp-zigbee-sdk #314: Getting signal quality

acha666, 2024-04-10:
### Question

Hi,

Is there a API to get the signal quality(maybe LQI or RSSI) of current connection?

xieqinan, 2024-04-11:
@acha666 ，

No, there isn't a single API specifically for obtaining the RSSI. However, the general command response (ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID, ESP_ZB_CORE_CMD_WRITE_ATTR_RESP_CB_ID) includes RSSI information. You can try sending a read or write attribute packet to the remote device; it will respond with the `esp_zb_zcl_cmd_read_attr_resp_message_t`, which contains the RSSI member.

acha666, 2024-04-11:
@xieqinan

Thank you. However, it seems that this method is for coordinator devices only. Could you advise how to get the signal quality in an end device application?

xieqinan, 2024-04-15:
@acha666 ,

Why do you think this method can only be used in the coordinator? Is it because the coordinator does not include any cluster? I believe any device capable of sending read/write commands can use it to retrieve signal quality.

simonborje, 2024-04-15:
@xieqinan 
A related question: When I read the `info.header.rssi` value in the `ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID` callback handler on the coordinator side, the value is always 0 no matter the distance between the devices. Is there anything else I need to do to get the rssi value?
This is with SDK version 1.2.3 and IDF 5.1.3.

closed 2024-04-28 by espressif-bot

xieqinan, 2024-04-28:
@simonborje, @acha666 ,

The issue has bee resolved in esp-zigbee-sdk v1.3.0, could you please update the version and test again?