espressif/esp-zigbee-sdk #315: Adding tolerance attribute for pressure measurement cluster

acha666, 2024-04-10:
### Is your feature request related to a problem?

[image]

https://github.com/espressif/esp-zigbee-sdk/blob/3fbf2325b77c8601703def202911566a99dfe098/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_pressure_meas.h#L16-L26

The ZCL Spec shows there is an `tolerance` attribute (ID 0x0003) in pressure measurement cluster but it doesn't exist in `esp_zb_zcl_pressure_measurement_attr_t`

xieqinan, 2024-04-11:
@acha666 ，

No problem, as it is only an enumerated type that will not affect your project.

closed 2024-04-12 by espressif-bot

xieqinan, 2024-04-12:
@acha666,

The issue has been fixed in the latest esp-zigbee-sdk version, please update the version and test again.