espressif/esp-zigbee-sdk #316: crash after zigbee sterring while using BLE NIMBLE

diazmanuel, 2024-04-12:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.3

### esp-zigbee-lib version.

1.2.3

### esp-zboss-lib version.

1.2.3

### Espressif SoC revision.

ESP32-H2 & ESP32-C6

### What is the expected behavior?

The expected behavior is that it does not restart

### What is the actual behavior?

I have the following ecosystem
a cell phone that connects via ble to a zigbee gateway
a gateway that is put into address mode using the cell phone command
an end device zigbee device that enters the network when the gateway is in address mode

What is happening is that after the cell phone connects via ble and puts the gateway in Steering mode, and the end device enters the network, and as soon as the device manages to get the said device (the end device) into the network device) restarts with the following log:
```
I (976) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
I (1016) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
I (1023) taskZigbee:  Zigbee Task Created
W (1024) taskZigbee: ZDO signal: 23, status: -1
I (1025) taskZigbee: Zigbee signal: SKIP_STARTUP
I (1161) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1263) taskPerifericos: perifericos Handler Task Initialized
I (2323) BLE_INIT: Using main XTAL as clock source
I (2329) BLE_INIT: ble controller commit:[217f1bf]
I (2330) NimBLE_SPP_BLE_PRPH: BLE Host Task Started
I (2530) taskZigbee: Zigbee signal: DEVICE_FIRST_START
I (2530) taskZigbee: Zigbee stack initialized
I (2672) taskZigbee: Zigbee signal: STEERING
I (2672) taskZigbee: Network steering was not successful (status: -1)
I (7816) taskZigbee: Zigbee signal: STEERING
I (7816) taskZigbee: Network steering was not successful (status: -1)
I (12960) taskZigbee: Zigbee signal: STEERING
I (12960) taskZigbee: Network steering was not successful (status: -1)
I (18104) taskZigbee: Zigbee signal: STEERING
I (18104) taskZigbee: Network steering was not successful (status: -1)
I (23248) taskZigbee: Zigbee signal: STEERING
I (23248) taskZigbee: Network steering was not successful (status: -1)
I (28392) taskZigbee: Zigbee signal: STEERING
I (28392) taskZigbee: Network steering was not successful (status: -1)
I (33536) taskZigbee: Zigbee signal: STEERING
I (33536) taskZigbee: Network steering was not successful (status: -1)
Zigbee stack assertion failed common/zb_error_indication.c:81
 
abort() was called at PC 0x4205272d on core 0
0x4205272d: zb_assert at ??:?
 
Core  0 register dump:
MEPC    : 0x408005ea  RA      : 0x4080c6fa  SP      : 0x40823e20  GP      : 0x40812840  
0x408005ea: panic_abort at /home/aebrecht/esp/v5.1.3/esp-idf/components/esp_system/panic.c:472
0x4080c6fa: __ubsan_include at /home/aebrecht/esp/v5.1.3/esp-idf/components/esp_system/ubsan.c:313
 
TP      : 0x407fed48  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x40823e5c  S1      : 0x40823e40  A0      : 0x40823e5c  A1      : 0x40823e3e  
A2      : 0x00000000  A3      : 0x40823e89  A4      : 0x00000001  A5      : 0x4081c000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x4081824c  
S4      : 0x00000001  S5      : 0x40823fd8  S6      : 0x00000040  S7      : 0x00000000  
S8      : 0x0000003f  S9      : 0x00000000  S10     : 0x0000001b  S11     : 0x00000007  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?
 
MHARTID : 0x00000000  
 
Stack memory:
40823e20: 0x0000001b 0x00000000 0x40823e3c 0x40811004 0x00000040 0x40823fd8 0x00000001 0x40810030
0x40811004: __assert_func at /home/aebrecht/esp/v5.1.3/esp-idf/components/newlib/assert.c:34
0x40810030: spimem_flash_ll_auto_wait_idle_init at /home/aebrecht/esp/v5.1.3/esp-idf/components/hal/esp32c6/include/hal/spimem_flash_ll.h:221
(inlined by) spi_flash_hal_setup_auto_suspend_mode at /home/aebrecht/esp/v5.1.3/esp-idf/components/hal/spi_flash_hal_iram.c:143
 
40823e40: 0x35303234 0x64323732 0x420fb200 0x40813b78 0x40823e40 0x40813b94 0x40823e3c 0x726f6261
40823e60: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x32353032 0x20643237
40823e80: 0x63206e6f 0x2065726f 0x00000030 0x00000000 0x0000002f 0x40818648 0x420fb241 0x42052730
0x42052730: zb_aps_multi_pre_link_key_init at zb_init_common.c.obj:?
 
40823ea0: 0x00000040 0x40823fd8 0x0000003f 0x00000051 0x00000040 0x40823fd8 0x000000fe 0x42090300
0x42090300: zb_error_raise at ??:?
 
40823ec0: 0x00000000 0x40818648 0x40823f20 0x42053d82 0x6e61696c 0x39306563 0x000001c0 0x0000000e
0x42053d82: zb_nvram_write_end at ??:?
 
40823ee0: 0x00000004 0x00000000 0x00000000 0x00000000 0x00000040 0x40823fd8 0x40823fb0 0x00000000
40823f00: 0x0000000e 0x00000030 0x40818648 0x4205608c 0x00000000 0x00000000 0x40829720 0x00000000
0x4205608c: zb_nvram_write_dataset at ??:?
 
40823f20: 0x00000015 0x000e0060 0x00000000 0x00000000 0x40823f60 0x40816648 0x408284be 0x42056194
0x42056194: zb_aps_keypair_write at ??:?
 
40823f40: 0x00000001 0x40823f60 0x00000000 0x00000035 0x00000001 0x00000000 0x00000000 0x4206c516
0x4206c516: zb_secur_update_key_pair at ??:?
 
40823f60: 0xfe4eeb84 0x404ccaff 0x4267695a 0x6c416565 0x6e61696c 0x39306563 0x00000001 0x00000000
40823f80: 0x00000001 0x420f8a9c 0x0000003f 0x00000000 0x00000000 0x00000000 0x420f9000 0x00000000
40823fa0: 0x4082401e 0x0000001b 0x40823fd8 0x4206cb4c 0x4267695a 0x6c416565 0x6e61696c 0x39306563
0x4206cb4c: zb_secur_create_best_suitable_link_key_pair_set at ??:?
 
40823fc0: 0x4082401e 0x0000001b 0x40825249 0x420690b0 0x0000001b 0x40825231 0xfe4eeb84 0x404ccaff
0x420690b0: zb_aps_unsecure_frame at ??:?
 
40823fe0: 0x00000000 0x0000001b 0x4eeb841f 0x00000000 0x00000000 0x4204e6de 0x00000000 0x4082952f
0x4204e6de: zb_nlde_data_indication at ??:?
 
40824000: 0x00000000 0x0000001b 0x00000021 0x4204e000 0x00000000 0x00000000 0x02ffffff 0xffff0000
0x4204e000: zb_nlde_data_indication_continue at ??:?
 
40824020: 0x30000021 0x000000fb 0x000000ff 0x00006f00 0xf4fffb30 0x00000000 0x0000001d 0x00000000
40824040: 0x0000001b 0x00000000 0x0000003f 0x00000000 0x00000000 0x4204e6de 0x7fffffff 0x000000ff
0x4204e6de: zb_nlde_data_indication at ??:?
 
40824060: 0x0000000f 0x00000000 0x40816648 0x4205732c 0x00000000 0x4087eed0 0x408299cc 0x00000002
0x4205732c: zb_sched_loop_iteration at ??:?
 
40824080: 0x00000011 0x00000000 0x40829664 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408240a0: 0x00000011 0x4082bb2c 0x408299cc 0x4082ba28 0x00000011 0x0000fdee 0x00000001 0x42048996
0x42048996: esp_zb_main_loop_iteration at ??:?
 
408240c0: 0x00000011 0x0000fdee 0x00000001 0x42007d04 0xffff0001 0xffff0001 0x00000005 0x00000000
0x42007d04: MyZigbeeHandlertTask at /home/aebrecht/Documents/Zentraly/Multizona_Zigbee/firmware/firmware_G2_Multizona/src/task_zigbee/taskZigbee.c:462 (discriminator 2)
 
408240e0: 0x00000000 0x00000000 0x00000000 0xf2010410 0x000000ff 0xffff0001 0xffff0001 0x00000005
40824100: 0x00000001 0x00000000 0x0000000a 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824120: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824140: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00003340 0x00000000 0x00440010
40824160: 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824180: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408241a0: 0x00000000 0xc6fdc823 0xc6fec863 0xe400000a 0xfffe450b 0x04404cca 0x00000000 0x00000000
408241c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408241e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824200: 0x00000000 0x00010001 0x80020900 0x000000ca 0x00000000 0xff100000 0x00a8004f 0x00000000
 
 
ELF file SHA256: a28afcedf2fc3a84
 
Rebooting...
```

and after the reboot it can enter the network without problems

### Steps to reproduce.

1. connect the gateway via BLE with a cell phone
2. Start the steering
3. add a end device to the zigbee network
4. check if the esp crash

agustinebrecht, 2024-04-23:
Hi, I have the same problem. Can I solve it?

kelin6, 2024-04-24:
@diazmanuel @agustinebrecht  Could you please confirm if this crash log is from the Zigbee gateway?
Is the Zigbee gateway using a dual-chip solution (ESP32S3+ESP32H2) or a single-chip solution (ESP32C6)?

diazmanuel, 2024-04-24:
hello @kelin6, 
I'm sorry for commenting on an error in the explanation, the device that is crashing is the final device, I also clarify that all the devices mentioned work with a single chip, the coordinator uses an ESP32C6 and the final device (which is the one that crashes) a ESP32H2. The login corresponds to the final device (ESP32H2).

The steps to reproduce it are as follows:

1. The coordinator (ESP32C6) is put in stering mode, follow the examples in this repository
2. The BLE nimble is turned on on the end device (ESP32H2) (following the example of nimble_spp_server) and in parallel the zigbee module (following the examples in this repository with the exception that the device does not begin to perform steering but waits for a command please to start it)
3. The connection is established between the end device and the phone via BLE and the command is sent to start steering mode
4. Finally the device tries to pair with the coordinator and it is at that moment that the failure and reboot occurs as shown in the logs

some clarifications:
- By logging in the coordinator we were able to verify that the device managed to pair correctly, that is, the restart occurs after the pairing is achieved
- In other tests we replaced the end device (ESP32H2) with a repeater (ESP32C6) respecting the previous steps and it ended up giving the same results
- Finally, it is not possible for us to send you a code extract to test because it is very intertwined with other parts which makes it impossible to isolate it for testing. I believe that by following the examples we mentioned before you will be able to reproduce it, but if you do not succeed, we will try to provide you with more information.

lpy4105, 2024-04-26:
Hi @diazmanuel,
I failed to reproduce the issue, here is the quick example code I used: 
zed: https://github.com/lpy4105/esp-zigbee-sdk/tree/test/ble_zb_light/examples/ble_zb/ble_zb_light
zc: HA_on_off_switch in this repo.

It would be helpful if you could bridge the gap between the two code bases, especially the pairing process and additional flash operations if any.

diazmanuel, 2024-05-02:
hello @lpy4105 ,
We tested the example code that you gave us and it does not actually crash like in our case, we are working to bring your example closer to our use case, at the moment we could not replicate it in your code. On the other hand, in ours it continues to happen systematically, we continue analyzing our code to see if we have any errors but everything would indicate that this is not the case. 
With the Log that we gave you, could you give us a little more information about where the problem could be?

lpy4105, 2024-05-07:
> With the Log that we gave you, could you give us a little more information about where the problem could be?

@diazmanuel As the log indicates, the ZB stack wrote something to nvram, but failed to verify the data. We do find some risks when multiple tasks operating the nvram at the same time and have pushed a fix in `1.2.3`, but it just relieves the situation. We are appropriated if you could help to reproduce the issue, so that we could find the root cause.

diazmanuel, 2024-05-22:
Hello @lpy4105 , sorry for the delay in the response, we were trying to reproduce this problem with the code you gave us but we couldn't, we believe it is due to the use of memory in different tasks, honestly we couldn't isolate it to be able to pass it on to you and we chose to avoid the problem , that is, avoid the condition in which this error occurs, in any case we will leave the problem open and we are at your disposal if you require us to try it again in incoming sdk updates

lpy4105, 2024-05-23:
Hi @diazmanuel, thanks for your feedback. We have just found that the similar issue from another customer was caused by calling SDK API from different tasks without protection. Please check in your project if all the API calls are protected by `esp_zb_lock_acquire` and `esp_zb_lock_release`. Hopefully this could help you resolve this issue.

diazmanuel, 2024-05-31:
Hi @lpy4105 I appreciate the mention of these functions, I understand that this reserves the zigbee resources to be able to use them from other tasks, in our case it turned out to be like this since we manage a large amount of resources from an external task, I wanted to ask you from which version was added these features because we hadn't seen them before, and reading the documentation we noticed that it was something really important that we missed. I also take this opportunity to recommend that you add some example of multitasking to be able to adopt the recommended practices, for this update as well as for future ones in case of a change as significant as this one.

lpy4105, 2024-06-04:
>I wanted to ask you from which version was added these features 

I believe it was `v1.1.0` that we added this support and suggested (actually it is mandatory) the user to use the pairs of lock function to protect each SDK API call outside the zigbee main task. 

> I also take this opportunity to recommend that you add some example of multitasking to be able to adopt the recommended practices.

This have been demonstrated in our examples, you could see the API calls in ISR context have been wrapped with the lock. Sure, we could consider adding such example.

diazmanuel, 2024-06-05:
Perfect, thank you very much, I close the issue since with the new implementation we made the error no longer occurs (indeed we were not using these lock functions that you mention so we believe it was due to that). On the other hand, I would appreciate if you could add an example of multitasking to corroborate that we are using the resources appropriately. If this example includes low consumption in multitasking, it would also be well received since this would cover most use cases.

diazmanuel, 2024-06-05:
Hi @lpy4105, 
Perfect, thank you very much, I close the issue since with the new implementation we made the error no longer occurs (indeed we were not using these lock functions that you mention so we believe it was due to that). On the other hand, I would appreciate if you could add an example of multitasking to corroborate that we are using the resources appropriately. If this example includes low consumption in multitasking, it would also be well received since this would cover most use cases.

closed 2024-06-05 by diazmanuel