espressif/esp-zigbee-sdk #318: bind and unbind error

diazmanuel, 2024-04-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.3

### esp-zigbee-lib version.

1.2.3

### esp-zboss-lib version.

1.2.3

### Espressif SoC revision.

ESP32-H2 & ESP32-C6

### What is the expected behavior?

the expected behavior is for both commands to work as they should

### What is the actual behavior?

there are 2 different problems

1.  binding error
In this case the error is found when performing the binding, the method correctly calls the command and the callcack is also called, but it responds with an error message 133 corresponding to a timeout, this error occurs quite often but not always
Below is an excerpt of the implemented code
```
void zigbeeDeviceBindReq(uint16_t rid,esp_zb_ieee_addr_t src_addr,esp_zb_ieee_addr_t des_addr,uint16_t cluster_id,uint8_t src_ep,uint8_t des_ep){
    esp_zb_zdo_bind_req_param_t bind_req;
    uint16_t *user_rid = malloc(sizeof(uint16_t));
    *user_rid=rid;
    bind_req.cluster_id=cluster_id;
    bind_req.dst_addr_mode=ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
    memcpy(bind_req.src_address,src_addr,sizeof(esp_zb_ieee_addr_t));
    memcpy(bind_req.dst_address_u.addr_long,des_addr,sizeof(esp_zb_ieee_addr_t));
    bind_req.dst_endp=des_ep;
    bind_req.src_endp=src_ep;
    bind_req.req_dst_addr=esp_zb_address_short_by_ieee(src_addr);
    esp_zb_zdo_device_bind_req(&bind_req, bind_cb, (void *)user_rid);
}
static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
ESP_LOGE("ZIGBEE","CALLBACK BIND");
free(user_ctx);
}
```

2. unbinding error
On the other hand, this error corresponds to unbind . and there is no response when calling it, not only that but I placed a log inside the unbind callback and it is never called
Below is an excerpt of the implemented code

```
void zigbeeDeviceUnbindReq(uint16_t rid,esp_zb_ieee_addr_t src_addr,esp_zb_ieee_addr_t des_addr,uint16_t cluster_id,uint8_t src_ep,uint8_t des_ep){
esp_zb_zdo_bind_req_param_t bind_req;
uint16_t *user_rid = malloc(sizeof(uint16_t));
*user_rid=rid;
bind_req.cluster_id=cluster_id;
bind_req.dst_addr_mode=ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
memcpy(bind_req.src_address,src_addr,sizeof(esp_zb_ieee_addr_t));
memcpy(bind_req.dst_address_u.addr_long,des_addr,sizeof(esp_zb_ieee_addr_t));
bind_req.dst_endp=des_ep;
bind_req.src_endp=src_ep;
bind_req.req_dst_addr=esp_zb_address_short_by_ieee(src_addr);
esp_zb_zdo_device_unbind_req(&bind_req, unbind_cb, (void *)user_rid);
}
static void unbind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
ESP_LOGE("ZIGBEE","CALLBACK UNBIND");
free(user_ctx);
}
```

### Steps to reproduce.

perform binding and unbindings

xieqinan, 2024-04-19:
@diazmanuel 
> 1. binding error
>    In this case the error is found when performing the binding, the method correctly calls the command and the callcack is also called, but it responds with an error message 133 [...]

Regarding the issue, I have checked your code, which appears to be correct. However, the binding request is extensively used in the esp-zigbee-sdk examples. Could you double-check the addresses in the binding request? Alternatively, could you please provide me with a simple reproducing code?

diazmanuel, 2024-04-23:
hello @xieqinan,
After carrying out more tests I was able to identify that the binding problem is linked to using repeaters and subjecting the end device to stress tests, that is, disconnecting the coordinator and the repeaters repeatedly, at some point it begins to fail, I will leave this to you.  I understand that the problem is entering and leaving the network continuously.

Regarding the unbind issue, I created an example to demonstrate the problem, in this case the problem as I mentioned before happens all the time, the example runs a coordinator on the ESP32C6 and an end device with the ESP32H2, when pressing the button on the board development of the end device, it sends a bind and when you press it again it sends an unbind and this is repeated cyclically when you continue pressing it.
What is happening is that the binding callback is called correctly with status 0 but when calling unbind it does not enter its callback so it is not being executed. Below I leave the log of these bindings and unbinding and attach the example so you can test it.

```
E (781) ESP_ZB_ON_OFF_LIGHT: mac address: 60:55:f9:ff:fe:f7:73:de
E (781) ESP_ZB_ON_OFF_LIGHT: mac address: 40:4c:ca:ff:fe:45:3b:9c
E (781) ZIGBEE: ready for BINDs
E (5131) ZIGBEE: Sending BIND
E (5131) ZIGBEE: CALLBACK BIND status 0
E (6641) ZIGBEE: Sending UNBIND
E (10291) ZIGBEE: Sending BIND
E (10291) ZIGBEE: CALLBACK BIND status 0
E (15041) ZIGBEE: Sending UNBIND
E (19621) ZIGBEE: Sending BIND
E (19621) ZIGBEE: CALLBACK BIND status 0
E (23941) ZIGBEE: Sending UNBIND

```
[custom_test_mod.zip](https://github.com/espressif/esp-zigbee-sdk/files/15078981/custom_test_mod.zip)

closed 2024-04-28 by espressif-bot