espressif/esp-zigbee-sdk #321: undefined reference to `zb_get_tx_power'

mightChamp, 2024-04-19:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### esp-zigbee-lib version.

0.9.5

### esp-zboss-lib version.

0.7.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

The Function
```
 esp_zb_get_tx_power(&power);
```
Should Return Transmit Power of Zigbee.

### What is the actual behavior?

I am getting compile error
```
managed_components/espressif__esp-zigbee-lib/lib/esp32/libesp_zb_api_zczr.a(esp_zigbee_core.c.obj): in function `esp_zb_get_tx_power':
(.text.esp_zb_get_tx_power+0x6): undefined reference to `zb_get_tx_power'
```

### Steps to reproduce.

Step 1: Use Zigbee Gateway Example
Step 2: Add Below Code in esp_zb_app_signal_handler()-> Case: ESP_ZB_BDB_SIGNAL_STEERING
```
case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Network steering started");
            int8_t power;
            esp_zb_get_tx_power(&power);
            ESP_LOGI(TAG,"Zigbee Power:%d",power);
        }
        break;
```

Step 3: Compiler will generate error:
```
managed_components/espressif__esp-zigbee-lib/lib/esp32/libesp_zb_api_zczr.a(esp_zigbee_core.c.obj): in function `esp_zb_get_tx_power':
(.text.esp_zb_get_tx_power+0x6): undefined reference to `zb_get_tx_power'
```

xieqinan, 2024-04-26:
@mightChamp ,

Please update the esp-zigbee-lib and esp-zboss-lib to the latest 1.2.3 version which has contain `esp_zb_get_tx_power()` API.

mightChamp, 2024-05-07:
Sorry for this, and thanks for info.

closed 2024-05-07 by mightChamp