espressif/esp-zigbee-sdk #322: How to get Feedback/Acknowledgment of Sent cmd_req

mightChamp, 2024-04-19:
### Question

Hello,

I am developing window Covering client, I am able to operate Window Covering Properly by below function:
```
esp_zb_zcl_window_covering_cluster_send_cmd_req_t cmd_req;
cmd_req.address_mode=ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
cmd_req.cmd_id=ESP_ZB_ZCL_CMD_WINDOW_COVERING_UP_OPEN;
cmd_req.zcl_basic_cmd.src_endpoint=1;
cmd_req.zcl_basic_cmd.dst_endpoint=1;
for(int i=0;i<8;i++)
    cmd_req.zcl_basic_cmd.dst_addr_u.addr_long[i]=dev_list[0].ieee_addr[i];
cmd_req.cluster_id=ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING;
esp_zb_zcl_window_covering_cluster_send_cmd_req(&cmd_req);
```
Curtain Motor Works Properly by this command. But I want to get Feedback that motor operated or not.
If motor is not operated, I get response like :
```
ZDO signal: NLME Status Indication (0x32), status: ESP_OK
```

1)  How can I retrieve the short address or IEEE address of a curtain motor that fails to communicate from the above message?

2) Currently, I don't receive a success message confirming the curtain operation. Even if the motor communicates and executes the command (UP/Down/Stop), there's no feedback or acknowledgment. How can I obtain a success message for cmd_req?

3) How can I access the Received Signal Strength Indication (RSSI) and Link Quality of the response message?

Thank You

xieqinan, 2024-04-23:
@mightChamp ,

> How can I retrieve the short address or IEEE address of a curtain motor that fails to communicate from the above message?

Could you refer to the [customized_switch example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices/customized_client) which provides a way of retrieving the short address and IEEE address?

> Currently, I don't receive a success message confirming the curtain operation. Even if the motor communicates and executes the command (UP/Down/Stop), there's no feedback or acknowledgment. How can I obtain a success message for cmd_req?

First, you should check whether the device is already on your network. Then, ensure the IEEE address is correct when you send the `esp_zb_zcl_window_covering_cluster_send_cmd_req()`.

> How can I access the Received Signal Strength Indication (RSSI) and Link Quality of the response message?

The message response carries the RSSI information, but the value may be incorrect. This issue will be fixed in the next version.

mightChamp, 2024-04-23:
Thanks for your response.

> First, you should check whether the device is already on your network. Then, ensure the IEEE address is correct when you send the esp_zb_zcl_window_covering_cluster_send_cmd_req()

A new device has been successfully incorporated into the network and operates without any issues. The Gateway Host code also receives reports from this device. However, the problem arises when I attempt to send a command using esp_zb_zcl_window_covering_cluster_send_cmd_req(). While the curtain node responds correctly to the command and operates as expected(Up/Down/Stop), but I do not receive confirmation that the request sent via esp_zb_zcl_window_covering_cluster_send_cmd_req() was successful. 

There are two test case:
1. Curtain Power Off: When Curtain Power Off, then on sending esp_zb_zcl_window_covering_cluster_send_cmd_req(), I get below error after some few seconds, 3-4 times:
```
ZDO signal: NLME Status Indication (0x32), status: ESP_OK
```
2. Curtain Power On: When Curtain Power On, then on sending esp_zb_zcl_window_covering_cluster_send_cmd_req(), Curtian works according to command  (UP/Down/Stop), But No signal or response or feedback, is received. 

So, I have feedback that Curtain Not Worked, But  I didn't have feedback for Curtain has been worked.

So, In My Previous Zigbee Controller of EFR32 Series, I always get both feedback, command received by Curtain and Command didn't received by Curtain.

xieqinan, 2024-04-25:
@mightChamp 

> A new device has been successfully incorporated into the network and operates without any issues. The Gateway Host code also receives reports from this device. However, the problem arises when I [...]

Fine, the `esp_zb_zcl_command_send_status_handler_register()` function may be helpful for you in this case.

mightChamp, 2024-05-23:
I have updated to latest version, now I getting success response in esp_zb_core_action_handler_register(zb_action_handler);

Ok Now last issue for me is How I will get Link Quality, I am getting rssi, but there is no option for Link Quality.

xieqinan, 2024-05-27:
@mightChamp ，

The [network neighbor table API](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/nwk/esp_zigbee_nwk.h#L263) may be helpful for you.

mightChamp, 2024-05-29:
@xieqinan Thanks for info, Will this iterator provide all nodes info(remote nodes), or only directly connected nodes?

xieqinan, 2024-05-29:
@mightChamp ，

The iterator provides information on all connected nodes and devices that can communicate directly.

mightChamp, 2024-05-29:
@xieqinan So For remote node, there is no info for Link Quality

xieqinan, 2024-05-30:
@mightChamp ，

> @xieqinan So For remote node, there is no info for Link Quality

Yes, I think if the remote device cannot communicate directly with the current device over radio, there is no way to obtain its Link Quality value.

chshu, 2024-06-12:
will follow up here https://github.com/espressif/esp-zigbee-sdk/issues/360

closed 2024-06-12 by chshu