espressif/esp-zigbee-sdk #324: Implementing Direct String Sending and Receiving via AF Layer on ESP32H2

zhang-zi-ang, 2024-04-19:
### Question

I'm attempting to utilize the AF (Application Framework) layer akin to CC2530 for communication, with the goal of directly sending and receiving strings. However, I couldn't find relevant APIs in my ESP32H2 environment.

I'd like to understand if there's a method to accomplish this process, or if there are other sample routines available to support it.

xieqinan, 2024-04-23:
@zhang-zi-ang ，

Does the AF (Application Framework) you said indicate the APS(application support) layer of Zigbee?

zhang-zi-ang, 2024-05-29:
I obtained what I wanted from the following link. [https://github.com/espressif/esp-zigbee-sdk/issues/202]
Could you please tell me if there are any updates to this example?

closed 2024-05-29 by zhang-zi-ang

xieqinan, 2024-05-29:
@zhang-zi-ang ,

We do not currently plan to add an example for this issue, but we are adding more documentation for the APIs of the esp-zigbee-sdk, which will be helpful for you.

zhang-zi-ang, 2024-06-03:
@xieqinan 

I encountered the message "receive buffer full, drop the current frame" while using this routine. My IDF version is 5.1.3, and the SDK version is 1.3.2.

xieqinan, 2024-06-03:
@zhang-zi-ang ，

Please provide the detailed steps for reproducing the `buffer full` issue. This issue has been encountered by other customers as well. Please comment out the log in esp-idf and test again. The log is printed in the ISR, which can block other tasks and is not necessary.