espressif/esp-zigbee-sdk #327: Enhanced Hue Issue

john159753, 2024-04-20:
### Question

Is it possible I have uncovered a bug in the library where enhanced color hue changes always start from the same starting point? When I configure the library to use enhanced hue (I have a 12-bit LED driver so enhanced hue is appealing), all hue level changes with transition, seem to start the transition at the same exact H/S point. This doesn't happen when I use XY like the example starts you off at.

So every time I change the hue to a new color, it snaps to an initial hue  and then transitions as expected to the new color.

I will get more details as I have time, Even share a clean up version of my code if it helps, but I wasn't sure if this is something that other people have experienced or not.

xieqinan, 2024-04-24:
@john159753 ,

The enhanced hue will start from the value you assigned during the creation process. What behavior are you expecting from the enhanced hue?

john159753, 2024-04-26:
Hi @xieqinan 

I think what you stated makes sense, for the first time it turns on. But every single time i change a color, it starts the color change transition from the same spot.

How i define the endpoint 
```c
 uint8_t colorX = ESP_ZB_ZCL_COLOR_CONTROL_CURRENT_X_DEF_VALUE;
 uint8_t colorY = ESP_ZB_ZCL_COLOR_CONTROL_CURRENT_Y_DEF_VALUE;
 uint8_t colorMode = 0x00; // init with hue/sat color mode 
 uint8_t colorOpts = 0x01; // not sure what coloropts mean, but i found example with this being 0x1
 uint8_t colorECM = 0x00; // init with standard hue/sat ecm
 uint8_t colorCapabilites = 0x000B; //XY, Hue and Enhanced Hue
 uint8_t colorHue = 0x00;
 uint8_t colorSaturation = 0x00;
 uint16_t enhancedHue = 0x0000;

//set color attrributes
esp_zb_attribute_list_t *rgb_light_color_cfg = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_X_ID, &colorX);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_Y_ID, &colorY);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_MODE_ID, &colorMode);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_OPTIONS_ID, &colorOpts);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_COLOR_MODE_ID, &colorECM);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_CAPABILITIES_ID, &colorCapabilites);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_HUE_ID, &colorHue);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_SATURATION_ID, &colorSaturation);
esp_zb_color_control_cluster_add_attr(rgb_light_color_cfg, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_CURRENT_HUE_ID, &enhancedHue);
  
//set level
esp_zb_level_cluster_cfg_t level_cfg = level_config();
  esp_zb_attribute_list_t *rgb_light_level_cfg = esp_zb_level_cluster_create(&level_cfg);

//set onoff
esp_zb_on_off_cluster_cfg_t onoff_cfg = onoff_config();
  esp_zb_attribute_list_t *rgb_light_onoff_cfg = esp_zb_on_off_cluster_create(&onoff_cfg);
// set color attributes
  esp_zb_attribute_list_t *esp_zb_basic_cluster_rgb = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
//set basic
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster_rgb, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster_rgb, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, manufname[0]);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster_rgb, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &zclVersion);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster_rgb, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &powerSource);

```

callbacks example
```c++
case ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL:
            
            if(message->attribute.id == ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_HUE_ID  && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_U8) {
                standardhue = message->attribute.data.value ? *(uint8_t *)message->attribute.data.value : standardhue;
                sat = *(uint8_t *)esp_zb_zcl_get_attribute(message->info.dst_endpoint, message->info.cluster,ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_SATURATION_ID)->data_p;
               // log_i("shit hue changes to %d", standardhue);
                colorMode = 0x00;
                enhancedColorMode = 0x00;
            } else if(message->attribute.id == ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_SATURATION_ID  && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_U8) {
                sat = message->attribute.data.value ? *(uint8_t *)message->attribute.data.value : sat;
                hue = *(uint16_t *)esp_zb_zcl_get_attribute(message->info.dst_endpoint, message->info.cluster,ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_CURRENT_HUE_ID)->data_p;
                log_i("Light color sat changes to 0x%d", sat);
                colorMode = 0x00;
                enhancedColorMode = 0x03;
            } else if(message->attribute.id == ESP_ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_CURRENT_HUE_ID && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_U16) {
                hue = message->attribute.data.value ? *(uint16_t *)message->attribute.data.value : hue;
                sat = *(uint8_t *)esp_zb_zcl_get_attribute(message->info.dst_endpoint, message->info.cluster,ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_SATURATION_ID)->data_p;
                log_i("Light enhanced color full %d upper8 %d lower8 %d sat %d", hue, ((hue >> 0) & 0xFF), ((hue >> 8) & 0xFF), sat);
                enhancedColorMode = 0x03;
                colorMode = 0x00;
            } else if (message->attribute.id == ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_X_ID && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_U16) {
                light_color_x = message->attribute.data.value ? *(uint16_t *)message->attribute.data.value : light_color_x;
                light_color_y = *(uint16_t *)esp_zb_zcl_get_attribute(message->info.dst_endpoint, message->info.cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_Y_ID)->data_p;
[... 7 lines cut ...]
                log_i("Light color y changes to 0x%x", light_color_y);
                enhancedColorMode = 0x01;
[... cut ...]
              set_color(standardhue , sat); //overload for uint8_t
            }
          

            break;
```

Please see the video to help show the behavior i am seeing. I cant see where i might be messing up things in code. As i understand it the value of enhancedhue should be updated/tracked in the library. It seems like the library might not be tracking the last known enhancedcurrenthue value, which is why it always starts at the same spot.

https://photos.app.goo.gl/UHLMofnoBBVCXcFm6

xieqinan, 2024-04-28:
@john159753，

I tested the `move_to_enhance_hue()` function using esp-zigbee-sdk v1.2.3, and it worked fine. Could you please test it again with the demo?
 
 [HA_color_dimmable_example.zip](https://github.com/espressif/esp-zigbee-sdk/files/15141021/HA_color_dimmable_example.zip)

john159753, 2024-05-14:
I moved off of the Arduino EDP-IDF and now am using straight ESP-IDP in VSCode and working much better now. 

Sorry about the noise.

closed 2024-05-14 by john159753