espressif/esp-zigbee-sdk #328: Zigbee2MQTT connection problem - ESP32-H2/C6 HA_on_off_light example

SCiunczyk, 2024-04-22:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### esp-zigbee-lib version.

v1.2.3

### esp-zboss-lib version.

v1.2.3

### Espressif SoC revision.

ESP32-H2 ESP32-C6

### What is the expected behavior?

Successful pairing and operation with Zigbee2MQTT add-on on HA. 

### What is the actual behavior?

The pairing and operation works only with ZHA integration - device is detected, paired and has undefined model and manufacture names, but I can switch on/off led on board from my HA interface. On Z2M add-on device is detected but model and manufacture are not recognised. So it can't be operated.

### Steps to reproduce.

1. I'm using the newest example from [examples/esp_zigbee_HA_sample/HA_on_off_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light) with modifications to `esp_zb_task` 

```
char modelid[14]   = "ESP32C6.Light";
char manufname[10] = "Espressif";

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID,  &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);

    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    esp_zb_device_register(esp_zb_on_off_light_ep);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

```

2. The boards are: [H2](https://www.aliexpress.com/item/1005006020651611.html) and [C6](https://www.aliexpress.com/item/1005005565914909.html)
 
3. HA (latest version) is installed on Odroid N2+ (Armbian) with ConBee III as coordinator flashed with latest firmware

4. After switching to Zigbee2MQTT Edge - latest version I can't pair correctly

5. Board is cleared / re-flashed pairing on device side is reported as successful but model and manufacture are not recognised. So device can't be used with Z2M 

### More Information.

Log on ZHA pairing after fresh clear/flash/monitor and switching light from HA interface

```
I (159886) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (159886) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (163156) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (167246) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (167246) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (170516) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (174426) ESP_ZB_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (174426) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (175446) ESP_ZB_ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: 81:c8:28:43:21:e8:e0:73, PAN ID: 0x8bc4, Channel:11, Short Address: 0xe357)
W (195456) ESP_ZB_ON_OFF_LIGHT: Receive Zigbee action(0x12) callback
W (220006) ESP_ZB_ON_OFF_LIGHT: Receive Zigbee action(0x12) callback
I (283336) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x6), attribute(0x0), data size(1)
I (283336) ESP_ZB_ON_OFF_LIGHT: Light sets to On
I (284666) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x6), attribute(0x0), data size(1)
I (284666) ESP_ZB_ON_OFF_LIGHT: Light sets to Off
```

The device status ZHA.

[image]

The device status in Z2M
[image]
[image]

I suppose that problem is with device/manufacture name setup, but I have no idea how to fix it.

lhespress, 2024-04-23:
@SCiunczyk Please Add an new device as the follow links: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

SCiunczyk, 2024-04-23:
But I think that I can't add new device to Z2M with unknown _model_ and _manufacturer_ 

Let's assume that we are starting from [examples/esp_zigbee_HA_sample/HA_on_off_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light) and the example code of `esp_zb_task`

```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    esp_zb_device_register(esp_zb_on_off_light_ep);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

What is the recommended approach to define _model_ and _manufacturer_  property in code?

lhespress, 2024-04-24:
@SCiunczyk 
First, It can be define any value which you want if you do it as the [Support new devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html) step.

Second, We also could use the already existing devices which match the function, eg: 

https://github.com/Koenkk/zigbee-herdsman-converters/blob/4a8d546d9c1150d81e42d56a85e2315c32c4ed11/src/devices/xiaomi.ts#L2699

define model as `HLQDQ01LM` and manufacturer as `Xiaomi`.

SCiunczyk, 2024-04-24:
Look at my log:
[image]

First I have to pair device - but pairing failed - I have same symptoms as here: https://github.com/espressif/esp-idf/issues/12657

Again: the problem exists only with Zigbee2MQTT. With ZHA (Zigbee Home Automation) demo works perfectly.

kwerdenker, 2024-04-27:
> But I think that I can't add new device to Z2M with unknown _model_ and _manufacturer_
>
> Let's assume that we are starting from [...]

I had the same issue and was able to cobble together the following from different snippets around the issues. Compiled with ESP-IDF 5.2.1 and works with my ESP32-C6 and HA via deCONZ. 

```
char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

static void esp_zb_task(void *pvParameters)
{
    ESP_LOGI(TAG, "Entering ESP_ZB_TASK");
    // initialize Zigbee stack with Zigbee end-device config 
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    uint8_t zcl_version, null_values;

    zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE;
    null_values = ESP_ZB_ZCL_BASIC_POWER_SOURCE_DEFAULT_VALUE;

    // basic cluster create with fully customized 
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &zcl_version);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &null_values);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    // identify cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    ESP_ERROR_CHECK(esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &null_values));
    // group cluster create with fully customized 
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    ESP_ERROR_CHECK(esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &null_values));
    // scenes cluster create with standard cluster + customized
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_SCENES);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_CURRENT_GROUP_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_CURRENT_SCENE_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_SCENE_VALID_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_SCENE_COUNT_ID, &null_values);
    // on-off cluster create with standard cluster config
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    // create cluster lists for this endpoint
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    // add created endpoint (cluster_list) to endpoint list
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

lhespress, 2024-04-28:
@SCiunczyk What do you mean it paired failed when you changed the `model` as HLQDQ01LM and manufacturer as `Xiaomi`? Could you provide the sniffer packet?

SCiunczyk, 2024-04-28:
At the moment zero success unfortunately, I have adapted the code to new zboss 1.2.3

```
char modelid[] = {9, 'H', 'L', 'Q', 'D', 'Q', '0', '1', 'L', 'M'};
char manufname[] = {6, 'X', 'i', 'a', 'o', 'm', 'i'};

static void esp_zb_task(void *pvParameters)
{
    ESP_LOGI(TAG, "Entering ESP_ZB_TASK");
    // initialize Zigbee stack with Zigbee end-device config 
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    uint8_t zcl_version, null_values;

    zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE;
    null_values = ESP_ZB_ZCL_BASIC_POWER_SOURCE_DEFAULT_VALUE;

    // basic cluster create with fully customized 
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &zcl_version);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &null_values);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    // identify cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    ESP_ERROR_CHECK(esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &null_values));
    // group cluster create with fully customized 
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    ESP_ERROR_CHECK(esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &null_values));
    // scenes cluster create with standard cluster + customized
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_SCENES);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_CURRENT_GROUP_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_CURRENT_SCENE_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_SCENE_VALID_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_SCENE_COUNT_ID, &null_values);
    // on-off cluster create with standard cluster config
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    // create cluster lists for this endpoint
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));

    esp_zb_endpoint_config_t EPC = {
        .endpoint=HA_ESP_LIGHT_ENDPOINT, 
        .app_profile_id=ESP_ZB_AF_HA_PROFILE_ID, 
        .app_device_id=ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID, .app_device_version=1,
    };

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    // add created endpoint (cluster_list) to endpoint list
//    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, EPC);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

```
The model as HLQDQ01LM and manufacturer as Xiaomi changed according to @lhespress suggestions. ZHA integrations works:
[image]

But Zigbee2MQTT nor Zigbee2MQTTEdge is able to pair only joining:

[image]

I have no sniffer handy, have to obtain one. Do you have recommendation for not expensive device?

kwerdenker, 2024-04-28:
> I have no sniffer handy, have to obtain one. Do you have recommendation for not expensive device?
According to the docs, you can configure an esp32-c6 to work as a sniffer, if you have a second one on hand: https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32c6/developing.html#sniffer-and-wireshark

SCiunczyk, 2024-04-28:
I have two H2 and to C6 boards so will try....

lhespress, 2024-05-11:
@kwerdenker @SCiunczyk 

I changed `model` and `manufacturer` as follows:

```
char modelid[] = {17, 'l', 'u', 'm', 'i', '.', 'l', 'i', 'g', 'h', 't', '.', 'c', 'b', 'a', 'c', 'n', '1'};
char manufname[] = {4, 'L', 'U', 'M', 'I'};
```

SCiunczyk, 2024-05-12:
Will try, but now I'm travelling AFK. What coordinator do you have?

lhespress, 2024-05-13:
@SCiunczyk I used ConBee II as coordinator.

OOHehir, 2024-06-03:
For anyone looking at this I managed to connect to a Conbee II network successfully using the code from above but slightly modified. When joining a new co-ordinator (or probably rejoining the exisiting one) make sure to erase NVRAM:

```
    // Erase NVRAM before creating connection to new Coordinator
    // Comment out next line to erase NVRAM data if you are connecting to new Coordinator
    esp_zb_nvram_erase_at_start(true);
```

Working code:

```
char modelid[] = {9, 'H', 'L', 'Q', 'D', 'Q', '0', '1', 'L', 'M'};
char manufname[] = {6, 'X', 'i', 'a', 'o', 'm', 'i'};

static void esp_zb_task(void *pvParameters)
{
    ESP_LOGI(TAG, "Entering ESP_ZB_TASK");
    // initialize Zigbee stack with Zigbee end-device config
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    uint8_t zcl_version, null_values;

    zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE;
    null_values = ESP_ZB_ZCL_BASIC_POWER_SOURCE_DEFAULT_VALUE;

    // basic cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &zcl_version);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &null_values);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    // identify cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    ESP_ERROR_CHECK(esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &null_values));
    // group cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    ESP_ERROR_CHECK(esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &null_values));
    // scenes cluster create with standard cluster + customized
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_SCENES);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_CURRENT_GROUP_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_CURRENT_SCENE_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_SCENE_VALID_ID, &null_values);
    esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_SCENE_COUNT_ID, &null_values);
    // on-off cluster create with standard cluster config
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    // create cluster lists for this endpoint
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));

    esp_zb_endpoint_config_t EPC = {
        .endpoint=HA_ESP_LIGHT_ENDPOINT,
        .app_profile_id=ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id=ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID, .app_device_version=1,
    };

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    // add created endpoint (cluster_list) to endpoint list
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, EPC);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    // Erase NVRAM before creating connection to new Coordinator
    // Comment out next line to erase NVRAM data if you are connecting to new Coordinator <----------------------------
    // esp_zb_nvram_erase_at_start(true);

    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

lienbacher, 2024-06-07:
Hello Gentlemen, 
I have just spent a few hours unsuccessfully trying to modify the code of the color light example using all code snippets in all these threads to solve the problem with manufacturer reported. 

This issue would be a none issue if the example would be written so that it would provide espressif as manufacturer and the example name from the start to comply with the zigbee standard right away. I do not quite understand why this is not the case? Shouldn't the goal be to get things started quick and easy? Instead of having nicely working code, we have to hack some code snippets that do almost the same thing but differently. I do not think this should be the goal here. 

The snippet OOHehir posted above does not work for me. The device builds, flashes and starts up, but it wouldn't start to interview with my zigbee network. The default version of the example does, but with undefined model and manufacturer.

lienbacher, 2024-06-07:
> @SCiunczyk First, It can be define any value which you want if you do it as the [Support new devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html) step. [...]

I find it interesting that you are trying to solve the problem in z2m, when the problem we describe is in the example code not providing the information that z2m needs in order to create a converter for the device. It seems like there is a misunderstanding on your side?

lienbacher, 2024-06-08:
Update: After a full flash erase things are finally working on my end, seems most snippets i pieced together would've worked.

lhespress, 2024-06-11:
@lienbacher That's because of Zigbee2MQTT uses [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) which don't support `espressif` as manufacturer, so we don't provide it on example as default.

SCiunczyk, 2024-06-11:
@lienbacher I agree with you, and what surprises me even more, ZHA integration works without any problem with these H2/C6 boards. With Z2M I still have no success :-(

0rion12, 2024-06-11:
Hi all, 

I have been following this thread for a few days to try to get my esp32-c6 board to work with my home automation and Z2M. 

By following the various tips for adding a manufacturer and a model ID, I managed to get my esp32 to join my zigbee network. I am able to turn the LED on my esp32 on and off with the ON/OFF switch located in the "expose" tab.

[image]

Now I would like to do more advanced things and be able to customize the example code a little. For example, I would like to be able to manage the intensity. For this, in the esp32 code, I added the "level_cluster" cluster. 
This is well recognized under Z2M because in the features tab, I can clearly see the "genLevelCtrl" cluster added. 

[image]

However, I don't know how to act on it. In the "expose" tab, there is nothing. I tried to use the console dev, but all the fields are read only. Besides, I can read the value of the new cluster using the console, but not write a value because of the write-only mode.

I tried adding a converter, but it didn't work.
Here is my device informations: 
[image]

I created a file named "TEST1.js" at the location "/homeassistant/zigbee2mqtt/TEST1.js".

The file looks like this : 
```
const {state, brightness} = require('zigbee-herdsman-converters/lib/modernExtend');
// No other imports are necessary.

const definition = {
    zigbeeModel: ['TEST1'],
    model: 'TEST1',
    vendor: 'ESPRESSIV',
    description: 'MiJia temperature & humidity sensor',
    extend: [
      // Modern extends are incapsulating fromZigbee, toZigbee, exposes and more.
      state(),
      brightness(),
    ],
};

module.exports = definition;
```

And I also modified my zigbee2mqtt "configuration.yaml" file to add this at the end : 
```
external_converters:
  - TEST1.js
```

But that hasn't changed anything, I have the impression that Z2M does not make the link between my file and my device.
 I think I'm not doing things right, but I can't figure out how to do it. Can you help me please ?

Thank you !

lienbacher, 2024-06-11:
@0rion12 this has nothing to do with the issue you are commenting on, this is not a thread in a forum, it is an issue in a repository. It discusses an issue and it has a status, which in this case is "Open". 

Apart from this you are also looking for help that is much more related to zigbee2mqtt and not the espressif zigbee repository as the people reponsible for this repository have little to no experience with zigbee2mqtt. It's a bit like asking for help with baking bread in a factory for ovens.

lienbacher, 2024-06-11:
> @lienbacher I agree with you, and what surprises me even more, ZHA integration works without any problem with these H2/C6 boards. With Z2M I still have no success :-(

Well zha working is less a surprise because it seems to be less strict about those two clusters being present, z2m uses them to identify which converter to use. 

OOhenirs code worked or me, make sure you completely delete your devices flash before trying again, that was the problem for me after all.

SCiunczyk, 2024-06-12:
To erase flash I'm using the VSC plug-in. It's correct way?

[image]

lienbacher, 2024-06-12:
Yes, or 'idf.py erase-flash'

lienbacher, 2024-06-12:
So you say the device joins but does not pair. What exactly do you mean by that, and why do you show the device map and not about page of the device?

SCiunczyk, 2024-06-13:
Hi @lienbacher . The steps I'm going are:

1. Open Zigbee2MQTT web interface

[image]

2. Delete espressif board (from previous try)

[image]

3. Verify if is deleted:

[image]

4. In VSC Full Clean / Build / Flash Erase / Flash board 

5. Disconnect board from USB

6. Click "Allow to connect..." in HA/Z2M

[image]

7. Connect board to PC USB

[image]

8. Check devices list:

[image]

9. Device page:

[image]

10. and map:

[image]

Environment: Windows 11 / Microsoft Surface Book 2 / ConBee III coordinator / HA v. 2024.6.2 / espressif H2 / C6 from Aliexpress / esp_zb_task code snippet upladed bby @OOHehir / SDK v1.2.3

BTW I feel like I'm banging my head against a wall :-)

SCiunczyk, 2024-06-13:
here is the log screeneshot:

[image]

Look that interview failed.

> Also, if can you switch your z2m temporarily to english by simply clicking the flag in the top right and select english, it's easier for us to see whats going on.

Silly me. I miss the flag icon. Sorry

lienbacher, 2024-06-13:
that error message is a very important piece in the puzzle. If you switch debug logging on in your z2m you will likely get more information on why the interview failed. but the reason is likely in your esp_zb_task. Can you post the current code you are using?

SCiunczyk, 2024-06-13:
[Yes Master](https://youtu.be/bGhshJ4c2Mw)

Debug logging:

[image]

Code I'm compiling:

```

/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * Zigbee HA_on_off_light Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */
#include "esp_zb_light.h"
#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light (End Device) source code.
#endif

static const char *TAG = "ESP_ZB_ON_OFF_LIGHT";
/********************* Define functions **************************/
static esp_err_t deferred_driver_init(void)
{
    light_driver_init(LIGHT_DEFAULT_OFF);
    return ESP_OK;
}

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, , TAG, "Failed to start Zigbee commissioning");
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Initialize Zigbee stack");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Deferred driver initialization %s", deferred_driver_init() ? "failed" : "successful");
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
[... 130 lines cut ...]
    // add created endpoint (cluster_list) to endpoint list
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, EPC);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    // Erase NVRAM before creating connection to new Coordinator
    // Comment out next line to erase NVRAM data if you are connecting to new Coordinator <----------------------------
    // esp_zb_nvram_erase_at_start(true);

    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}
```

SCiunczyk, 2025-06-19:
Yes my C6 is working with z2m, but I'm using XIAO ESP32C6 board with examples provided by Seed Studio. They cover some complexity of "pure" zigbee libraries  by Espressif. Also I replaced unsupported Conbee III to SMLIGHT SLZB-MR1...

xieqinan, 2025-09-30:
Hi @SCiunczyk,

Do you have any other questions that need to be discussed regarding this issue? Please let me know your status.

SCiunczyk, 2025-10-06:
At the moment I have no questions.