espressif/esp-zigbee-sdk #329: APS pkt 12 for ep 1 is not recognized - drop         Why?

vritzka, 2024-04-22:
### Question

Hello,

I am running **esp_zigbee_gateway** on ESP32s3 + **ot_rcp** on ESP32H2.

As soon as I join a device I get this error:

ZBOSS: zdo/af_rx.c:463   APS pkt 18 for ep 1 is not recognized - drop    
ZBOSS: zdo/af_rx.c:463   APS pkt 3 for ep 1 is not recognized - drop   
ZBOSS: zdo/af_rx.c:463   APS pkt 6 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 12 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 15 for ep 1 is not recognized - drop

.... and more lines like this.

I am joining a switch or a temperature sensor .... both cause this error.

Those switch or temp sensor work well when I run esp_zigbee_host + esp_zigbee_rcp

What do those errors mean?

Thanks

### Additional context.

`I (513) spi_flash: flash io: dio
W (517) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (530) sleep: Configure to isolate all GPIO pins in sleep state
I (537) sleep: Enable automatic switching of GPIO sleep configuration
I (544) main_task: Started on CPU0
I (554) main_task: Calling app_main()
I (574) ESP_RADIO_SPINEL: spinel UART interface initialization completed
I (574) ESP_RADIO_SPINEL: Spinel UART interface has been successfully enabled
I (574) main_task: Returned from app_main()
I(584) OPENTHREAD:[I] P-RadioSpinel-: RCP reset: RESET_POWER_ON
I(594) OPENTHREAD:[I] P-RadioSpinel-: Software reset RCP successfully
ZBOSS: common/zb_init_default.c:183   FFD build
ZBOSS: common/zb_init_default.c:187   sizes: g_zb 7552 sched 224 bpool 16 nwk 1032 aps 3048 addr 284 zdo 840
ZBOSS: common/zb_init_default.c:196   sec 2
ZBOSS: common/zb_init_default.c:198   zcl 600
ZBOSS: common/zb_init_default.c:201   zll 960
ZBOSS: common/zb_init_default.c:207   nvram 140
ZBOSS: common/zb_init_default.c:210   buttons 200
ZBOSS: common/zb_init_default.c:212   err_ind 4
ZBOSS: common/zb_init_default.c:218   scheduler q size 80
ZBOSS: common/zb_init_default.c:221   g_mac 824 g_imac 28
ZBOSS: common/zb_init_default.c:225   Configurable mem build, use ZBOSS lib defaults = 0
ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 80 ZB_NWK_IN_Q_SIZE 40 ZB_MAC_PENDING_QUEUE_SIZE 20 ZB_APS_DST_BINDING_TABLE_SIZE 16 ZB_APS_BIND_TRANS_TABLE_SIZE 20 ZB_N_APS_RETRANS_ENTRIES 27
ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 64 ZB_IEEE_ADDR_TABLE_SIZE 80 ZB_NEIGHBOR_TABLE_SIZE 64 ZB_NWK_ROUTING_TABLE_SIZE 64 ZB_APS_DUPS_TABLE_SIZE 32
I (704) ESP_ZB_GATEWAY: Running RCP Version: openthread-esp32/fcb76f0671-456c44828; esp32h2;  2024-04-22 12:47:02 UTC
ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
I (724) ESP_ZB_GATEWAY: Production configuration is ready
W (734) ESP_ZB_GATEWAY: Production configuration is not present
I (744) example_connect: Start example_connect.
I (744) pp: pp rom version: e7ae62f
I (754) net80211: net80211 rom version: e7ae62f
I (764) wifi:wifi driver task: 3fca7bdc, prio:23, stack:6656, core=0
I (764) wifi:wifi firmware version: a0511a3
I (764) wifi:wifi certification version: v7.0
I (764) wifi:config NVS flash: enabled
I (774) wifi:config nano formating: disabled
I (774) wifi:Init data frame dynamic rx buffer num: 32
I (784) wifi:Init static rx mgmt buffer num: 5
I (784) wifi:Init management short buffer num: 32
I (794) wifi:Init dynamic tx buffer num: 32
I (794) wifi:Init static tx FG buffer num: 2
I (794) wifi:Init static rx buffer size: 1600
I (804) wifi:Init static rx buffer num: 10
[... 81 lines cut ...]
ZBOSS: zdo/af_rx.c:463   APS pkt 18 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 3 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 6 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 12 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 15 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 20 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 4 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 9 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 14 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 18 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 3 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 6 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 12 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 15 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 20 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 4 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 9 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 13 for ep 1 is not recognized - drop
ZBOSS: zdo/af_rx.c:463   APS pkt 16 for ep 1 is not recognized - drop
W (255794) ESP_ZB_GATEWAY: Network(0xc543) closed, devices joining not allowed.
I (1084244) example_connect: Wi-Fi disconnected, trying to reconnect...
I (1086654) example_connect: Wi-Fi disconnected, trying to reconnect...
I (1089064) example_connect: Wi-Fi disconnected, trying to reconnect...
I (1091474) example_connect: Wi-Fi disconnected, trying to reconnect...
`

xieqinan, 2024-04-24:
@vritzka ，

If you haven't modified the original gateway example, it doesn't create any endpoints. Consequently, commands sent from the temperature sensor to the gateway will be dropped.

If you add an endpoint to the gateway, you should double-check the command sent by the temperature sensor. If the command isn't directed to the existing endpoints of the gateway, it will also be dropped.

chshu, 2024-06-12:
@vritzka please reopen if any follow up questions.

closed 2024-06-12 by chshu