espressif/esp-zigbee-sdk #331: Make an End Device join a Router after Coordinator power loss

HOSTEER, 2024-04-25:
### Question

Hello, I'm trying to connect an End Device to a Router in a specific case. I want to be sure that an End Device can join a Router. 
To do so I use a Coordinator to startup the PAN, then I let the router connect itself to the PAN and I power off the Coordinator.
After this I turn on an End Device and I'm expecting it to connect itself to the Router, but it's not the case.

- On the zigbee sniffer I'm only getting router's zigbee frame after coordinator power off and End device turned on
- On the End Device the esp-idf monitor only display `Stack BDB Steering failure with ESP_FAIL status, steering`

xieqinan, 2024-04-26:
@HOSTEER ，

If the end device is factory new, it cannot join a network directly with the coordinator because it needs certification from the coordinator first. Could you clarify what you mean by "making an End Device join a Router"? Are you suggesting that the end device should become a child node of the router? Or do you mean that the new end device can join the router without involving the coordinator? Alternatively, are you suggesting that the end device should operate solely with the router?

If the third option aligns with your intention, I recommend conducting a test where the router connects itself to the PAN (Personal Area Network) and then connects the end device to the PAN as well. After that, power off all devices and then power on the router and end device again to observe their behavior.

HOSTEER, 2024-04-26:
> Are you suggesting that the end device should become a child node of the router? Or do you mean that the new end device can join the router without involving the coordinator?

I mean that the new end device can join the router without involving the coordinator. 

> If the end device is factory new, it cannot join a network directly with the coordinator because it needs certification from the coordinator first.

Did you want to write "it cannot join a network directly with**out** the coordinator" ? 

Also tried the third option test and what I saw was that the end device was able to connect to the router even if the coordinator was offline.

xieqinan, 2024-04-26:
@HOSTEER ,

> Also tried the third option test and what I saw was that the end device was able to connect to the router even if the coordinator was offline.

If the network is distributed, I believe it can be done.

HOSTEER, 2024-04-26:
Also, is there a way to know to which router an end device is connected to ? 
Code wise or through a zigbee sniffer ?

xieqinan, 2024-04-28:
@HOSTEER ,

The Zigbee sniffer will offer a more intuitive way to visualize the network topology. But we plan to incorporate related API functionality for network topology into the esp-zigbee-sdk in the future.

closed 2024-04-29 by HOSTEER