espressif/esp-zigbee-sdk #332: How can I join an esp32c6 to an MSP Zigbee Network

TomPrenderville, 2024-05-01:
### Question

Hello,
I have a Zigbee implementation dating back to before the Zigbee Profiles were designed. It is running a Manufacturer Specific Profile - we have our own Trust Centre Link Key, and we use Cluster IDs as Message IDs and build our own Unicast Messages for communication within the network.

How can I get an esp32c6 to join this network, and then send or receive our own unicast messages. We would effectively be bypassing the Zigbee Cluster Library. 

The network itself operates with the Coordinator/Trust Centre operating as a "Sink" for data messages from "Sensor" nodes which join the network. Sensors can operate as Routers or Sleepy End Devices.

Many Thanks
Tom

xieqinan, 2024-05-07:
@TomPrenderville ，

Regarding the issue, I believe there are two questions that need to be addressed. The first question is how the Zigbee device can join a network using its own TC link key. The second question is how messages can be handled within a custom manufacturer profile and cluster ID. Is that correct? If so, the first question can be resolved by using `esp_zb_secur_TC_standard_preconfigure_key_set()` to set the same TC link key as the coordinator, or by using `esp_zb_secur_link_key_exchange_required_set(false)` to disable the link key exchange. For the second question, the esp-zigbee-sdk provides manufacturer profile settings and a custom cluster handler for users, which is helpful. I believe we need further discussion to address the specific cases.

TomPrenderville, 2024-05-07:
@xieqinan ,
Many thanks for the reply. You are correct in the two questions that need to be addressed! 
For now I am struggling to set the link key with the esp_zb_secur_TC_standard_preconfigure_key_set() function you mentioned. 
Once I have that working I will need further information on creating the custom cluster handlers and the manufacturer profile settings. Are there examples or documents I can look at?

xieqinan, 2024-05-08:
@TomPrenderville ,

The ESP-Zigbee-SDK currently does not provide an example for custom clusters. However, you can refer to issues on the ESP-Zigbee-SDK GitHub repository for assistance, such as https://github.com/espressif/esp-zigbee-sdk/issues/202#issuecomment-2005690377. Additionally, the [zcl docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/zcl/esp_zigbee_zcl_command.html) may not be applicable to your needs.

If you encounter any questions or issues with the ESP-Zigbee-SDK while implementing your project, feel free to discuss them here.

TomPrenderville, 2024-05-09:
Hello @xieqinan ,

I have tried using the esp_zb_secur_TC_standard_preconfigure_key_set() in a number of locations in the sample HA_on_off_light project but I can't get it to work. It always joins a HA network rather than the network with the alternative Link Key.

I have erased the flash on the HA Light device so any previous keys are removed and it starts in "Factory Reset" mode and is looking for a network. I have two coordinators, one with the HA Key (ZigbeeAlliance09) and another with a custom key. When I specify the custom key in the HA_on_off_light project using esp_zb_secur_TC_standard_preconfigure_key_set(), it continues to join the coordinator with the HA Key.

Perhaps I'm using the function in the wrong place ... 
I have tried just after the esp_zb_init() function, Then I tried just before the esp_zb_start() function, and I've also tried in the esp_zp_app_signal_handler just before esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING); 

In each case it won't join Custom Key network and continues to join the HA network. I have made sure to erase the flash on the device before each attempt so it doesn't remember the old network settings and each time it gets a new short address so I'm sure it's joining again rather than starting up on the previous network.

Any help appreciated.
Tom

TomPrenderville, 2024-05-09:
Hello again. For no particular reason, the device joined the network with the alternative Key.

Flori1007, 2024-05-10:
@TomPrenderville can you please share your config files and the way how you get that running? I Try VSCODE and ESP IDF and also ESPHOME on HomeAssistant to get zigbee and an esp32C6 running and connecting to my existing zigbee devices.... I dont get it how to get that set up. Would be very nice :)

xieqinan, 2024-05-11:
@TomPrenderville 

> Hello again. For no particular reason, the device joined the network with the alternative Key.

The device will autonomously select the better parent to join during the beacon request process. However, you can use `esp_zb_set_extended_pan_id()` to specify the extended PAN ID for the device to join. This approach allows you to verify whether `esp_zb_secur_TC_standard_preconfigure_key_set()` takes effect as expected.

TomPrenderville, 2024-05-13:
Hello @Flori1007 ,
I'm not sure exactly what you're looking for. However, I am simply using the HA_on_off_light example that can be copied from the Espressif samples in VS Code and I want it to join one of our networks with a different Security Key. So, I simply needed to change the Trust Center Link Key, also called the Preconfigured Key in the esp_zb_task function before starting the Stack with esp_zb_start. The Key shown below is as an example only. Remember to erase the exp32c6 before programming so it forgets any previous network and starts the Network Steering once the stack is up and running. Also, remember to open the network on the Coorinator/Trust Center to allow new devices to join that network.

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    esp_zb_device_register(esp_zb_on_off_light_ep);

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

     // Our Key
    uint8_t key[16] =  { 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F, 0x10};
    // HA Key
    // uint8_t key[16] =  { 0x39, 0x05, 0x65, 0x54, 0x30, 0x23, 0x44, 0x09, 0x83, 0x84, 0x22, 0x70, 0x76, 0x36, 0x47, 0x37 };
    esp_zb_secur_TC_standard_preconfigure_key_set(key);
    ESP_LOGI(TAG, "Preconfigured Key: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x",
                key[0], key[1], key[2], key[3], key[4], key[5], key[6], key[7], 
                key[8], key[9], key[10], key[11], key[12], key[13], key[14], key[15]);

    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

xieqinan, 2024-06-12:
@TomPrenderville ,

Are there any other topics that need to be discussed here? If not, please consider closing this issue.

closed 2024-09-29 by chshu