espressif/esp-zigbee-sdk #333: Calling a cluster specific command

Andelarrr, 2024-05-02:
### Is your feature request related to a problem?

Hello,

Our device is a client of a cluster (0x0800) that is not supported by the library at this time, I am looking for a way to call a cluster specific command for this cluster.

I would need this request to have:
- Frame type: 1 cluster-specific
- Manufacturer specific: 0
- Direction: 0 client to server
- Command: 0x00 (we will have other values too in the future)

Please advise how I can make such a call.

Thanks!

### Describe the solution you'd like.

I guess ideal would be to have something like the esp_zb_zcl_custom_cluster_cmd_req().

### Describe alternatives you've considered.

I was looking at ESP_ZB_CORE_CMD_PRIVILEGE_COMMAND_REQ_CB_ID, but I did not see how could I use that on the client side.
I was also thinking about calling some raw methods, maybe that is possible, I could not figure it out though.

xieqinan, 2024-05-07:
@Andelarrr,

The cluster id 0x0800 corresponds to the "key establishment" cluster, which is temporarily unsupported in the library. Therefore, if you intend to handle commands from the "key establishment" cluster, you will need to rely solely on the APS APIs. However, if your goal is only to send commands from this cluster, I believe that using `esp_zb_zcl_custom_cluster_cmd_req()` would be valid. You can try using the following code snippet for experimentation. It's advisable to use a sniffer to verify whether the command is structured correctly.

```
    uint16_t key_establishment_suite = 0x0001;
    uint8_t ephemeral_data_generate_time = 0;
    uint8_t confirm_key_generate_time =0;
    uint8_t value[4] = {};
    memcpy(value, &key_establishment_suite, sizeof(uint16_t));
    memcpy(value + 2, &ephemeral_data_generate_time, sizeof(uint8_t));
    memcpy(value + 3, &confirm_key_generate_time, sizeof(uint8_t));

    esp_zb_zcl_custom_cluster_cmd_req_t cmd_req = 
    {
        .zcl_basic_cmd.src_endpoint = 1,
        .zcl_basic_cmd.dst_endpoint = 1,
        .zcl_basic_cmd.dst_addr_u.addr_short = 0x0000,
        .address_mode =  ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .cluster_id = 0x0800,
        .profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI,
        .custom_cmd_id = 0x00,
        .data = {ESP_ZB_ZCL_ATTR_TYPE_SET, sizeof(value), value},
    };
    esp_zb_zcl_custom_cluster_cmd_req(&cmd_req);
```

Andelarrr, 2024-05-08:
@xieqinan,

Thanks for the help, this is indeed working!

I have now a crash a few seconds after the exchange, but this is another story. :-)

closed 2024-05-08 by Andelarrr