espressif/esp-zigbee-sdk #334: Custom data package send

LorenzoGrandi7, 2024-05-02:
### Question

Goodmorning, I'm trying to check the performance of the ESP32-C6, and in particular I'm studying the Zigbee protocol. The setup I'm working on are two controller, one acting as a coordinator and the other as an endpoint, that transfer a data package of 120kB from the endpoint to the coordinator (virtually as a generic sensor that sends its data to a coordinator). Looking to the examples I see that it's easy to send single values, but I haven't found a way to send a stream of data. I'm attepting to send the package using esp_zb_zcl_write_attr_cmd_req, but I'm not sure it's the correct way to proceed. Any suggestion?

### Additional context.

End point code:
```

#include <stdio.h>

#include "esp_zb_zed.h"
#include "switch_driver.h"

#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile sensor (End Device) source code.
#endif

static const char *TAG = "ESP_ZB_ZED";

typedef struct rx_device_params_s {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t  endpoint;
    uint16_t short_addr;
} rx_device_params_t;

typedef struct zdo_info_ctx_s {
    uint8_t endpoint;
    uint16_t short_addr;
} zdo_info_user_ctx_t;

typedef struct esp_zb_data_pkg_cluster_cfg_s {
    uint32_t data;
} esp_zb_data_pkg_cluster_cfg_t;

typedef struct esp_zb_package_cfg_s  {
    esp_zb_basic_cluster_cfg_t basic_cfg;
    esp_zb_identify_cluster_cfg_t identify_cfg;
    esp_zb_data_pkg_cluster_cfg_t data_cfg;
} esp_zb_package_cfg_t;

rx_device_params_t rx_device;

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

static void esp_app_buttons_handler(switch_func_pair_t *button_func_pair) {
    uint32_t data_package[VECTOR_SIZE];
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {

    for (uint32_t i = 0; i < VECTOR_SIZE; i++) {
        data_package[i] = i;
    }

        /* Send report attributes command */
        esp_zb_zcl_write_attr_cmd_t write_req;
        write_req.zcl_basic_cmd.dst_addr_u.addr_short = rx_device.short_addr;
        write_req.zcl_basic_cmd.dst_endpoint = rx_device.endpoint;
        write_req.zcl_basic_cmd.src_endpoint = ESP_TX_ENDPOINT;
        write_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        write_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ANALOG_VALUE;
        write_req.attr_field->data.value = data_package;
[... 482 lines cut ...]
    /* Create customized thermostat endpoint */
    esp_zb_package_cfg_t package_cfg = ESP_ZB_DEFAULT_CONFIGURATION_TOOL_CONFIG();
    esp_zb_ep_list_t *esp_zb_device_ep = custom_thermostat_ep_create(ESP_RX_ENDPOINT, &package_cfg);

    /* Register the device */
    esp_zb_device_register(esp_zb_device_ep);

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

```

xieqinan, 2024-05-07:
@LorenzoGrandi7 ,

This issue is similar to the one discussed in this GitHub link: https://github.com/espressif/esp-zigbee-sdk/issues/202. I recommend referring to it first for further insights.

kgkask, 2024-06-07:
@LorenzoGrandi7 could you please provide the full code, I mean the .h file as well. I am trying convert the code to Arduino IDE and do some customized modification. 
or mainly the function `esp_zb_analog_value_create()` since I was trying before to create a structure attribute and I couldn't succeed.

LorenzoGrandi7, 2024-06-07:
> @LorenzoGrandi7 could you please provide the full code, I mean the .h file as well. I am trying convert the code to Arduino IDE and do some customized modification. or mainly the function [...]
.
The header file are the same provided in the examples of the sdk library, I made no modifications on them. Still the code isn't working, so I suggest you to check the linked discussed which is more complete and provides a working solution (even if I didn't managed to make mine working with it).

kgkask, 2024-06-07:
@LorenzoGrandi7 Thanks for quick reply, 

I have seen that one and even convert it, but I want to make a single structure attribute contains all information I want to report from the EP, so basically looking for a way to create custom attribute and handle it. 

anyway thanks for your feedback.

kgkask, 2024-06-07:
I have question but I don't to initiate a thread, is it possible to make a packet (structure attribute) with multiple set of data like uint8_t, char, string and more, and then make it reportable ? like each time the EP send the full packet.

I know I need to make a custom cluster with its own attribute.

chshu, 2024-09-29:
> I have question but I don't want to initiate a thread, is it possible to make a packet (structure attribute) with multiple set of data like uint8_t, char, string and more, and then make it reportable ? like each time the EP send the full packet.
> 

Yes, you can define the data structure format per your request in [custom cluster](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_custom.html).

Please reopen if any follow up questions.

closed 2024-09-29 by chshu