espressif/esp-zigbee-sdk #335: Zigbee gateway coordinator doesn't associate any device because doesn't send association response command

nalves23, 2024-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### esp-zigbee-lib version.

1.3.1

### esp-zboss-lib version.

1.3.1

### Espressif SoC revision.

ESP32-S3

### What is the expected behavior?

It was expected that zigbee router devices (with ESP zigbee stack) join with success my zigbee gateway!

### What is the actual behavior?

I can't associate any zigbee device to coordinator since Zigbee-SDK and ZBoss 1.0.2 version.
The first issue detection was detected on the version 1.2.3 of Zigbee-sdk and zboss-lib.

In the wireshark analyser I can check that zigbee router device start the beacons request to search and join the network (my zigbee gateway is the only network available and it is in permit association mode). 
When the zb_router receive the beacon response from my gateway start the commissioning process, send the association request message but my coordinator doesn't send the Association response as it is supposed.

Has I said previously, my gateway was working (same hardware and software) since the last version os 1.0.2 of zigbee SDK and zboss lib. 

### Steps to reproduce.

I have one ESP Thread Border Router/Zigbee Gateway Board and I follow all the procedure to create a zigbee gateway procedure in  https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_gateway/README.md page.

I program the example code ot_rcp example code in ESP32-H2 module and the Zigbee Gateway example in the ESP32-S3 as in instructions of the example. Any change was made on menuconfig as is referee in the README file.

### More Information.

Version 1.2.3 has been out for a while and I'm worried that no one has checked this issue yet.

[image]

diazmanuel, 2024-05-02:
Hello, we are possibly having the same problem, in our case we jumped from version 1.2.2 to 1.3.0 and 1.3.1 and we detected errors in the pairing of the devices. The coordinator responds that it was able to add the end devices but these, in most cases, do not finish connecting. This problem started since we jumped from version 1.2.2 to 1.3.0. We couldn't analyze the plot with wireshark but we think it is due to the same issue.

xieqinan, 2024-05-07:
@diazmanuel @nalves23 ,

Have you configured [OPENTHREAD_NCP_VENDOR_HOOK](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_gateway/README.md#create-the-rcp-firmware-image) on the RCP side? I've just successfully used [dimmable_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light) as the router for a test.

Could you please confirm which Zigbee router type was used in the test? The CI of esp-zigbee-sdk has tested the gateway item, so I believe it should be fine.

nalves23, 2024-05-07:
@xieqinan 

The suggestion you made worked perfectly. In fact, the joining process wasn't working because of this configuration, and I couldn't understand it from the beginning because it's not documented.

Kind Regards.

closed 2024-05-07 by nalves23