espressif/esp-zigbee-sdk #336: Connected LED

BaWe155, 2024-05-03:
### Question

I want an indication led, that indicates when a Zigbee node is connected with coordinator. Led must off by a restart and on by a connection.
Were in the code can I call the led function?

HOSTEER, 2024-05-03:
If you use led's functions provided by esp-zigbee-sdk in light_driver.c, the way you can do this is by initializing the LED in the beggining of `app_main()` 

```c
void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    light_driver_init(0);
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    switch_driver_init(button_func_pair, PAIR_SIZE(button_func_pair), esp_zb_buttons_handler);
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
    xTaskCreate(temp_transmit, "temp_transmit_main", 4096, NULL, 4, NULL);
}
``` 
Then use `light_driver_set_color_RGB()` in the `esp_zb_app_signal_handler()` to change led color and intensity. For example if you want to know when the device is connected or disconnected you can put the functions here.
```c
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status != ESP_OK) {
            connected = false;
            ESP_LOGW(TAG, "Stack %s failure with %s status, steering",esp_zb_zdo_signal_to_string(sig_type), esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
            light_driver_set_color_RGB( 1, 0 0);
        } else {
            /* device auto start successfully and on a formed network */
            connected = true;
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
                    extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                    extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                    esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());
            light_driver_set_color_RGB( 0, 1, 0);
        }
        break;
``` 

I'm not Espressif's employee, so there is maybe a better way to do this

BaWe155, 2024-05-06:
Hello,

I tested it, the led goes on for one time, only by a factory new.

    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK)
        {
            set_connected_led(true);

            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            esp_zb_ieee_addr_t Macaddr;
            esp_zb_get_long_address(Macaddr);
            ESP_LOGI(TAG, "----------------------------------------------------------------------------");
            ESP_LOGI(TAG, "LZB node joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel());

            ----------------------------------------------------------------");

            // Connected, bind
            bind_to_coordinator();
        }
        else
        {
            set_connected_led(false);
            ESP_LOGI(TAG, "Network steering was not successful (status: %s)", esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;

xieqinan, 2024-05-07:
@BaWe155 ，

The ESP-Zigbee-SDK examples use the `ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");` statement to inform the user about the device state. Does this meet your requirements?

BaWe155, 2024-05-07:
@xieqinan 
 When a Zigbee node is connected with coordinator a LED must on and of by a restart or no connection must the LED off

I believe the ESP_ZB_BDB_SIGNAL_STEERING case is not the right place.

xieqinan, 2024-05-07:
@BaWe155 

> I believe the ESP_ZB_BDB_SIGNAL_STEERING case is not the right place.

Yes, but I think the below code can meet your requirement.
```
void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        light_driver_init(0);
        ESP_LOGI(TAG, "Initialize Zigbee stack");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Deferred driver initialization %s", deferred_driver_init() ? "failed" : "successful");
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                light_driver_init(0);
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                light_driver_init(1);
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
            light_driver_init(0);
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());
            light_driver_init(1);
        } else {
            light_driver_init(0);
            ESP_LOGI(TAG, "Network steering was not successful (status: %s)", esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
    case ESP_ZB_ZDO_SIGNAL_LEAVE: {
        esp_zb_zdo_signal_leave_params_t *leave_params = NULL;
        leave_params = (esp_zb_zdo_signal_leave_params_t *)esp_zb_app_signal_get_params(p_sg_p);
        if (err_status == ESP_OK) {
            light_driver_init(0);
        }
        if (leave_params->leave_type == ESP_ZB_NWK_LEAVE_TYPE_RESET) {
            ESP_LOGI(TAG, "Reset device");
        }
    } break;
    case ESP_ZB_NWK_SIGNAL_PERMIT_JOIN_STATUS:
        if (err_status == ESP_OK) {
            if (*(uint8_t *)esp_zb_app_signal_get_params(p_sg_p)) {
                ESP_LOGI(TAG, "Network(0x%04hx) is open for %d seconds", esp_zb_get_pan_id(), *(uint8_t *)esp_zb_app_signal_get_params(p_sg_p));
            } else {
                ESP_LOGW(TAG, "Network(0x%04hx) closed, devices joining not allowed.", esp_zb_get_pan_id());
            }
        }
        break;
    default:
        ESP_LOGI(TAG, "ZDO signal: %s (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type, esp_err_to_name(err_status));
        break;
    }
}
```

BaWe155, 2024-05-07:
@xieqinan 
Thank you, It almost works.

But when a node has made a join with the coordinator the LED goes on. Ather that I switch off the coordinator and do a power cycle by the Zigbee node. The connection LED goes on, it can’t have a connection on that moment because the coordinator is switched off.

The connection LED must only ON when there is a real connection with an online coordinator. So when the LED is on I know there is a proper and real connection.

xieqinan, 2024-05-08:
@BaWe155 

> The connection LED goes on, it can’t have a connection on that moment because the coordinator is switched off.

The end device requires time to confirm whether the coordinator is offline, so it is not possible for the LED of the end device to switch off immediately when the coordinator goes offline. The confirmation time is approximately 30 seconds (with data requests every 3 seconds for a total of 10 attempts). Refer to `ED_KEEP_ALIVE`.

BaWe155, 2024-05-08:
@xieqinan I use a router not an end device. After 60sec (coordinator off) LED is still on. When I do a power cycle without a coordinator (switched off) LED on the router goes on.

xieqinan, 2024-05-08:
@BaWe155 ,

I believe we should clarify the issue. You are expecting an LED on a node connected to the coordinator to indicate the state of the coordinator. However, it's important to note that the router operates somewhat independently of the coordinator in a Zigbee network. The coordinator functions as both a router and a trust center. When the coordinator powers off, it means that other devices cannot join the network and there is a loss of part of the routing information.

However, Zigbee routers in the network have routing capabilities that allow them to maintain the original links even without direct coordination from the coordinator. This means that routers and the coordinator are relatively independent, and the coordinator powering off does not necessarily result in network paralysis.

Therefore, routers and the coordinator only share a communication link and do not have a dependent relationship. When the coordinator powers off, routers will not be affected in application.

xieqinan, 2024-05-09:
@BaWe155 ，

> Thank you for your explanation I understand it, we have an application with a control bridge for every action we need the control bridge.

Regarding the issue, I need to know more details. What are the primary functions of a control bridge? Additionally, what actions does the control bridge monitor, only concerning the state of the coordinator?

BaWe155, 2024-05-14:
@xieqinan 
Thank you, for now the function of the LED is enough

closed 2024-05-14 by BaWe155