espressif/esp-zigbee-sdk #337: Adding HA Extended Color Light to the ZigBee example projects

mjdswan, 2024-05-04:
### Is your feature request related to a problem?

We are developing a ZigBee color light using ESP32C6, which for commercial reasons must join a SONOFF ZigBee Bridge network.  So far I cannot get your HA color light example to join.

  

### Describe the solution you'd like.

According to my tests using your example projects while sniffing the joining sequence, I am able to join the SONOFF bridge when using the example **HA_on_off_light** and it works fine.  However if I try to use the example **HA_color_dimmable_light**, the SONOFF sends a Leave request about a half second after we send it our Simple Descriptor response (ack'd) for the endpoint.
I cannot see a reason for that other than the bridge not supporting device type 0x0102, although it's a bit of an educated guess.

I also have another commercial ZigBee light device which joins this SONOFF bridge successfully as device type 0x010D (Extended color light).

### Describe alternatives you've considered.

I wonder if it would be a simple task for you to add this 0x010D device as an example project.
Alternatively, if there is a way to manually configure the endpoint descriptor as such via the API, I'll be happy to hear it.

xieqinan, 2024-05-07:
@mjdswan，

Could you please give the following code a try?

```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZR_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_color_dimmable_light_cfg_t light_cfg = ESP_ZB_DEFAULT_COLOR_DIMMABLE_LIGHT_CONFIG();
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = HA_COLOR_DIMMABLE_LIGHT_ENDPOINT,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = 0x010D,
        .app_device_version = 0,
    };
    esp_zb_ep_list_add_ep(ep_list, esp_zb_color_dimmable_light_clusters_create(&light_cfg), endpoint_config);
    esp_zb_device_register(ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

mjdswan, 2024-05-08:
Hi @xieqinan,
Thanks a lot, your sample works and the board joins the SONOFF bridge now.
I have been exploring the examples more and I think I understand enough to create what I need.
Thank you for the prompt and excellent help.

Regards,
Mark.

closed 2024-05-08 by mjdswan