espressif/esp-zigbee-sdk #338: information about zigbee color modes color capabilities

ilker-aktuna, 2024-05-05:
### Question

I am trying to use C6 devkit as a zigbee dimmable light bulb.
I can successfully power on/off the led , and set level
But when I try to change color modes, it fails.
I am trying to control the device from my zigbee controller (Hubitat Hub)
I am not sure which mode the driver on the controller is using. It has both RGB and HueSat inputs.
But in general, I can't find information about color modes and color capabilities supported by esp zigbee libraries.

Where can I find a list of the types ?

Also, an example for RGB and HueSat endpoint configuration would be great.

### Additional context.

.

ilker-aktuna, 2024-05-05:
I tried the example from issue https://github.com/espressif/esp-zigbee-sdk/issues/327
I can't receive any color control commands from the controller..
If I send a hue / sat command, the EP reports:
```
I (101571) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (59861) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
```

but on level and on/off commands it works fine:
```
I (45271) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x8), attribute(0x0), data size(1)
I (45281) ESP_ZB_ON_OFF_LIGHT: Light level changes to 0
I (41451) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0x6), attribute(0x0), data size(1)
I (41461) ESP_ZB_ON_OFF_LIGHT: Light sets to On
```

mjdswan, 2024-05-06:
Hi Ilker,

If you didn't already find it, I can suggest trying the example project called **HA_color_dimmable_light** under the ZigBee SDK example projects.

Cheers, Mark.

ilker-aktuna, 2024-05-06:
@mjdswan I am already using it as a template. But it does not answer my questions. It seems like it uses only one color control mode and that is not supported by my controller's driver.
I need to learn and test other color modes and color capabilities

xieqinan, 2024-05-07:
@ilker-aktuna ,

Do you mean that when the "zigbee controller (Hubitat Hub)" sends the hue/sat command to the light, the "I (101571) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK" message will be triggered?

The log seems to indicate that the coordinator is requesting the light to leave its network. I recommend checking the logic of the "Hubitat Hub" to see if it requires the device's cluster with specific attributes or capabilities.

ilker-aktuna, 2024-05-07:
@xieqinan 
thanks for your reply. Actually I solved that "ZDO leave" issue. If I select the correct zigbee profile on the zigbee coordinator (Hubitat Hub) leave does not occur. I assume ESP decides to leave if the hub sends an incompatible command.

My current issue is that the color cluster does not receive any commands.
I see from the logs on Hubitat that it sends hue and saturation values (in hex) together with transition time to the "ESP_ZB_ZCL_CMD_COLOR_CONTROL_MOVE_TO_HUE_SATURATION" register (0x06) 
But on ESP side I do not receive this command.

On my attribute handler I have both ESP_ZB_ZCL_CLUSTER_ID_LEVEL_CONTROL and ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL checks.
level control is triggered but color control is NEVER triggered.

These are the commands that I used for registering the color control cluster:

```
uint8_t color_mode2 = 0x0000;
color_capabilities = 0x0018;
uint16_t zero = 0;
 uint8_t colorX = ESP_ZB_ZCL_COLOR_CONTROL_CURRENT_X_DEF_VALUE;
 uint8_t colorY = ESP_ZB_ZCL_COLOR_CONTROL_CURRENT_Y_DEF_VALUE;
 uint8_t colorOpts = 0x0001; // not sure what coloropts mean, but i found example with this being 0x1
 uint8_t colorECM = 0x0003; // init with standard hue/sat ecm
 uint8_t colorHue = 0x0050;
 uint8_t colorSaturation = 0x0000;
 uint16_t enhancedHue = 0x0000;

esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_MODE_ID, &color_mode2);
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_CAPABILITIES_ID, &color_capabilities);
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMPERATURE_ID, &color_attr);
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMP_PHYSICAL_MIN_MIREDS_ID, &min_temp);
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMP_PHYSICAL_MAX_MIREDS_ID, &max_temp);
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_HUE_ID, &colorHue);
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_SATURATION_ID, &colorSaturation);
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_CURRENT_HUE_ID, &enhancedHue);

    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_level_cluster(esp_zb_cluster_list, esp_zb_level_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_color_control_cluster(esp_zb_cluster_list, esp_zb_color_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_COLOR_DIMMABLE_LIGHT_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
```

I just don't know what color capability and color mode I have to set. It is not documented anywhere.
Any idea on that ?

xieqinan, 2024-05-07:
@ilker-aktuna ,
 
>  uint8_t colorOpts = 0x0001; // not sure what coloropts mean, but i found example with this being 0x1

The `coloropts` equals to 0x01 means that the color command will be handled, even the light is off.
 
Could you please verify whether the endpoint value matches with the one expected by Hubitat?

It would be beneficial to provide a sniffer packet as well, as it can help us quickly resolve this issue.

ilker-aktuna, 2024-05-07:
@xieqinan 
Actually, the Hubitat hub pairs with the ESP device when it initialize. It is sending the commands to the paired device. And on Hubitat device page shows "endpointId: 0A" This is equivalent to the ID set on header file (#define HA_ESP_LIGHT_ENDPOINT           10 )

I don't have a sniffer so I can't provide packets, but on the device driver (on Hubitat)  , I enabled logging.
When I send a "set color" command , Hubitat driver logs:

`[he cmd 0xB302 0x0A 0x0300 0x06 {00 F4 0A00}, 
delay 200, 
he cmd 0xB302 0x0A 0x0008 4 {0xE8 0x0A00}, 
delay 1400, 
he rattr 0xB302 0x0A 0x0300 0x0000 {}, 
delay 200, 
he rattr 0xB302 0x0A 0x0300 0x0001 {}, 
delay 200, 
he rattr 0xB302 0x0A 0x0008 0 {}, 
delay 200, 
he rattr 0xB302 0x0A 0x0300 0x0008 {}]`

the first line is the color setting (hue + sat values) sent to register 0x0300 (this is the color control cluster)
then it sends a level setting to 0x0008 (this is the level control cluster)

then it reads the attributes after a delay of 1400 ms.

The level command is working. But the color is not working.

If I change the color to blue, the log displays:
`
[he cmd 0xB302 0x0A 0x0300 0x06 {A8 AF 0A00}, 
delay 200, 
he cmd 0xB302 0x0A 0x0008 4 {0xDB 0x0A00}, 
delay 1400, 
he rattr 0xB302 0x0A 0x0300 0x0000 {}, 
delay 200, 
he rattr 0xB302 0x0A 0x0300 0x0001 {}, 
delay 200, 
he rattr 0xB302 0x0A 0x0008 0 {}, 
delay 200, 
he rattr 0xB302 0x0A 0x0300 0x0008 {}]`

xieqinan, 2024-05-08:
@ilker-aktuna ,

> [he cmd 0xB302 0x0A 0x0300 0x06 {A8 AF 0A00}, delay 200, he cmd 0xB302 0x0A 0x0008 4 {0xDB 0x0A00}, delay 1400, he rattr 0xB302 0x0A 0x0300 0x0000 {}, delay 200, he rattr 0xB302 0x0A 0x0300 0x0001 {}, delay 200, he rattr 0xB302 0x0A 0x0008 0 {}, delay 200, he rattr 0xB302 0x0A 0x0300 0x0008 {}]

The provided information may not be sufficient. Installing a sniffer is relatively straightforward; you can refer to the guide at https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark for detailed instructions.

ilker-aktuna, 2024-05-08:
I don't have another ESP32-H2 or ESP32-C6 to do that sniffing process.
Why isnt the provided information sufficient ?
I have all the zigbee commands sent out by the coordinator. (Hubitat)
You are looking for a proof I understand that but why don't you believe the logs ?

If you won'T be able to help with the problem , maybe you can point me to the values that color capabilities can get.
The documentation of zigbee support is not sufficient.

xieqinan, 2024-05-08:
@ilker-aktuna ，

> You are looking for a proof I understand that but why don't you believe the logs ?

I trust the log, but it doesn't provide enough information. While it does show the values of hue/sat in the ZCL layer, which is helpful, I also need confirmation on whether the command was sent by the Hubitat and received by the ESP device. Additionally, I need to verify that the IEEE802154, network, APS, and ZCL information of the command are correct. If the log mentioned above can resolve these issues, then I won't need any further information from you.

> If you won'T be able to help with the problem , maybe you can point me to the values that color capabilities can get.
The documentation of zigbee support is not sufficient.

After calling `esp_zb_device_register()`, you can use `esp_zb_zcl_get_attribute()` to retrieve the value of the `ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_CAPABILITIES_ID` attribute.

ilker-aktuna, 2024-05-08:
well, there is no reason to think that Hubitat does not send the command  to the network. Because this hub works with a lot of other zigbee devices (several different brands and models) and there is no reliability issue AFAIK.
So I can say that Hubitat is successful controlling other zigbee color ligths that follow the standards.
In addition to that I can clearly see that the level command works with my ESP32-C6 devkit.
Combining these 2 information, I really don't see any reason to doubt that Hubitat sends the command. I believe it sends the command and to the correct network and endpoint. (level command works between same hub and endpoint - so why not color)

> After calling esp_zb_device_register(), you can use esp_zb_zcl_get_attribute() to retrieve the value of the ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_CAPABILITIES_ID attribute.

And how will that help me ? I am not looking for what is set on the device. I am asking for a list of possible settings, and I can't find it anywhere.

> By the way, the esp-zigbee-sdk also provides the debug mode for user to achieve the debug purpose, you can refer to [debug mode](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#enable-debug-mode-and-trace-logging). More log will be outputed in this mode.

I am out at the office today. When I get back home, I can try debug mode. But if the device does not register to the color cluster , or does not use the correct color capability, it may not show any useful log. Correct ?

I am just asking for a list of possible color capabilities to try. It is a small information which should be available to users because otherwise how shall we use that setting ?

ilker-aktuna, 2024-05-08:
Think simple. There are 2 devices (Hubitat and ESP)
Hubitat sends a level command to ESP and it works.
In the same code, same device (Hubitat) sends a color control command to the same ESP device and it does not work.
There is no reason to search for a problem on the sending device hardware /software or the network. Even there is no reason to search for a problem on the receiving device hardware. I am 100% sure that Hubitat sends the packets to the "same" device with "same" network parameters. Only difference is the address and command format. 
And these are available in the log. If you don't see a problem about format of the command and used address (0x0300) then the problem is on the configuration of the receiving device. It probably does  use the correct capability/mode settings. 
Since there are not many zigbee examples on ESP Zigbee SDK documentation, (only 1 example with color control) I am 100% sure that the code I am using is not correct for this case.

ilker-aktuna, 2024-05-08:
> By the way, the esp-zigbee-sdk also provides the debug mode for user to achieve the debug purpose, you can refer to [debug mode](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#enable-debug-mode-and-trace-logging). More log will be outputed in this mode.

In menuconfig Zigbee options I don't see a "debug mode"
There is a zigbee trace enable opiton and a trace level (default -1) 
Is this the debug mode ? What should be the trace level ?

xieqinan, 2024-05-09:
@ilker-aktuna ,

Based on the log, it seems that no color control commands have been received in the ZCL layer. Could you please add `esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_INFO, ESP_ZB_TRACE_SUBSYSTEM_APS);` before calling `esp_zb_init();` to enable logging for APS?

> still can't find what color capabilities can be set...

There are two ways to set the color capabilities for the control cluster.

The first method involves adding the following code before `esp_zb_device_register()`.

```
uint16_t color_capabilities = 1 + 2 + 8; // support hue/sat, enhance_hue and color x/y mode
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_CAPABILITIES_ID, &color_capabilities);
```

The second method is to add the following code after `esp_zb_device_register()`.
```
uint16_t color_capabilities = 1 + 2 + 8; // support hue/sat, enhance_hue and color x/y mode
 esp_zb_zcl_set_attribute_val(HA_ESP_LIGHT_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_CAPABILITIES_ID, &color_capabilities, false);
```

ilker-aktuna, 2024-05-09:
> The first method involves adding the following code before `esp_zb_device_register()`
>
> ```
> uint16_t color_capabilities = 1 + 2 + 8; // support hue/sat, enhance_hue and color x/y mode [...]

When I set color_capabilities to 11 as you suggested, ESP crashes when it receives a color command.
color_capabilities= 1 + 2 + 8;
Log is here:
https://pastebin.com/EZkDSuGL

if I set color capabilities to 0x0018 as I see in an example, it does not crash but it does not also change color.
color_capabilities = 0x0018; 
Log is here:
https://pastebin.com/wmhPnAqA

I added traces as you asked but they do not mean much to me.
do you see the problem ?

xieqinan, 2024-05-10:
@ilker-aktuna 

> When I set color_capabilities to 11 as you suggested, ESP crashes when it receives a color command.

The crash is likely triggered by the absence of the "enhanced color mode" attribute. To add this attribute to the color control, you can use the following code:

```c
esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_COLOR_MODE_ID, &enhancedmode);
```

I have attached a color control example in the link provided (https://github.com/espressif/esp-zigbee-sdk/issues/327#issuecomment-2081348867). You can refer to it for your project.

ilker-aktuna, 2024-05-10:
@xieqinan 

thanks but what should be the value for &enhanced ?
the example you provided does not have this ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_COLOR_MODE_ID  attribute...

ilker-aktuna, 2024-05-10:
ok. I got it working , but only using the default dimmable light config as in the example.
I could not find what is missing if I individually add attributes instead of default light config.

working code:

> 
>     esp_zb_color_dimmable_light_cfg_t light_cfg = ESP_ZB_DEFAULT_COLOR_DIMMABLE_LIGHT_CONFIG();
>     light_cfg.color_cfg.color_capabilities += 3;
>     light_cfg.color_cfg.current_y = 21000;
>     light_cfg.color_cfg.options = 1;
>     esp_zb_attribute_list_t *attr_list = esp_zb_color_control_cluster_create(&(light_cfg.color_cfg));
>     uint8_t sat = 0x22;
>     uint8_t hue = 0x00;
>     uint16_t en_hue = 0x0000;
>     esp_zb_color_control_cluster_add_attr(attr_list, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_HUE_ID, &(hue));
>     esp_zb_color_control_cluster_add_attr(attr_list, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_SATURATION_ID, &(sat));
>     esp_zb_color_control_cluster_add_attr(attr_list, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_CURRENT_HUE_ID, &(en_hue));
>     esp_zb_cluster_list_add_color_control_cluster(esp_zb_cluster_list, attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

xieqinan, 2024-05-11:
@ilker-aktuna 

> ok. I got it working , but only using the default dimmable light config as in the example.
> I could not find what is missing if I individually add attributes instead of default light config.

I'm glad to hear that. The attribute marked with `M` in the figure (https://github.com/espressif/esp-zigbee-sdk/issues/338#issuecomment-2101841628) is necessary for the color control cluster, as well as for other clusters. You can refer to the default light configuration in the [code](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/ha/esp_zigbee_ha_standard.h#L105).

mjdswan, 2024-05-11:
Hi @xieqinan ,
I've been following this thread and also managed to use the supplied example to add Hue and Saturation controls to my light endpoint successfully.

Quick note: the .app_device_id appeared to be incorrectly set to ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID in the example.

What I now need to add is Color Temperature control, but I notice you mentioned that it's not supported.  Do we have some idea when it will be added?  It seems like a fairly common feature of color lights.

xieqinan, 2024-05-11:
@mjdswan ,

> What I now need to add is Color Temperature control, but I notice you mentioned that it's not supported. Do we have some idea > when it will be added? It seems like a fairly common feature of color lights.  

Do you mean the color temperature attribute of the color control cluster? The esp-zigbee-sdk supports it; you can refer to the [color temperature command](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L2367). If so, please note that the color capabilities need to be adjusted by adding (2 << 4), and the attributes `ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMP_PHYSICAL_MIN_MIREDS_ID` and `ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMP_PHYSICAL_MAX_MIREDS_ID` are necessary for supporting the color temperature function.

mjdswan, 2024-05-11:
@xieqinan many thanks, I will try it.

mjdswan, 2024-05-11:
@xieqinan, thanks again, I successfully added the color temperature control.
On checking the ZCL spec, I discovered that you must have meant (1 << 4) in your last post for the color capabilities.
"Color temperature supported" is bit 4, and this works.

ilker-aktuna, 2024-05-11:
Now that color control works, I will try to add electrical measurement cluster and try to measure DC current that ESP is using.
is that possible ?

xieqinan, 2024-05-15:
@ilker-aktuna ,

> Now that color control works, I will try to add electrical measurement cluster and try to measure DC current that ESP is using.
> is that possible ?

Yes, you can give it a try. The ESP-Zigbee-SDK supports electrical measurement cluster.

By the way, if you feel the issue has been resolved, please consider closing it.

ilker-aktuna, 2024-05-15:
> Yes, you can give it a try. The ESP-Zigbee-SDK supports electrical measurement cluster.

yes, I see that but I could not find a library/method to measure the DC current used by the module. At least not without adding anything to the circuit.
any ideas on that ?

I know this is not relevant to the zigbee discussion but before closing the issue, I'd like to ask...

xieqinan, 2024-05-30:
@ilker-aktuna 

> yes, I see that but I could not find a library/method to measure the DC current used by the module. At least not without adding anything to the circuit.

If you only need a method to measure DC current, I’m afraid I cannot provide a better idea. However, opening an issue in the [esp-idf](https://github.com/espressif/esp-idf) repository may be helpful for you.

ilker-aktuna, 2024-05-30:
Ok. Thank you.

closed 2024-05-30 by ilker-aktuna

mukeshbauskar, 2024-07-01:
@ilker-aktuna I am desperately trying to pair esp32c6 (flashed with [on off light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light) ) with hubitat c7 and c8 and it fails to pair. 
Can you please help me.

ilker-aktuna, 2024-07-16:
well, I am on a vacation away from home. can't add the code now. but the code I provided above should work.
what's not working for you ?
what do you get ?

mukeshbauskar, 2024-07-16:
@ilker-aktuna  I finally got it working. The issue was with Hubitat. It has problem working with zigbee 2.0 devices. There is  a workaround mentioned [here](https://community.hubitat.com/t/2-3-9-157-zigbee-plugs-are-found-but-do-not-function/139919/34). And it has been working smoothly. Thank you.

ilker-aktuna, 2024-07-16:
glad it finally worked for you.
and the solution is interesting. I will note it for future use.
However, I did not have this issue with my esp32-c6
why did you ?