espressif/esp-zigbee-sdk #339: Buffer corruption assert / crash while receiving fragmented message

Andelarrr, 2024-05-08:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### esp-zigbee-lib version.

1.3.1

### esp-zboss-lib version.

1.3.1

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

No buffer corruption.

### What is the actual behavior?

After a few messages are exchanged, there is an assert/crash with a buffer corruption message. This seems to happen after the first fragment of a new fragmented message is received, but the corruption could be already there from before.
I am not sure if this is a lib issue or an issue in our code. We don't work directly with the buffers though. Some messages are received in 5 fragments, not sure if this is relevant.

### Steps to reproduce.

The coordinator is up.
We start the ESP-C6 with our software.
A few messages get exchanged, after which the assert happens.

### More Information.

Trace/core dump:
[CoruptBufWithCoreDump.txt](https://github.com/espressif/esp-zigbee-sdk/files/15245481/CoruptBufWithCoreDump.txt)
Elf and map files:
[Esp32C6Zigbee.zip](https://github.com/espressif/esp-zigbee-sdk/files/15245486/Esp32C6Zigbee.zip)
Wireshark, I have selected the last line that is acked by the ESP:
[image]

lpy4105, 2024-05-11:
Hi @Andelarrr,

I've tested that the APS fragmentation basically works with up to 1000+ message length. Could you share the steps to reproduce the issue? 

We also want more details about the case:
- The information about the peer device
- The content of the exchanged message, if possible

Andelarrr, 2024-05-13:
Hi @lpy4105,

Thanks for your answer!

I have attached the pcap file for the exchange:
[zigbee-sniffer-2024-05-08_05-32-46.zip](https://github.com/espressif/esp-zigbee-sdk/files/15292065/zigbee-sniffer-2024-05-08_05-32-46.zip)

Network key is 00f18e3c94310b9075bb91fb9e665292

Our ESP is 40:4c:ca:ff:fe:40:7e:48 (0xcc33). The communication hub is a silab device, from DCC.

I am unsure as well if this is related to fragmentation, I just see that it happens after the first fragment of that message.

In lines 28, 30 and 65, 101 we see successful exchanges. After this in line 110 we send another request, the response starts in line 144. Here the first message is acked, but not the following ones anymore, as the assert happens at that point.

Can you perhaps give me more info on the assert that did happen? I am assuming that it is related to the message buffers, we are not manually allocating / freeing those. Maybe you will tell me that it has another meaning and I could find more clues based on that.

lpy4105, 2024-05-13:
Thanks for sharing the pcap file. After going over the traffics, I found that the ESP received and ACKed the APS fragment messages (couter = 6) correctly at line 108. But received the fragment 1 message with couter = 6 again at line 136. That's would be weired. I will check how the SDK would deal with this case. 

Meanwhile, could you try to enable the trace log? See [here](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32h2/developing.html#enable-debug-mode-and-trace-logging) for more detail. For time being, I think only APS trace is enough:
```C
#if CONFIG_ESP_ZB_TRACE_ENABLE
   esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_INFO, ESP_ZB_TRACE_SUBSYSTEM_APS);
#endif
```
>Can you perhaps give me more info on the assert that did happen? 

As far as I could tell, the problem should be APSDE-DATA.indication was triggered with a invalid buffer (id == 0) for some reason.

Andelarrr, 2024-05-13:
I've added the APS log and executed again, this is the trace and new pcap: [APS_trace.zip](https://github.com/espressif/esp-zigbee-sdk/files/15294619/APS_trace.zip)

The network key was this: B919C710954998E285504DF2C13E0990

Let me know if I should execute with other traces.

lpy4105, 2024-05-14:
Thanks for your quick tests. We do find some issues in our APS layer and plan to fix it in recent releases.

Andelarrr, 2024-05-14:
Thanks, let me know when there is a new version, I will test it.

lpy4105, 2024-05-27:
Hi @Andelarrr, The esp-zigbee-sdk v1.3.2 has fixed this issue. Could you please update the libraries and test it?

chshu, 2024-06-12:
Closing as the issue has been fixed.

closed 2024-06-12 by chshu