espressif/esp-zigbee-sdk #340: What call should be used to get attribute responses on a timed interval?

djorr5, 2024-05-09:
### Question

I have a gateway example successfully binding a Zigbee smart plug. As soon as it binds I am getting attribute updates via the ESP_ZB_CORE_REPORT_ATTR_CB_ID. These appear to be sent to me (the gateway) every 30 seconds. I assumed that this was as a result of me calling esp_zb_zcl_config_report_cmd_req with a .max_interval = 30. However when I changed this value to say 10 the attribute reporting still stayed at 30 seconds. 

So to test this further I commented I commented out out the esp_zb_zcl_config_report_cmd_req call completely and the attribute still reports every 30 seconds. 

Is this something that the zigbee end device just does automatically? Or does this happen as part of the binding? I can't seem to figure out where this attribute report is coming from. 

I am interested to know because if I reboot the smart plug the attribute reporting stops, and I would like to be able to start it up again. 

This might be similar to or related to #can

lpy4105, 2024-05-09:
Hi @djorr5, by default the stack wouldn't report any attributes unless:
- It recieves `Configure Reporting Command` from other devices, OR
- The device calls `esp_zb_zcl_update_reporting_info` to configure the report itself, OR
- Actively send `Report Attributes Command` 

These three use cases are all demostrated in [HA_thermostat](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_thermostat) and [HA_temperature_sensor](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor). 

If this doesn't answer you question, please share the logs of both devices so that we could know the use case clearly.

djorr5, 2024-05-13:
I assume then that the end device (TS011F) if calling the `esp_zb_zcl_update_reporting_info` (or a variant of it) on its own? Below is the logs when I first pair the device 
```
I (50844) ESP_ZB_GATEWAY: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (50854) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (51144) ESP_ZB_GATEWAY: New device commissioned or rejoined (short: 0xa326)
I (51194) ESP_ZB_GATEWAY: Found smart plug
I (51194) ESP_ZB_GATEWAY: Request smart plug to bind us
I (51194) ESP_ZB_GATEWAY: Bind smart plug
I (51204) ESP_ZB_GATEWAY: Successfully bind the smart plug from address(0xa326) on endpoint(1)
I (51284) ESP_ZB_GATEWAY: The Zigbee Smart plug from address(0xa326) on endpoint(1) successfully binds us
I (51294) ESP_ZB_GATEWAY: Read attribute response: from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x0)
I (51304) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x0), attribute(0x4), type(0x42), value(16)
I (51314) ESP_ZB_GATEWAY: Peer Manufacturer is "_TZ3000_8mukcg2g"
I (51314) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x0), attribute(0x5), type(0x42), value(6)
I (51324) ESP_ZB_GATEWAY: Peer Model is "TS011F"
I (52214) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (52264) ESP_ZB_GATEWAY: Network(0xd2d2) is open for 180 seconds
I (55934) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (55934) ESP_ZB_GATEWAY: Zigbee Smart plug is currently Off
I (55934) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (55944) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (55964) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (55974) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (55984) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (60584) FENCE VOLTAGE: Highest ADC Value in the last minute: 0rc endpoint(1) to dst endpoint(1) cluster(0x6)
I (60584) FENCE VOLTAGE: CPU Temperature value 29.60 ℃
I (60604) MQTT_CLIENT: MQTT_EVENT_PUBLISHED, msg_id=27884
I (60614) MQTT_CLIENT: MQTT_EVENT_PUBLISHED, msg_id=6268
I (81284) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (81294) ESP_ZB_GATEWAY: Zigbee Smart plug is currently Off
I (111294) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (111294) ESP_ZB_GATEWAY: Zigbee Smart plug is currently Off
I (120604) FENCE VOLTAGE: Highest ADC Value in the last minute: 0
I (120604) FENCE VOLTAGE: CPU Temperature value 30.60 ℃
I (120614) MQTT_CLIENT: MQTT_EVENT_PUBLISHED, msg_id=48638
I (120614) MQTT_CLIENT: MQTT_EVENT_PUBLISHED, msg_id=9232
I (141294) ESP_ZB_GATEWAY: Received report from address(0xa326) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (141294) ESP_ZB_GATEWAY: Zigbee Smart plug is currently Off
```
I have commented out the `configure_zigbee_reporting` which calls the `esp_zb_zcl_config_report_cmd_req` and that command is not called anywhere else in the code. However I see it reporting back via the `zb_attribute_reporting_handler`. Code snippet below:

```
void configure_zigbee_reporting(esp_zb_zcl_config_report_cmd_t *report_cmd, uint8_t src_endpoint, uint16_t cluster_id, bool report_change) {
    // Set the address mode
    report_cmd->address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
    
    // Set the source endpoint
    report_cmd->zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
    
    // Set the cluster ID
    report_cmd->clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
    
    // Define the report records
    esp_zb_zcl_config_report_record_t records[] = {
        {
            .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
            .attributeID = ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID,
            .attrType = ESP_ZB_ZCL_ATTR_TYPE_BOOL,
            .min_interval = 0,
            .max_interval = 10,
            .reportable_change = &report_change,
        },
    };
    
    // Set the number of records
    //report_cmd->record_number = sizeof(records) / sizeof(esp_zb_zcl_config_report_record_t);
    
    // Set the record field
    //report_cmd->record_field = records;

[... 19 lines cut ...]
            read_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
            read_req.zcl_basic_cmd.dst_endpoint = smart_plug.endpoint;
            read_req.zcl_basic_cmd.dst_addr_u.addr_short = smart_plug.short_addr;
            read_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_BASIC;

            uint16_t attributes[] = {
                ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID,
                ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID,
            };
            read_req.attr_number = ARRAY_LENTH(attributes);
            read_req.attr_field = attributes;

            esp_zb_zcl_read_attr_cmd_req(&read_req);
        }
        if (bind_req->req_dst_addr == smart_plug.short_addr) {
            ESP_LOGI(TAG, "The Zigbee Smart plug from address(0x%x) on endpoint(%d) successfully binds us",
                     smart_plug.short_addr, smart_plug.endpoint);
            //esp_zb_zcl_config_report_cmd_t report_cmd_instance;
[... cut ...] _CLUSTER_ID_ON_OFF, true);
        }
        free(bind_req);
```
It's annoying because when I power cycle the Zigbee power plug I no longer receive the reports. From memory I did try to call `configure_zigbee_reporting` (with the config_report_cmd no longer commented out) from `ESP_ZB_NLME_STATUS_INDICATION` but it didn't re-establish the reports. 

Is it advisable to periodically send the `report_attributes_command` every 30 seconds as long as `esp_zb_lock_acquire(portMAX_DELAY)` is adhered to?

lpy4105, 2024-05-13:
> Is it advisable to periodically send the `report_attributes_command` every 30 seconds as long as `esp_zb_lock_acquire(portMAX_DELAY)` is adhered to?

Generally, the `report_attributes_command` should be generated according to the report configuration, but it depends on the device implementaion.

I think there are several things you can try:
- If you are able to access the manual of TS011F, please help to check if there are special comments on the reporting function.
- Check the `Configure Reporting Response Command` from the peer device after you send `Configure Reporting Command`, you could see if the configurations succeed or not. This could be done by deal with `ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID`.
- Send `Read Reporting Configuration Command` and check the `Read Reporting Configuration Respo nse Command` from the peer device, you could see the actual report configuration on the peer device. But note that we are not officially support sending `Read Reporting Configuration Command`, you could use the zboss api or wait for our support on the next release.

djorr5, 2024-05-16:
Thanks @lpy4105 I couldn't get the plug to keep reporting its attributes on a power cycle. Will wait for the next release so I can implement 'Read Reporting Configuration Command' and in the meantime I will keep looking for a manual on the device to try to figure out why.

closed 2024-05-16 by djorr5

lpy4105, 2024-05-27:
Hi @djorr5, The esp-zigbee-sdk v1.3.2 provides the API to send 'Read Reporting Configuration Command'. Please refer to [here](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32h2/api-reference/zcl/esp_zigbee_zcl_command.html#_CPPv437esp_zb_zcl_read_report_config_cmd_reqP35esp_zb_zcl_read_report_config_cmd_t). You could update the library and have a try.