espressif/esp-zigbee-sdk #341: Attribute reporting not working using esp_zb_zcl_update_reporting_info

luar123, 2024-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### esp-zigbee-lib version.

1.2.3

### esp-zboss-lib version.

1.2.3

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

attributeReport send to the coordinator (zigbee2mqtt) when attribute is changed.

### What is the actual behavior?

I am using code similar to [HA_temperature_sensor](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor) with
```
/* Config the reporting info  */
    esp_zb_zcl_reporting_info_t reporting_info = {
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
        .ep = endpoint_id,
        .cluster_id = cluster_id,
        .cluster_role = role,
        .attr_id = attr_id,
        .manuf_code = ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC,
    };

    reporting_info.dst.short_addr = 0;
    reporting_info.dst.endpoint = 1;
    reporting_info.dst.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
    reporting_info.u.send_info.min_interval = 10; /*!< Actual minimum reporting interval */
    reporting_info.u.send_info.max_interval = 0; /*!< Actual maximum reporting interval */
    reporting_info.u.send_info.def_min_interval = 10; /*!< Default minimum reporting interval */
    reporting_info.u.send_info.def_max_interval = 0; /*!< Default maximum reporting interval */
    reporting_info.u.send_info.delta.s16 = 0; /*!< Actual reportable change */
    esp_zb_zcl_update_reporting_info(&reporting_info);
```
However, no report is received. Tried to debug with
```
esp_zb_zcl_reporting_info_t *info = esp_zb_zcl_find_reporting_info(attr_info);
    ESP_LOGD(TAG,"reporting Info: dir: %u, flags: %u, min_i: %u, max_i: %u, delta: %d, last: %d, addr: %u, ep: %u, profile: %u",
    info->direction, info->flags, info->u.send_info.min_interval, info->u.send_info.max_interval, info->u.send_info.delta.s16, info->u.send_info.reported_value.s16,
    info->dst.short_addr, info->dst.endpoint, info->dst.profile_id);
```
and got:
`reporting Info: dir: 0, flags: 7, min_i: 10, max_i: 0, delta: 0, last: 2511, addr: 0, ep: 1, profile: 260`

Using the coordinator to configure reporting with the same settings reporting is working and I get the same response.
Also tried `esp_zb_zcl_start_attr_reporting()` without success.
And what is the meaning of the flag value? I got multiple values while testing.

### Steps to reproduce.

1. configure reporting as described
2. join network
3. set attribute to new value

xieqinan, 2024-05-10:
@luar123 ，

I believe this issue is not a bug but rather a configuration issue. The reporting feature depends on cluster binding. Have you bound the cluster to the coordinator (zigbee2mqtt) on the end device side? You can refer to the example provided [here](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L123). Alternationly, check if the coordinator (zigbee2mqtt) has sent a `config_report_request` to the end device.

lpy4105, 2024-05-10:
Hi @luar123 

According to ZCL spec :
>2.5.7.1.6 Maximum Reporting Interval Field
...
If this value is set to 0x0000, and the minimum reporting interval field does not equal 0xffff there SHALL
be no periodic reporting, but change based reporting SHALL still be operational.

HOSTEER, 2024-05-13:
@luar123 you could try this, it worked for me.

- On the router or endDevice `static void esp_zb_task(void *pvParameters)`

```c
    esp_zb_temperature_meas_cluster_cfg_t temp_meas_cfg_server = {
        .measured_value = ESP_ZB_ZCL_TEMP_MEASUREMENT_VALUE_DEFAULT_VALUE,
        .min_value = ESP_ZB_ZCL_TEMP_MEASUREMENT_MIN_VALUE_DEFAULT_VALUE,
        .max_value = ESP_ZB_ZCL_TEMP_MEASUREMENT_MAX_VALUE_DEFAULT_VALUE,
    };
    esp_zb_attribute_list_t *esp_zb_temp_meas_server_cluster = esp_zb_temperature_meas_cluster_create(&temp_meas_cfg_server);
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_temperature_meas_cluster(esp_zb_cluster_list, esp_zb_temp_meas_server_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
``` 
- On the coordinator `static void esp_zb_task(void *pvParameters)`

```c
    esp_zb_temperature_meas_cluster_cfg_t temp_meas_cfg_client = {
        .measured_value = ESP_ZB_ZCL_TEMP_MEASUREMENT_VALUE_DEFAULT_VALUE,
        .min_value = ESP_ZB_ZCL_TEMP_MEASUREMENT_MIN_VALUE_DEFAULT_VALUE,
        .max_value = ESP_ZB_ZCL_TEMP_MEASUREMENT_MAX_VALUE_DEFAULT_VALUE,
    };
    esp_zb_attribute_list_t *esp_zb_temp_meas_client_cluster = esp_zb_temperature_meas_cluster_create(&temp_meas_cfg_client);
    esp_zb_cluster_list_add_temperature_meas_cluster(esp_zb_cluster_list, esp_zb_temp_meas_client_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
```

- finally in a custom task of a router or endDevice put 
```c
            temperature = (int16_t) tsens_value;
            esp_zb_lock_acquire(portMAX_DELAY);
            esp_zb_zcl_set_attribute_val(HA_ONOFF_SWITCH_ENDPOINT, 
                ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, 
                ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, 
                &temperature, 
                1);
            esp_zb_lock_release();
            esp_zb_zcl_report_attr_cmd_t cmd_req = ESP_ZB_TEMP_ATTR_CMD_CONFIG();
            esp_zb_lock_acquire(portMAX_DELAY);
            esp_zb_zcl_report_attr_cmd_req(&cmd_req);
            esp_zb_lock_release();
``` 

where tsens is a temperature multiplied by 100 and ESP_ZB_TEMP_ATTR_CMD_CONFIG is
```c
#define ESP_ZB_TEMP_ATTR_CMD_CONFIG()                                   \
    {                                                                   \
        .zcl_basic_cmd = {                                              \
            .dst_addr_u.addr_short = 0x0000,                            \
            .dst_endpoint = HA_ESP_LIGHT_ENDPOINT,                      \
            .src_endpoint = HA_ONOFF_SWITCH_ENDPOINT,                   \
        },                                                              \
        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,           \
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT,            \
        .attributeID = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID,       \
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,                 \
    }
``` 

In this code HA_ESP_LIGHT_ENDPOINT and HA_ONOFF_SWITCH_ENDPOINT define are base on esp's examples and 0x0000 is the coordinator addr.

the `esp_zb_zcl_reporting_info_t ` from what I understood is supposed to be used with binding, without implementing binding it will not work. It allow you to modify reporting config of remote devices but you can manually report things by creating a task which report every seconds or ten seconds your data. Then you don't need to define `esp_zb_zcl_reporting_info_t ` but your reporting interval will be hardcoded

luar123, 2024-05-13:
Thank you all for your input! I will try binding.
I have tried already `esp_zb_zcl_report_attr_cmd_t`, this is working great without binding, but it would be nice not to hardcode the interval.

closed 2024-05-30 by luar123

kgkask, 2024-06-14:
I have related question, 

Is it possible to define a custom structure attribute and report it ? I am sure it is doable for custom single attribute but I want to report a packet based on single value within it. So how to do it?

lpy4105, 2024-06-14:
@kgkask 
Please raise new issue and describe you issue clearly. It is not clear what is "custom structure attribute". If you meant to report an attribte of "Structure" type, the answer should be no, the ZCL spec doesn't support reporting attributes of "array", "structure", "set" and "bag" type.

kgkask, 2024-06-14:
You answered my question, Thanks that's what I was looking for.