espressif/esp-zigbee-sdk #343: Way to know when an end device has re-joined a coordinator

djorr5, 2024-05-16:
### Question

I am trying to figure out how to handling when an end device has rejoined the coordinator from say a power cycle. I noticed that when I power cycled my end device the coordinator showed up the 'ESP_ZB_NLME_STATUS_INDICATION'. So I added that with code to find out the status messages. 
```c++
case ESP_ZB_NLME_STATUS_INDICATION:
       printf("%s, status: 0x%x\n", esp_zb_zdo_signal_to_string(sig_type), *(uint8_t *)esp_zb_app_signal_get_params(p_sg_p));
``` 
This gave me 'NLME Status Indication, status: 0x0' which I assumed means  success. However when I was playing around I noticed that if I send an attribute read command to an end device that is not connected I also get the same message. 
```c++
I (2160947) ESP_ZB_GATEWAY: Send 'read attribute' command
I (2161007) ESP_ZB_GATEWAY: Device is no longer available
I (2161007) MQTT_CLIENT: Sent MQTT offline message
I (2161627) MQTT_CLIENT: MQTT_EVENT_PUBLISHED, msg_id=28694
I (2161637) MQTT_CLIENT: MQTT_EVENT_PUBLISHED, msg_id=55406
I (2161677) MQTT_CLIENT: MQTT_EVENT_PUBLISHED, msg_id=53431
NLME Status Indication, status: 0x0
NLME Status Indication, status: 0x0
NLME Status Indication, status: 0x0
NLME Status Indication, status: 0x0
``` 
Is this correct? Or is there another way to know when an end device re-joins?

xieqinan, 2024-05-17:
@djorr5 ,

> Is this correct? Or is there another way to know when an end device re-joins?

When the device joins or rejoins a network, it will send a device announcement request to the network. Has the coordinator received the `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE` signal when the end device joins the network?

djorr5, 2024-05-17:
No. I have a pretty standard `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE` based on the examples and I never see the ESP_LOGI when an end device rejoins. It is obviously called when commissioning the end device but that is the only time I have seen it called. Which is interesting as if that was the case it would want to re-bind every time it reconnects right? Even in your Thermostat example if a device rejoins from a power cycle with DEVICE_ANNCE it would re-bind each time. Or is this expected with Zigbee devices?

```c++
    case ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE:
        dev_annce_params = (esp_zb_zdo_signal_device_annce_params_t *)esp_zb_app_signal_get_params(p_sg_p);
        ESP_LOGI(TAG, "New device commissioned or rejoined (short: 0x%04hx)", dev_annce_params->device_short_addr);
        esp_zb_zdo_match_desc_req_param_t  cmd_req;
        cmd_req.dst_nwk_addr = dev_annce_params->device_short_addr;
        cmd_req.addr_of_interest = dev_annce_params->device_short_addr;
        find_temperature_sensor(&cmd_req, user_find_cb, NULL);
        break;
```

djorr5, 2024-05-22:
Looking at wireshark packets this end device appears to send a Link Status when re-joining and then a route request. I am unsure if either of these would be a DEVICE_ANNCE to the coordinator?
[image]
[image]

Then the end device proceeds to send attribute reports from then on in. Are any of the re-join packets captured in the ESP SDK?

xieqinan, 2024-05-23:
@djorr5 

> No. I have a pretty standard `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE` based on the examples and I never see the ESP_LOGI when an end device rejoins. It is obviously called when commissioning the end device [...]

The `device_annce` is an important signal for device re-join as defined in the BDB spec; you can refer to the figure below. The re-bind operation is application logic in the Thermostat example. You can choose to re-bind or not; it does not matter because the stack will only keep one binding for the same device.

[image]

There is no other signal currently triggered to indicate device re-join in the esp-zigbee-sdk.

djorr5, 2024-05-24:
Understood. It must be how they have implemented in this particular end device.

closed 2024-05-24 by djorr5