espressif/esp-zigbee-sdk #344: ESP32-C6 Zigbee witch DHT11, MG90S and 3 LED

sorgonas, 2024-05-18:
### Question

I'm currently in despair. I'm now trying to set up a Zigbee device with my ESP32 C6 for my mailbox project, which is currently still running via WiFi and MQTT and I would like to replace it.
I only managed to get the temperature displayed with the DHT11 (GPIO 3). just no matter what I try to get the servo MG90S into Zigbee fails.
I also need a button to move the servo (GPIO 16) up and one to move it down. I managed to do it without Zigbee, but I just can't keep up with my 90 degree movement.
The last hurdle, which I haven't started with yet, is to create 3 LEDs (GPIO 4,5,6) with a button each.

Maybe one of you can help me get this sorted

### Additional context.

this is my main.c which works with zigbee and dht11:
`
/*
 * SPDX-FileCopyrightText: 2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * Zigbee HA_temperature_sensor Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */
#include <zb_config_platform.h>
#include <stdio.h>
#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"
#include "dht.h"
#include "switch_driver.h"
#include "esp_zb_temperature_sensor.h"
#include "ha/esp_zigbee_ha_standard.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile sensor (End Device) source code.
#endif

#define DHT_GPIO_PIN GPIO_NUM_3
#define SENSOR_TYPE DHT_TYPE_DHT11

static const char *TAG = "ESP_ZB_TEMP_SENSOR";

static int16_t zb_temperature_to_s16(float temp)
{
    return (int16_t)(temp * 100);
}

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

static void esp_app_buttons_handler(switch_func_pair_t *button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        /* Send report attributes command */
        esp_zb_zcl_report_attr_cmd_t report_attr_cmd;
        report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
        report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID;
        report_attr_cmd.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
        report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
        report_attr_cmd.zcl_basic_cmd.src_endpoint = HA_ESP_SENSOR_ENDPOINT;

        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd);
[... 252 lines cut ...]
    // go low on compare threshold
    ESP_ERROR_CHECK(mcpwm_generator_set_action_on_compare_event(generator,
                    MCPWM_GEN_COMPARE_EVENT_ACTION(MCPWM_TIMER_DIRECTION_UP, comparator, MCPWM_GEN_ACTION_LOW)));

    ESP_LOGI(TAG, "Enable and start timer");
    ESP_ERROR_CHECK(mcpwm_timer_enable(timer));
    ESP_ERROR_CHECK(mcpwm_timer_start_stop(timer, MCPWM_TIMER_START_NO_STOP));
}

// bei bestimmten Ereignis hoch unter 
void servo_up()
{
    mcpwm_comparator_set_compare_value(comparator, example_angle_to_compare(-90));
}

// bei bestimmten Ereignis runter unter 
void servo_down()
{
    mcpwm_comparator_set_compare_value(comparator, example_angle_to_compare(0));
}

void app_main(void){
    servo_up(); // or servo_down();
}
`

quentains, 2024-06-03:
Hello there. 

I am almost in the same situation. I have few ESP32 at home, doing temperature and humidity data collection. Powering WIFI from a battery is not ideal in term of lifetime. I wanted to switch to a complete Zibgee setup, but after 2 complete days trying to make it work, only the temperature sensor worked (not the humidity one).

I mainly based my attempts on the example codes, but once you want to modify something, it is really painfull to understand how to do ! I simply decided to wait until some more documentation pops out.

Good luck with your project @sorgonas.

xieqinan, 2024-06-12:
@sorgonas ,

The issue is caused by the missing `binding` behavior for reporting attributes. In the `temperature_sensor and thermostat` example, the `thermostat` requests the binding from the `temperature_sensor` to the `thermostat`. You can refer to `https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c#L178`. As a result, the `temperature_sensor` knows where to send the report. In your project, you can call `esp_zb_zdo_device_bind_req()` to bind your device to MQTT in the `ESP_ZB_BDB_SIGNAL_STEERING` signal with `ESP_OK`. Below is the code snippet for reference:

```
        esp_zb_zdo_bind_req_param_t bind_req;
        bind_req.req_dst_addr = esp_zb_get_short_address();
        esp_zb_ieee_address_by_short(bind_req.req_dst_addr, bind_req.src_address);
        bind_req.src_endp = HA_ESP_SENSOR_ENDPOINT;
        bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        esp_zb_ieee_address_by_short(0x0000, bind_req.dst_address_u.addr_long); // if MQTT is coordinator, its short address is 0x0000
        bind_req->dst_endp = MQTT_EP; // please call `esp_zb_zdo_active_ep_req()` to get MQTT endpoint or find it in APP
        esp_zb_zdo_device_bind_req(bind_req, bind_cb, bind_req);
```

sorgonas, 2024-06-14:
Thank you @xieqinan  for your answer and your attempt to help me, but I don't even understand how I should put them together and where this code should go. The linked post that you linked to this one is also very complicated for me. And when I think that I actually wanted to integrate a servo, 3 LEDs and a battery indicator for my mailbox guard, I get even more dizzy.

I'll probably wait until there's a book about Zigbee that can teach you how it all works together from the beginning, etc. As I said, I'm not a programmer, I learned system administration.

xieqinan, 2025-09-30:
Hi @sorgonas @quentains ,

Sorry for the delay, the [ZCL reporting](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_general_report.html) document has been provided. Please follow it to understand the reporting mechanism. If it does not help resolve the problem, kindly share the difficulties you are facing.