espressif/esp-zigbee-sdk #345: ESP32-C6 Zigbee windows covering example

Lorem96, 2024-05-19:
### Question

Hi, I’m trying to make my smart blinds. I use zigbee2mqtt with a SONOFF ZBDongle and an ESP32-C6. Can you provide an example of code that will register in the ZigBee network as a ‘window covering’ device and react to OPEN, CLOSE, and STOP commands?

ninoweg, 2024-05-20:
I am also struggling to setup a window covering device. I use an ESP32-H2. With the custom_switch as a template I was able to add a window-covering cluster to it by adding

```
esp_zb_attribute_list_t *esp_zb_window_covering_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING);
esp_zb_cluster_list_add_window_covering_cluster(esp_zb_cluster_list, esp_zb_window_covering_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
```

I am able to join my zigbee network over ZHA but I'm not able to add attributes. I tried to add an attribute by adding the following code

```
esp_zb_window_covering_cluster_add_attr(esp_zb_window_covering_cluster, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_PHYSICAL_CLOSED_LIMIT_LIFT_ID, &test_attr3);
esp_zb_window_covering_cluster_add_attr(esp_zb_window_covering_cluster, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_PHY_CLOSED_LIMIT_TILT_ID, &test_attr3);
...
```

and also by setting up the cluster the following way:

```
/* window-covering cluster create with standard cluster config*/
esp_zb_window_covering_cluster_cfg_t window_covering_cfg;
esp_zb_attribute_list_t *esp_zb_window_covering_cluster = esp_zb_window_covering_cluster_create(&window_covering_cfg);
esp_zb_window_covering_cluster_add_attr(esp_zb_window_covering_cluster, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_WINDOW_COVERING_TYPE_ID, &test_attr3);
```

neither of which created the cluster with the expected attributes or in some cases even led to the device not being able to connect to the network. I would appreciate some guidance on this or a working example.

xieqinan, 2024-05-27:
@Lorem96  @ninoweg ，

The esp-zigbee-sdk does not provide a window covering example temporarily, but it is not difficult for users to implement the window covering data model themselves. You can refer to the code below for it.

```
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = endpoint_id,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_WINDOW_COVERING_DEVICE_ID,
        .app_device_version = 0,
    };
    /* Set the NULL to esp_zb_window_covering_cluster_create() means that using the default attribute value for window covering cluster */
    esp_zb_ep_list_add_ep(ep_list, esp_zb_window_covering_cluster_create(NULL), endpoint_config);
    esp_zb_device_register(ep_list);
    
```

ninoweg, 2024-05-27:
@xieqinan thanks for the response!

There seems to be an issues with the `esp_zb_window_covering_cluster_create` function call as the device runs into errors as soon as I add it to the code. My esp_zb_task function looks something like this:

```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);  

    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = HA_WINDOW_COVERING_ENDPOINT,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_WINDOW_COVERING_DEVICE_ID,
        .app_device_version = 0,
    };
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    /* Set the NULL to esp_zb_window_covering_cluster_create() means that using the default attribute value for window covering cluster */
    esp_zb_cluster_list_add_basic_cluster(cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_window_covering_cluster(cluster_list, esp_zb_window_covering_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);
    esp_zb_device_register(ep_list);

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(true));
    esp_zb_main_loop_iteration();
}
```

If I remove the window covering cluster and just add other clusters everything is working as expected and I can add my device to my zigbee network. As soon as I add the `esp_zb_window_covering_cluster_create` the monitor shows a bunch of errors. I attached the log: [eso32h2_zigbee_log.txt](https://github.com/espressif/esp-zigbee-sdk/files/15453965/eso32h2_zigbee_log.txt)

My idf_component.yml is 
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zboss-lib: "~1.3.0"
  espressif/esp-zigbee-lib: "~1.3.0"
  ## Required IDF version
  idf:
    version: ">=5.0.0"`
```

I installed everything as described here: https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32h2/developing.html

xieqinan, 2024-05-28:
@ninoweg 

Sorry for the inconvenience. You can use the code below to replace it and test again.
```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = HA_WINDOW_COVERING_ENDPOINT,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_WINDOW_COVERING_DEVICE_ID,
        .app_device_version = 0,
    };
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    /* Set the NULL to esp_zb_window_covering_cluster_create() means that using the default attribute value for window covering cluster */
    esp_zb_cluster_list_add_basic_cluster(cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_attribute_list_t *window_attr_list = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING);
    esp_zb_window_covering_cluster_cfg_t config = {
        .covering_type = ESP_ZB_ZCL_WINDOW_COVERING_WINDOW_COVERING_TYPE_DEFAULT_VALUE,
        .covering_status = ESP_ZB_ZCL_WINDOW_COVERING_CONFIG_STATUS_DEFAULT_VALUE,
        .covering_mode = ESP_ZB_ZCL_WINDOW_COVERING_MODE_DEFAULT_VALUE,
    };
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_WINDOW_COVERING_TYPE_ID,
                            ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &config.covering_type);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CONFIG_STATUS_ID,
                            ESP_ZB_ZCL_ATTR_TYPE_8BITMAP, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &config.covering_status);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_MODE_ID,
                            ESP_ZB_ZCL_ATTR_TYPE_8BITMAP, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &config.covering_mode);
    esp_zb_cluster_list_add_window_covering_cluster(cluster_list, window_attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);
    esp_zb_device_register(ep_list);

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

ninoweg, 2024-06-03:
@xieqinan thank you! 

I managed to pair the device to my network. I only had to change `ESP_ERROR_CHECK(esp_zb_start(false));` to `ESP_ERROR_CHECK(esp_zb_start(true));` to get it working.

I'm now running into another issue when trying to send commands to device but I'm not sure whether this is an esp-zigbee or rather an ZHA issue. I can set/get attributes of the window covering endpoint but as soon as I send a command (open/stop/close cover) I'm getting the following error message: `Failed to call service cover/open_cover. Failed to open cover: <Status.HARDWARE_FAILURE: 192>`. 

As it suggest an hardware failure I'm wondering if I'm still having issues with the window covering cluster (no issues ). When using one of the other examples the device works as expected. I'm not sure how to debug this issue and would appreciate some help. Here's the code I'm using:

```
/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * Zigbee customized client Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */
#include "esp_HA_customized_switch.h"
#include "esp_check.h"
#include "esp_err.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "string.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "zcl/esp_zigbee_zcl_common.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light switch (End Device) source code.
#endif

static const char *TAG = "ESP_HA_WINDOW_COVERING";
static const char *TAG_MOTOR = "MOTOR";

char modelid[] = {24, 'E', 'S', 'P', '3', '2', '-', 'H', '2', '-', 'W', 'i', 'n', 'd', 'o', 'w', '-', 'C', 'o', 'v', 'e', 'r', 'i', 'n', 'g'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

typedef struct light_bulb_device_params_s {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t  endpoint;
    uint16_t short_addr;
} light_bulb_device_params_t;

typedef struct zdo_info_ctx_s {
    uint8_t endpoint;
    uint16_t short_addr;
} zdo_info_user_ctx_t;

/* remote device struct for recording and managing node info */
light_bulb_device_params_t on_off_light;

static void zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    switch (button_func_pair->func) {
    case SWITCH_ONOFF_TOGGLE_CONTROL: {
        /* send on-off toggle command to remote device */
        esp_zb_zcl_on_off_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
        cmd_req.zcl_basic_cmd.dst_endpoint = on_off_light.endpoint;
        cmd_req.zcl_basic_cmd.src_endpoint = HA_WINDOW_COVERING_ENDPOINT;
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
[... 268 lines cut ...]
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_WINDOW_COVERING_TYPE_ID, ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &config.covering_type);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CONFIG_STATUS_ID, ESP_ZB_ZCL_ATTR_TYPE_8BITMAP, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &config.covering_status);
    esp_zb_cluster_add_attr(window_attr_list, ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING, ESP_ZB_ZCL_ATTR_WINDOW_COVERING_MODE_ID, ESP_ZB_ZCL_ATTR_TYPE_8BITMAP, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &config.covering_mode);
    
    esp_zb_cluster_list_add_window_covering_cluster(cluster_list, window_attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);
    esp_zb_device_register(ep_list);

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(true));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    xTaskCreate(esp_zb_task, "Zigbee_main", 16384, NULL, 5, NULL);
}
```

Lorem96, 2024-06-09:
@ninoweg 
I have implemented **tilt** percentage functionality with 

> `esp_zb_device_cb_id_handler_register` 

and **OPEN CLOSE** functionality with 

> `esp_zb_raw_command_handler_register`

It's not the best implementation but it works with zigbee2mqtt. 

Here is the full code:
```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "esp_zigbee_core.h"
#include "light_driver.h"
#include "stepper_motor_driver.h"
#include "switch_driver.h"
#include "zboss_api.h"
#include "esp_zigbee_cluster.h"
#include "zb_types.h"

// Zigbee configuration
#define INSTALLCODE_POLICY_ENABLE false
#define ED_AGING_TIMEOUT ESP_ZB_ED_AGING_TIMEOUT_64MIN
#define ED_KEEP_ALIVE 3000
#define HA_ESP_BLINDS_ENDPOINT 1
#define ESP_ZB_PRIMARY_CHANNEL_MASK ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK

#define ESP_ZB_DEFAULT_RADIO_CONFIG()       \
    {                                       \
        .radio_mode = ZB_RADIO_MODE_NATIVE, \
    }
#define ESP_ZB_DEFAULT_HOST_CONFIG()                          \
    {                                                         \
        .host_connection_mode = ZB_HOST_CONNECTION_MODE_NONE, \
    }

typedef struct
{
    int steps;
    int direction;
} MotorParams;

char modelid[] = {19, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'S', 'm', 'a', 'r', 't', 'B', 'l', 'i', 'n', 'd', 's'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

static MotorParams paramsCalibrationOpen = {999, 1};
static MotorParams paramsCalibrationClose = {999, -1};

static int stepsCount = 250;
static int closedStep = 0;
static int openStep = 250;
static int currentStep = 0;

static const char *TAG = "ESP_ZB_SMART_BLINDS";

esp_zb_cfg_t zb_nwk_cfg = {
    .esp_zb_role = ESP_ZB_DEVICE_TYPE_ED,
    .install_code_policy = INSTALLCODE_POLICY_ENABLE,
    .nwk_cfg.zed_cfg = {
        .ed_timeout = ED_AGING_TIMEOUT,
        .keep_alive = ED_KEEP_ALIVE,
    }};

[... 342 lines cut ...]
    }

    if (click_count == 3)
    {
        clear_motor_task();
        xTaskCreate(motor_control_task, "CalibrationMotorMove", 4096, &paramsCalibrationClose, 5, &motorControlTaskRef);
    }
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));

    light_driver_init();
    load_blind_step();
    switch_driver_init(button_func_pair, PAIR_SIZE(button_func_pair), button_click_handler);
    xTaskCreate(zigbee_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

```

xieqinan, 2024-06-12:
@ninoweg 

Please ensure that the "window covering" device is on the same network as the "smart blinds" first. Then, you need to determine the packet sent by the "smart blinds" to control the "window covering" device using a [sniffer](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark). If the control packet is sent correctly by the "smart blinds," you can refer to `https://github.com/espressif/esp-zigbee-sdk/issues/345#issuecomment-2156449315` to implement the function.

ninoweg, 2024-06-12:
@Lorem96 thank you! 

@xieqinan I'm only using a single device, something similar to [#345 (comment)](https://github.com/espressif/esp-zigbee-sdk/issues/345#issuecomment-2156449315). This also works for me. 

The issue seems to be the missing callback id in `esp_zb_core_action_handler_register()` as it is not listed in `esp_zb_core_action_callback_id_t`? So `esp_zb_core_action_handler_register(zb_action_handler);` can actually not be used for this? Could a support for the cover callback be added?

Regarding handling raw data as in [#345 (comment)](https://github.com/espressif/esp-zigbee-sdk/issues/345#issuecomment-2156449315) - are there any other examples or documentation on this?

RahulRGaikwad, 2024-06-20:
@ninoweg Hello, I'm facing a similar issue. I'm also getting 'Hardware Failure' as a code in my default response.

The callback ID is accurate, as the server is supposed to generate default response to any received command.

I think the command handlers are not completely implemented for the window covering cluster. 
@xieqinan any update on this?

ninoweg, 2024-06-21:
Hey @RahulRGaikwad! 

The example from @Lorem96 didn't quite work for me but I think I managed to fix some of the issues.

First of all I think it is necessary to use both the raw command handler as well as the core action handler

```
esp_zb_raw_command_handler_register(raw_command_handler);
esp_zb_core_action_handler_register(zb_action_handler);
```

First the command is handled by the raw command handler and in case it returns false e.g. because a certain command is not handled by it, the command is passed to the core action handler so it can try to deal with it. If you only use the raw command handler you would need to deal with all commands yourself which you probably do not want.

I then used the following function to handle (or rather debug) the raw messages

```
bool raw_command_handler(zb_uint8_t bufid)
{
    printf("raw cmd:\n");
    uint8_t buf[zb_buf_len(bufid)];
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);
    printf("cluster_id: 0x%x \n", cmd_info->cluster_id);
    printf("profile_id: 0x%x \n", cmd_info->profile_id);
    printf("cmd_id: %d \n", cmd_info->cmd_id);
    printf("cmd_direction: %d \n", cmd_info->cmd_direction);
    printf("seq_number: %d \n", cmd_info->seq_number);
    printf("is_common_command: %d \n", cmd_info->is_common_command);
    printf("disable_default_response: %d \n", cmd_info->disable_default_response);
    printf("is_manuf_specific: %d \n", cmd_info->is_manuf_specific);
    printf("manuf_specific: %d \n", cmd_info->manuf_specific);
    memcpy(buf, zb_buf_begin(bufid), sizeof(buf));
    ESP_LOGI("RAW", "bufid: %d size: %d", bufid, sizeof(buf));
    for (int i = 0; i < sizeof(buf); ++i) {
        printf("0x%02X ", buf[i]);
    }
    printf("\n");

    if (cmd_info->is_common_command)
    {
        return false;
    }

    if (cmd_info->cluster_id == ESP_ZB_ZCL_CLUSTER_ID_BASIC)
    {
        ESP_LOGW(TAG, "Basic Cluster\n\n");
        return false;
    }
    else if (cmd_info->cluster_id == ZB_ZCL_CLUSTER_ID_WINDOW_COVERING)
    {
        ESP_LOGW(TAG, "Window Covering\n\n");
        zb_uint8_t command_id = cmd_info->cmd_id;
        switch (command_id)
        {
        case ESP_ZB_ZCL_CMD_WINDOW_COVERING_UP_OPEN:
            ESP_LOGW(TAG, "Open up\n");
            return true;
            break;
        case ESP_ZB_ZCL_CMD_WINDOW_COVERING_DOWN_CLOSE:
            ESP_LOGW(TAG, "Down close\n");
            return true;
            break;
        case ESP_ZB_ZCL_CMD_WINDOW_COVERING_STOP:
            ESP_LOGW(TAG, "Stop\n");
            return true;
            break;
        case ESP_ZB_ZCL_CMD_WINDOW_COVERING_GO_TO_LIFT_PERCENTAGE:
            ESP_LOGW(TAG, "Go to lift percentage\n");
            return true;
            break;
        case ESP_ZB_ZCL_CMD_WINDOW_COVERING_GO_TO_LIFT_VALUE:
            ESP_LOGW(TAG, "Go to tilt value\n");
            return true;
            break;
        case ESP_ZB_ZCL_CMD_WINDOW_COVERING_GO_TO_TILT_PERCENTAGE:
            ESP_LOGW(TAG, "Go to tilt percentage\n");
            return true;
            break;
        case ESP_ZB_ZCL_CMD_WINDOW_COVERING_GO_TO_TILT_VALUE:
            ESP_LOGW(TAG, "Go to tilt value\n");
            return true;
            break;
        default:
            return false;
            break;
        }
    }
    else 
    {
        ESP_LOGW(TAG, "Other\n\n");
        return false;
    }
    return false;
}
```

I only got it to properly pair the device when not dealing with `is_common_command` in the raw command handler but I'm not quite sure why this is. I'm also getting some strange behavior of receiving unrequested commands, so there might still be some issues.

closed 2024-07-04 by espressif-bot

mukeshbauskar, 2024-07-14:
@xieqinan It will be great to have a example of this in the repo.

mukeshbauskar, 2024-07-15:
@xieqinan can you please share example of ESP_ZB_CORE_WINDOW_COVERING_MOVEMENT_CB_ID.

xieqinan, 2024-07-22:
@mukeshbauskar ,

The `windows covering` example is similar to the `onoff switch` example, so you can refer to it first. We will provide an example named `all device types` that will implement various devices and clusters via CLI in the near future.

renatomotorline, 2024-07-26:
> @mukeshbauskar ,
> 
> The `windows covering` example is similar to the `onoff switch` example, so you can refer to it first. We will provide an example named `all device types` that will implement various devices and clusters via CLI in the near future.

The `all device types` example would be great.

mukeshbauskar, 2024-07-26:
@renatomotorline  thank you...

teodor-hristov, 2024-11-20:
> @ninoweg
>
> Sorry for the inconvenience. You can use the code below to replace it and test again.
>
> ```
> static void esp_zb_task(void *pvParameters)
> {
>     /* initialize Zigbee stack */
>     esp_zb_cfg_t [...]

Hey, thank you for the fix. It works for me too, but why does it happen? It is literally the same as using esp_zb_window_covering_cluster_create.

marrog86, 2025-03-25:
Hello, 
I took the esp_zb_light example as starting point. 
I bound the device successfully to my ZB network. I use Zb2Mqtt and it recognizes successfully my device as smart blind and offers state/position/tilt buttons to me. 
Open/Close and Stop command works as expected. I get the commands. 
I also get the SP_ZB_ZCL_CMD_WINDOW_COVERING_GO_TO_LIFT_PERCENTAGE commands, but the payload is empty. 

Please help me to get the percentage value from zb2mqtt for Lift and Tilt. 

So I tried to use esp_zb_device_cb_id_handler_register as in the example, but it never gets called. 

This is my Code: 
`

#include "esp_zb_light.h"
#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "zboss_api.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light (End Device) source code.
#endif

#define HA_ESP_BLINDS_ENDPOINT 1

static const char *TAG = "ESP_ZB_ON_OFF_LIGHT";
/********************* Define functions **************************/
static esp_err_t deferred_driver_init(void)
{
    static bool is_inited = false;
    if (!is_inited) {
        //light_driver_init(LIGHT_DEFAULT_OFF);
        is_inited = true;
    }
    return is_inited ? ESP_OK : ESP_FAIL;
}

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, , TAG, "Failed to start Zigbee commissioning");
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Initialize Zigbee stack");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Deferred driver initialization %s", deferred_driver_init() ? "failed" : "successful");
            ESP_LOGI(TAG, "Device started up in%s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : " non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
            ESP_LOGW(TAG, "%s failed with status: %s, retrying", esp_zb_zdo_signal_to_string(sig_type),
                     esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb,
[... 155 lines cut ...]
    esp_zb_device_register(esp_zb_covering_ep);
    //esp_zb_device_cb_id_handler_register(device_callback); //funktioniert nicht!
    esp_zb_raw_command_handler_register(raw_command_handler);
    esp_zb_core_action_handler_register(zb_action_handler);
    
    
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    esp_zb_set_primary_network_channel_set(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));

    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

`

xieqinan, 2025-03-26:
Hi @marrog86 ,

Your issue is independent of this closed issue. Please consider opening a new issue for discussion.