espressif/esp-zigbee-sdk #346: Problems with joining the zigbee network

DesignerAnGav, 2024-05-20:
### Question

Hello. I have four esp32-c6 boards. I found an example of esp_zigbee_cli, which allows two devices to connect. The connection commands are as follows:

INITIATOR
> bdb -r zr
> bdb -t d
> bdb -c 11
> tlk -k 00112233445566778899aabbccddeeff
> bdb -s
> bdb -i init
> bdb -a init
> bdb -a initiator

TARGET
> bdb -r zr
> bdb -t d
> bdb -c 11
> bdb -m 10
> tlk -t help
> tlk -t 60
> tlk -k 00112233445566778899aabbccddeeff
> tlk -r help
I (47351) : RSSI threshold(default -64) of Touchlink target, which ranges from -127 to 127
I (47351) : Done
> tlk -r -54
> tlk -r get
> bdb -s
> bdb -a init
> bdb -a target

I have successfully used this code to connect two esp32-c6 devices.
Please tell me how to connect four esp32-c6 devices via the command line. Thank you in advance.

xieqinan, 2024-05-23:
@DesignerAnGav 

The above commands aim to use the Zigbee Touchlink method to form a network, but it is typically used for pairing two devices.

Could you share your application scenario for Touchlink? If you only need to form a network for multiple devices, you can call `bdb -a steering` for the third device to join the Zigbee network created by the above commands.

chshu, 2024-06-12:
Please reopen if any follow up questions.

closed 2024-06-12 by chshu